pulquero/BatteryAggregator #2: Operation in combination with a smartshunt

towatai, 2023-04-02:
Hello and first of all thank you for your work.

I noticed that an existing smartshunt is also recognized as a battery and integrated accordingly. This of course falsifies all measurements and accordingly it would be useful to include a way to ignore this, if present!

pulquero, 2023-04-02:
For future reference, how does the smartshunt fit into the picture? Is it being used to monitor something other than a battery? Or being used to monitor the same battery bank as the BatteryAggregator?

Scheppo, 2023-04-09:
Hi, in my system I have 2 smartshunt for 2 parallel battery’s.
The system shows 3 Battery’s because one battery is BMS-CAN connected.
The aggeregtor shows and add all 3 Battery’s but I need only the data’s from the 2 smartshunt battery’s.
So is it possible to show only the 2 smartshunt data in aggregator?

commit f3044a74 (2023-04-09): Fixes #2. Add config to exclude certain battery services from aggregation.
closed 2023-04-09 by pulquero