pulquero/BatteryAggregator #4: Rename Battery Aggregator and other little Things.

Scheppo, 2023-04-10:
Hi Pulquero,
the „exclude Battery“ Option is working fine, thx.

Few Things:
1. I Will Change the Battery Aggregator Name in „400AH BAT“ because my battery’s are 280 and 120 Ah.
2. The total AH is not adjustable.
[image]
3. There are no Data from the Can Battery here
[image]

Maybe I have not seen the Options to Change this, sorry.

pulquero, 2023-04-10:
If you could run dbus-spy and attach screenshots of your battery services then I can check which dbus paths I am missing and add them. Total AH should be computed from the individual battery AHs, but maybe your batteries don't publish that?

Scheppo, 2023-04-10:
Ok, I can see that the smartshunt don’t send the Capacity, this is strange.

Here are the Infos for the 280ah can Battery:
I need only the Min and Max Cellvoltage here.
[image]

Here ist the 280ah Can Battery over the Smartshunt:
I see no Capacity Info, maybe you can use the Name for the info?
[image]

Here ist the 120ah Battery only over Smartshunt:
[image]

I have cut some first entry’s for Alarm to fit in one Picture.

commit c41c16f8 (2023-04-10): Fixes #4. Support for configurable installed capacity.
closed 2023-04-10 by pulquero