pulquero/BatteryAggregator #6: Capacity

Scheppo, 2023-04-16:
### The Capacity is not showing in aggregator Battery, only the installed capacity:
[image]
### Maybe the System need it?

pulquero, 2023-04-16:
Issue here is only the CAN battery provides capacity for the 280ah battery. No way to know the current capacity of the 120ah battery. What I could do is take the total capacity to be soc * installedCapacity if that is good enough?

Scheppo, 2023-04-16:
When the System don’t want this value then you don’t must do anything.

ps: The 120Ah Battery has also a Can-Bms, I will try to connect it to the ve-can bus.
When this works, I can tell you and maybe I can do same with 120ah battery like 280ah battery.

pulquero, 2023-04-16:
Config should be:
```
{
"capacity": 400,
"virtualBatteries": {
"com.victronenergy.battery.V280AH": ["com.victronenergy.battery.ttyS5", "com.victronenergy.battery.socketcan_can1"],
"com.victronenergy.battery.V120AH": ["com.victronenergy.battery.ttyUSB1", "com.victronenergy.battery.socketcan_can0"]
}
}
```

Scheppo, 2023-04-16:
Wow, it works!
Maybe anybody will do the same, here are my Systemconfig:
Cerbo GX
2* 48V Battery’s parallel connected.
Battery 1 - 280Ah, Seplos Can-Bms over GX-Can-BMS and Smartshunt connected.
Battery 2 - 120Ah, China-Bms over GX-VE-Can and Smartshunt connected.
Batteryaggregator do the following things:
Make Virtualbattery1 = Battery1 with Volt, Ampere and SOC Data from Smartshunt and other Data from Can-Bms.
Make Virtualbattery2 = Battery2 with Volt, Ampere and SOC Data from Smartshunt and other Data from Can-Bms.
Then aggregate the 2 Virtualbatteries to one 400Ah Battery used as Batterymonitor.

I hope I explained everything correctly.
if not please correct.

closed 2023-04-16 by pulquero