pulquero/BatteryAggregator #8: Bug: Smart Shunt is detected as Battery

WaldemarFech, 2023-05-08:
I've got 2 Batteries and one SmartShunt - so the BatteryAggregator detects 3 Batteries.

One Battery is currently not charging- and discharging. So the values of the other Battery is doubled, because of the Shunt.
[image]

WaldemarFech, 2023-05-08:
My Fault: RTFM
sorry!

Configuration
Optionally create the file /data/setupOptions/BatteryAggregator/config.json.

To exclude the battery service com.victronenergy.battery.shunt1, add:

{
	"excludedServices": ["com.victronenergy.battery.shunt1"]
}

pulquero, 2023-05-08:
The other option is to create a virtual battery from the merged combination of the shunt + battery it is monitoring.

closed 2023-05-08 by pulquero

WaldemarFech, 2023-05-09:
> My Fault: RTFM sorry!
> 
> Configuration Optionally create the file /data/setupOptions/BatteryAggregator/config.json.
> 
> To exclude the battery service com.victronenergy.battery.shunt1, add:
> 
> { "excludedServices": ["com.victronenergy.battery.shunt1"] }

In my case it is a VE-direct Shunt. So i had to insert:

com.victronenergy.battery.ttyS7

It would be great if we could choose which battery to aggregate or to exclude in the Remote-Console