pulquero/BatteryAggregator #11: Victron Shunt always added to BatteryAggregator

SlapJackNpNp, 2023-06-09:
Hello,

first of all, thanks for this tool!

I just set it up with two dalys but i always have the victron smartshunts added to the battery aggregation. My config looks like this: 

{
	"excludedServices": ["com.victronenergy.battery.shunt1"]
}
{
	"virtualBatteries": ["com.victronenergy.battery.virtual1": ["com.victronenergy.battery.shunt1", "com.victronenergy.battery.leastPrecedence"]]
}

i tried removing the batteryshunt from the virtualbatteries entry aswell but to no succes. Not sure what i'm doing wrong.

Thank you for your help. 
[image]

pulquero, 2023-06-09:
Services mentioned in virtualBatteries are automatically excluded. What are the service names of all your devices (or please attach a screen shot of dbus-spy)?

SlapJackNpNp, 2023-06-09:
What exactly is dbus-spy. Do you mean the list of devices shown in the cerbo? Or ist it a command i can execute over ssh?

pulquero, 2023-06-09:
Command over ssh.

SlapJackNpNp, 2023-06-09:
[image]
This is what i get with dbus-spy i hope this helps?

pulquero, 2023-06-09:
You want something like this (I may have the shunts the wrong way round)
```
	{
		"virtualBatteries": {
			"com.victronenergy.battery.virtual1": ["com.victronenergy.battery.ttyS6", "com.victronenergy.battery.ttyUSB0"],
			"com.victronenergy.battery.virtual2": ["com.victronenergy.battery.ttyS7", "com.victronenergy.battery.ttyUSB2"]
		}
	}
```

SlapJackNpNp, 2023-06-09:
So what exactly will this do? Like my goal is to ignore the smart shunts for the aggregator and only use the dalys. Atm i have 2 packs, each with 1 daly and 1 smart shunt. I want ti get the data and the control from the dalys and use the shunts only to check the consumptions while i monitor the system. 

Currently the aggregator uses all 4 so i get each current twice. 

So yeah how can i get the aggregator to ignore the shunts and only use the dalys? 

Thank you very much

pulquero, 2023-06-09:
This will combine a shunt with a daly to form a virtual battery - it will take the consumption from the shunt, and fill in any other parameters from the daly so you have the merged combination of the two. So, you'll have a virtual battery for each pack. And the aggregator will just aggregate these two virtual batteries.

    Aggregator ---- Virt Batt 1 ---- Shunt
                                ---- Daly
                                           
               ---- Virt Batt 2 ---- Shunt
                                ---- Daly

SlapJackNpNp, 2023-06-09:
Oooh that sounds amazing. Gonna test that tomorrow. Thank you very much!

SlapJackNpNp, 2023-06-10:
Okay, so i added the command you suggested: 

{
"virtualBatteries": {
"com.victronenergy.battery.virtual1": ["com.victronenergy.battery.ttyS7", "com.victronenergy.battery.ttyUSB2"],
"com.victronenergy.battery.virtual2": ["com.victronenergy.battery.ttyS6", "com.victronenergy.battery.ttyUSB0"]
}
}

I sorted it, so that the BMS and the shunt fit together. That is the only command i hace in the config.json atm. 
[image]
The Aggregator still uses 3 Units somehow tho, meaning that the showed values are not correct. Unfortunately i don't know what the 3rd unit is the aggregator is using. 

This is what the dbus-spy shows. 
[image]

Is there btw a way to name the Virtaul Batteries somehow? 

Thank you very much again for the help!

Marccatamaran, 2023-11-21:
Hi,
almost the same isssue, can't succed to exclud my shunt from the equation ? I create the file config.json with nano in the folder specified and try many way to try to exclud my shunt, but no cigar! 

I would like to see only my two Daly Bms the Battery Aggregator.
please advise

Services
com.victronenergy.adc
com.victronenergy.battery.aggregator                                                                                                                                       Battery Aggregator
com.victronenergy.battery.ttyS5                                                                                                                                          SmartShunt 500A/50mV
com.victronenergy.battery.ttyUSB0                                                                                                                                         SerialBattery(Daly)
com.victronenergy.battery.ttyUSB1                                                                                                                                         SerialBattery(Daly)
com.victronenergy.ble
com.victronenergy.digitalinput.input01                                                                                                                                      Haut niveau d eau
com.victronenergy.digitalinput.input02                                                                                                                                          Bilge pump on
com.victronenergy.fronius
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.packageManager                                                                                                                                              Package Manager
com.victronenergy.platform
com.victronenergy.qwacs
com.victronenergy.settings
com.victronenergy.solarcharger.ttyS6                                                                                                                           SmartSolar Charger MPPT 100/50
com.victronenergy.system
com.victronenergy.tank.adc_builtin0_1                                                                                                                                      Generic Tank Input
com.victronenergy.tank.adc_builtin0_2                                                                                                                                      Generic Tank Input
com.victronenergy.tank.adc_builtin0_3                                                                                                                                      Generic Tank Input
com.victronenergy.tank.adc_builtin0_4                                                                                                                                      Generic Tank Input
com.victronenergy.temperature.adc_builtin0_7                                                                                                                        Generic Temperature Input
com.victronenergy.temperature.adc_builtin0_8                                                                                                                        Generic Temperature Input
com.victronenergy.temperature.ruuvi_d6d9450a143f                                                                                                                                     RuuviTag
com.victronenergy.temperature.ruuvi_d76471a7ef46                                                                                                                                     RuuviTag
com.victronenergy.temperature.ruuvi_e49b90c9f9b4                                                                                                                                     RuuviTag
com.victronenergy.temperature.ruuvi_f616a54e79c2                                                                                                                                     RuuviTag
com.victronenergy.vebus.ttyS4                                                                                                                                   MultiPlus 12/3000/120-50 120V

pulquero, 2023-11-21:
@Marccatamaran  So you have one shunt across both batteries? So are you looking to aggregate the two Dalys and then have the result merged with the shunt?

Marccatamaran, 2023-11-21:
Yes I have my two Daly BMS on 560ah each  and after I have my victron
shunt! This one totalise whatever I take from the two Daly BMS!

At the end it would good if I drive my voltage with the Battery aggregator
and use the SOC from the victron shunt!
The Daly BMS have a tendency to drift on the SOC.

Right now if I charge with 100 amps
Each Daly show 50  amps that shut show 100  amps and the batteraggregetor
show 200 amps.

pulquero, 2023-11-21:
Currently this is not possible, but I'll see if I can make some changes to allow it.

Marccatamaran, 2023-11-22:
Can you show how to remove the victron shunt from the equation of the Batteryaggragator?

pulquero, 2023-11-22:
Yes, for now you can add
"excludedServices": ["com.victronenergy.battery.shunt1"],

Marccatamaran, 2023-11-23:
Hi,
I created the file using nano and paste the "excludedServices": ["com.victronenergy.battery.shunt1"], and reboot the Cerbo GX Still have 3 Battery ?

[image]

[image]

[image]
[image]

What I do wrong?
Thanks in advance for your support!

                                                                                       [ Read 1 line ]
^G Get Help      ^O Write Out     ^W Where Is      ^K Cut Text      ^J Justify       ^C Cur Pos       M-U Undo         M-A Mark Text    M-] To Bracket   M-Q Previous     ^B Back
^X Exit          ^R Read File     ^\ Replace       ^U Paste Text    ^T To Spell      ^_ Go To Line    M-E Redo         M-6 Copy Text    ^Q Where Was     M-W Next         ^F Forward

pulquero, 2023-11-24:
If you don't have an existing file then the full contents should be:
``
{
"excludedServices": ["com.victronenergy.battery.shunt1"]
}
``
See (updated) examples here: https://github.com/pulquero/BatteryAggregator

pulquero, 2023-11-25:
Ok, it's here, version 2.2. It is now possible to merge shunt data with aggregated data, example config:

### Aggregate all available batteries with additional info provided by a shunt

	{
		"auxiliaryServices": ["com.victronenergy.battery.shunt1"]
	}

closed 2023-11-25 by pulquero