pulquero/BatteryAggregator #12: Aggregator Current issue

nik-bok, 2023-11-01:
Hi,

Would you be so kind to check why my aggregator is showing far more amps then two my battery banks or shunt?
On the other hand, Lynx Shunt current is very close almost 100% correct if adding Batt1 and Batt2.
I have two JK BMSes, btw.

[image]
[image]

For some reason in the Aggregator/System it shows 3 batteries.
[image]

and...

[image]

pulquero, 2023-11-02:
If you look at /System/Batteries under the aggregator in dbus-spy it should list which 3 services it is aggregating.

pulquero, 2023-11-25:
Closing as no further response received.

closed 2023-11-25 by pulquero