pulquero/BatteryAggregator #13: [Question] Is it possible to use Smartshunt only for Soc and current?

SergioRius, 2023-11-21:
I have two battery banks in my system, each with a jkbms. Then I have a Smartshunt.
Is it possible to aggregate the two jkbms and then "override" Soc and Current readings coming from the smartshunt?
What would be the configuration?

pulquero, 2023-11-21:
Not currently. But I'll add it.

pulquero, 2023-11-25:
Ok, it's here, version 2.2. This is now possible, example config:

### Aggregate all available batteries with additional info provided by a shunt

	{
		"auxiliaryServices": ["com.victronenergy.battery.shunt1"]
	}

SergioRius, 2023-11-25:
@pulquero Wow, that was fast. Thank you very much. And when I entered my console, it already had the v2.2 on it.

Just please confirm one thing and then one question:
Previously had added the shunt as ignored. I have to replace the key entry with the one you provided, or the ignore key must still be present?
Is there a way to restart the battery aggregator service for those changes to be applied without rebooting the Cerbo?

Thanks again!

SergioRius, 2023-11-25:
I'm afraid that it doesn't work for me. Perhaps I'm doing anything wrong.
```
cat /data/setupOptions/BatteryAggregator/config.json
{
        "auxiliaryServices": ["com.victronenergy.battery.ttyS6"]
}
```

[image]

Seems that the Battery Aggregator name has changed. I remember changing it before but it didn't had an effect.
As you can see, there is a good difference in my readings.

pulquero, 2023-11-25:
My bad, needed a quick fix, try v2.2.1. Your config is fine.

SergioRius, 2023-11-25:
> My bad, needed a quick fix, try v2.2.1. Your config is fine.

I'm not a python dev, but seems that what the code is doing is pre-populating the values before, but they are overwritten with the BMSs values, isn't it? That should explain why I thought I was seeing the voltage value fluctuate fast.

pulquero, 2023-11-25:
No, you don't need to explicitly exclude the shunt, that should be taken care of by line https://github.com/pulquero/BatteryAggregator/blob/2f29046e4b7f9fc00076bd0b2243c8f504625332/battery_service.py#L228.
DBusMonitor is responsible for monitoring values directly from dbus. As the aux services were previously being excluded, then the values from the shunt would not be monitored and updated.

Ah, yes, I'll add primaryServices, then people can sandwich the aggregator between whatever services as they see fit.

pulquero, 2023-11-25:
Ok, try v2.4. It comprehensively allows for serviceName: [dbusPaths, ...] anywhere you could previously just give a serviceName. See README for examples.

Unfortunately, I don't have enough gear to fully test it, so you'll have to be my guinea pig again :)

closed 2023-11-25 by pulquero

SergioRius, 2023-11-28:
> Unfortunately, I don't have enough gear to fully test it, so you'll have to be my guinea pig again :)

Don't worry, I can do all the tests. Unfortunately I'll be busy this week.
I'll post the results as soon as I can get some time.

SergioRius, 2023-11-28:
With this settings:
```
{
        "primaryServices": {"com.victronenergy.battery.ttyS6": ["/Soc"]}
        "excludedServices": ["com.victronenergy.battery.ttyS6"]
}
```
I get the following values:
[image]

Soc doesn't get overwritten but Amps get duplicated.

pulquero, 2023-11-28:
You don't need excludedServices, anything used as a primaryService is automatically excluded.
(you are also missing a , between the two lines)

SergioRius, 2023-11-29:
> You don't need excludedServices, anything used as a primaryService is automatically excluded. (you are also missing a , between the two lines)

My bad.

It seems to work as expected. Also the metric seems more stable, in screen and in output nodes on node-red.

Here, only Soc being overwritten:
[image]

Marccatamaran, 2023-12-14:
Hi,
Would it be possible to get the final setting of config.json?
thanks

pulquero, 2023-12-14:
See the readme, I think I have example jsons for all the common use-cases now. If yours is not covered, raise a new issue and I'll add it, thanks.