pulquero/BatteryAggregator #15: Shows wrong values for current

Dropdragger, 2024-01-28:
I have recently finished the installation of my Cerbo and observed that the Battery-Aggregator shows a wrong value for the current. This wrong value can also be seen in the VRM.

I am using two JK-BMS (B2A8S20P). Both BMS are connected with the Cerbo via RS485 to USB. The Cerbo gets the data from the "dbus-serialbattery". The values which are displayed in the Remote-Console for each device are correct. The sum of current are nearly equal to the value shown by the Smart-Shunt. The value which is shown by the BatteryAggregator appears to be about twice as high.

For the installation of the "BatteryAggregator" I used the PackageManager. 

What could have gone wrong here? - Any help?

BR

pulquero, 2024-01-28:
I'm going to guess it is adding the batteries and the shunt together, thus double counting. If you use dbus-spy from the command line, you can look at System/Batteries for BatteryAggregator and that will list what it is actually aggregating.

Also, check the examples at the bottom of the readme https://github.com/pulquero/BatteryAggregator, hopefully one of these configurations will do the trick.

Dropdragger, 2024-01-31:
Hi again,
first of all, thanks for your quick reply. And sorry for not reading the entire manual.

I have tried one of the listed solution but without success. Here is what I have done.

I have created a 'config.json' file with the following content and rebooted the CerboGX afterwards:

[image]

The output of the 'dbus-spy' shows:

[image]

and the details of BatteryAggregator:

[image]

I am wondering why there is shown '3' for the parameter 'SystemBatteryParallel'. The same value can be found in the 'BatteryAggregator' (named it 'BATTERY'):

[image]

In the same way, I am surprised that beside the 'B_SHUNT' and ''BATTERY' I also see three instances of the 'SerialBattery(Jkbms)' in the VRM, numbered [1], [2] and [4]:

[image]

The current shown in 'BATTERY' is still nearly twice as high compared to 'B-SHUNT:

[image]

What I am doing wrong?

Regards

pulquero, 2024-01-31:
Spotted the error, it's excludedServices.

Dropdragger, 2024-01-31:
That was a very stupid mistake and I'm sorry I wasted your time. I should have better done a copy&paste.
Things now look better in the remote console and also in the VRM. I can't tell from the current values at the moment because there's no sun shining at the moment and I can't turn on the inverter for maintenance at the moment. But I assume that it will be correct now.

Just one more comment: I noticed that the number of 'SerialBatteries' have now increased in my VRM display only. Do you know this effect? I'm just curious if this is normal. It looks like this:

[image]

Thanks for your help and best regards

pulquero, 2024-01-31:
No worries. The extra batteries maybe due to a change of ids. If some of them stop updating that is probably why, and somewhere in vrm you can delete the old copies.

Dropdragger, 2024-02-01:
Hi,
I have just checked the current values and now the figures are correct.
Thx

Dropdragger, 2024-02-01:
I close this issue.

closed 2024-02-01 by Dropdragger