pulquero/BatteryAggregator #17: Pace Gobel BMS on RS485, Battery Aggregator on RS232 ?

Starter4711, 2024-02-09:
Hello
would like to ask, it it would be possible to connect Battery Aggregator via RS232, not on RS485 there the Gobel Pace BMS is communicating with the Cerbo automatically. But I would like to see the cell voltages. Thx

pulquero, 2024-02-09:
Please attach a screenshot of dbus-spy showing the BMS details. Battery aggregator reads directly from dbus so doesnt care about RS232 vs RS485. No reason why it shouldnt work.

Starter4711, 2024-02-10:
I will do (later), the BA shows the values, same lithe the Cerbo already does if the Gobel is connected to Cerbo.
What I want is more, I would like to see the cell voltages, but this does not depend on BA? If the Can connection does not handover from BMS these values, they arent there and BA does not get these values directly from the Battery, than I need to look for an other addon.

pulquero, 2024-02-10:
Exactly right.

closed 2024-02-27 by pulquero