pulquero/BatteryAggregator #18: Problem when batteries are not charged synchronously

ckvsoft, 2024-02-14:
Today I noticed a problem - I have 2 serial batteries that can each be charged with 50A

Battery Aggregator shows me the correct sum of 100A

For some reason the two batteries were not switched to charging synchronously - then an attempt was made to charge the first of the batteries with the 100A - this caused the BMS to switch off - then an attempt was made to charge the second battery with 100A and it was also switched off by the BMS

This continued alternately

lg Chris

pulquero, 2024-02-14:
Please can you give complete details about your setup. Are you using DVCC?

ckvsoft, 2024-02-14:
Yes
3x Multiplus II 5000
2x jkbms

[image]
[image]

pulquero, 2024-02-14:
Can you also include dbus-spy of the batteries and multiplus. How are the batteries wired to multiplus so they can be charged independently? If we can find some sync charging flag in dbus, then I can use that to adjust the charging current reported by battery aggregator.

ckvsoft, 2024-02-14:
now it's getting difficult. I have no idea what to do with it.
The battery packs have their own BMS and are connected in parallel to the Victron via a strip.
If the serial battery regulates down, the aggregator does it correctly and divides it up. But if only one of them means that it should be charged then the amps, the 2x 50A, are delivered to a battery.

ckvsoft, 2024-02-14:
If serial battery regulates the value for one of the two packs down, this also works, only if one of the two does not go into charging mode then the full amps are charged into a battery pack. Which then has a high voltage error and switches off. Of course it could also be that the error is in the serial battery

ckvsoft, 2024-02-15:
This is one battery pack

[image]
[image]
[image]
[image]
[image]
[image]
[image]
5 at 10 01 49 (7)](https://github.com/pulquero/BatteryAggregator/assets/16374447/5e51c5d2-ecf5-46f4-8070-c6028a252ad3)
[image]
[image]

pulquero, 2024-02-15:
Thanks. I think I need to add support for Io/AllowToCharge and that should hopefully fix it.

pulquero, 2024-02-15:
Released a new version that should fix it, please test.

ckvsoft, 2024-02-15:
Thanks for the quick processing. I will test it. But it can take some time because it is not always the case that the batteries have different charging cycles.
I'll get in touch again tomorrow

ckvsoft, 2024-02-17:
So far there is no error. Thanks again, hoping that was it.

SergioRius, 2024-02-17:
> Released a new version that should fix it, please test.

I have a problem where one of the batteries enters float before the other and then that last one doesn't get charged anymore. The problem is that the driver get the CVL from the floating battery.
Do you think that new version could also solve it?

pulquero, 2024-02-17:
No, this sounds like a new separate issue. Probably need to be smarter with how the maxchargevoltage is being aggregated. New issue plus screenshots of this would be helpful.

Nurmi78, 2024-02-26:
Hi

have the similar issue like ckvsoft and already installed newest version - but noticed the same "deadlock behaviour" again:

my setup is also 3x Multiplus II 5000 with 2 battery packs with 2x jkbms (max charge current = 55A) and Louisvdw/dbus-serialbattery driver installed where I set MAX_BATTERY_CHARGE_CURRENT to 50A

aggregator shows correct CCL 100A

but then this happens:

[image]

as you can see batteries are not charged equally and once one battery hits 55Amps BMS stops charging and in the same second full 100A hits the second BMS which also goes down

you think you can help me with that?

Nurmi78, 2024-02-26:
max charge current of each battery is linear with max of 50A regarding the linear mode of dbus-serialbattery driver - so when both batteries are allowed to charge with 50A aggregator shows 100A - when driver puts the limit down for some reasons like cell voltage, temperature,... CCL in aggregator seems to be right for me

at the moment it looks like this:
battery 1
[image]
battery2
[image]
aggregator
[image]

as far as I can see aggregator always aggregates right from the batteries but I cant say if there is a delay because I have to go through several clicks to see parameters of batteries and aggregator

so the timing might be an issue because when one BMS goes down I can see way too much Amps going to second BMS

what I wonder about is why aggregator overrides the 50Amps charge limit of dbus-serialbattery driver - I think thats the beginning of the issue

Nurmi78, 2024-02-26:
no you can see in screenshot of my first posting what I mean with "override" - aggregator says CCL is 97.770A but first battery is charged with 53.90A although CCL of battery is set to 50A in driver

in this situation two days ago I had CCL of 50A in both batteries and CCL of 100A in aggregator

I have no plan where the "unequal" charging of 53.90A and 43.87A comes from

short after my screenshots charging of the first battery passes 55A and so BMS goes down

pulquero, 2024-02-26:
Ah gotcha. So maybe something like:
sum the charging currents, and scale to reduce any overcurrent, e.g. 97.77 * 50/53.9.
I can try adding that in addition to scaling back during balancing.

pulquero, 2024-02-26:
New version released.

Nurmi78, 2024-02-27:
thanks I already installed it
unfortunately this behaviour doesnt occur regularly so Im heading back to you once I know more

Nurmi78, 2024-03-03:
Hi

so far I didnt met all the conditions to load with full 100A where the problem occurs in the past

but today i noticed another strange thing...

CCL of first battery is 5A (due to wrong SOC delivered from BMS)
CCL of second battery is 42A (due to temperature limitation from serialbattery driver)

but CCL from aggregator says 23A ? how can this happen? in my opinion ist should aggregate to 47Amps, right?

pulquero, 2024-03-03:
This maybe caused by the new scale-back functionality. I've added more logging so you should know why.

Nurmi78, 2024-03-03:
log says:

@4000000065e4915105165094 INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e4915113f8e464 INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e4915122e17f2c INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e4915131d79afc INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e4915205356984 INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e49152142c5844 INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e49152231740e4 INFO:battery:Max charge current is 49.0 but scaling back to 19.276160503540517 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.5420000000000003)
@4000000065e49152320cf724 INFO:battery:Max charge current is 49.0 but scaling back to 18.992248062015502 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.58)
@4000000065e49153057c2eb4 INFO:battery:Max charge current is 49.0 but scaling back to 18.992248062015502 as limit exceeded for battery com.victronenergy.battery.ttyUSB1 (overcurrent ratio: 2.58)

pulquero, 2024-03-03:
That would suggest a ccl of 49A is causing about 13A on battery 1, so it has scaled to 19A to keep battery 1 within the 5A ccl. I've just added even more logging to make this even clearer.

Nurmi78, 2024-03-08:
in thias moment it happended again that aggregator tried to charge one battery with over 150 !! Amps although CCL of battery is 48A

log file without date or time is neot very useful to me ;)

here it is:

@4000000065e9ac7c2865935c INFO:battery:Max charge current is 79.167 but scaling back to 77.11339853892216 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0266309292546782)
@4000000065e9ac7c2865b29c INFO:battery:Battery currents:
@4000000065e9ac7c2865be54 com.victronenergy.battery.ttyUSB0: 41.75, limit 40.667
@4000000065e9ac7c2865ca0c com.victronenergy.battery.ttyUSB1: 34.29, limit 38.5
@4000000065e9ac7c2865d5c4 
@4000000065e9ac7d28c0a7c4 INFO:battery:Max charge current is 79.167 but scaling back to 76.4178587467363 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.035975114958074)
@4000000065e9ac7d28dc3e44 INFO:battery:Battery currents:
@4000000065e9ac7d28dc4de4 com.victronenergy.battery.ttyUSB0: 42.13, limit 40.667
@4000000065e9ac7d28dc5d84 com.victronenergy.battery.ttyUSB1: 33.72, limit 38.5
@4000000065e9ac7d28dc693c 
@4000000065e9ac7e28843334 INFO:battery:Max charge current is 79.167 but scaling back to 77.46593813763234 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0219588364029804)
@4000000065e9ac7e288e9b44 INFO:battery:Battery currents:
@4000000065e9ac7e288eaae4 com.victronenergy.battery.ttyUSB0: 41.56, limit 40.667
@4000000065e9ac7e288eba84 com.victronenergy.battery.ttyUSB1: 33.72, limit 38.5
@4000000065e9ac7e288ec63c 
@4000000065e9ac8028d3223c INFO:battery:Max charge current is 79.167 but scaling back to 77.82171595358957 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0172867435512822)
@4000000065e9ac8028de7894 INFO:battery:Battery currents:
@4000000065e9ac8028de8834 com.victronenergy.battery.ttyUSB0: 41.37, limit 40.667
@4000000065e9ac8028de97d4 com.victronenergy.battery.ttyUSB1: 33.53, limit 38.5
@4000000065e9ac8028dea38c 
@4000000065e9ac81293297bc INFO:battery:Max charge current is 79.167 but scaling back to 77.46593813763234 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0219588364029804)
@4000000065e9ac81293b2b0c INFO:battery:Battery currents:
@4000000065e9ac81293b3aac com.victronenergy.battery.ttyUSB0: 41.56, limit 40.667
@4000000065e9ac81293b4a4c com.victronenergy.battery.ttyUSB1: 33.33, limit 38.5
@4000000065e9ac81293b5604 
@4000000065e9ac82292443c4 INFO:battery:Max charge current is 79.167 but scaling back to 78.54316635764822 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0079425578478864)
@4000000065e9ac82292d773c INFO:battery:Battery currents:
@4000000065e9ac82292d8ac4 com.victronenergy.battery.ttyUSB0: 40.99, limit 40.667
@4000000065e9ac82292d967c com.victronenergy.battery.ttyUSB1: 33.33, limit 38.5
@4000000065e9ac82292da234 
@4000000065e9ac8329590f64 INFO:battery:Max charge current is 79.167 but scaling back to 76.76405314735337 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0313030221063761)
@4000000065e9ac832990bd4c INFO:battery:Battery currents:
@4000000065e9ac832990d0d4 com.victronenergy.battery.ttyUSB0: 41.94, limit 40.667
@4000000065e9ac832990dc8c com.victronenergy.battery.ttyUSB1: 33.33, limit 38.5
@4000000065e9ac832990e844 
@4000000065e9ac8529d7e424 INFO:battery:Max charge current is 79.167 but scaling back to 76.4178587467363 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.035975114958074)
@4000000065e9ac8529dcc23c INFO:battery:Battery currents:
@4000000065e9ac8529dcd1dc com.victronenergy.battery.ttyUSB0: 42.13, limit 40.667
@4000000065e9ac8529dce17c com.victronenergy.battery.ttyUSB1: 34.1, limit 38.5
@4000000065e9ac8529dced34 
@4000000065e9ac862a0df3fc INFO:battery:Max charge current is 79.167 but scaling back to 77.11339853892216 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0266309292546782)
@4000000065e9ac862a12a334 INFO:battery:Battery currents:
@4000000065e9ac862a12b2d4 com.victronenergy.battery.ttyUSB0: 41.75, limit 40.667
@4000000065e9ac862a12c274 com.victronenergy.battery.ttyUSB1: 33.91, limit 38.5
@4000000065e9ac862a12ce2c 
@4000000065e9ac872a792d34 INFO:battery:Max charge current is 79.167 but scaling back to 75.7347539167255 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0453193006614698)
@4000000065e9ac872a794c74 INFO:battery:Battery currents:
@4000000065e9ac872a79582c com.victronenergy.battery.ttyUSB0: 42.51, limit 40.667
@4000000065e9ac872a7963e4 com.victronenergy.battery.ttyUSB1: 33.91, limit 38.5
@4000000065e9ac872a796f9c 
@4000000065e9ac892ae2b8bc INFO:battery:Max charge current is 79.167 but scaling back to 77.46593813763234 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0219588364029804)
@4000000065e9ac892ae82b44 INFO:battery:Battery currents:
@4000000065e9ac892ae83ae4 com.victronenergy.battery.ttyUSB0: 41.56, limit 40.667
@4000000065e9ac892ae8469c com.victronenergy.battery.ttyUSB1: 33.72, limit 38.5
@4000000065e9ac892ae85254 
@4000000065e9ac8b2b3009b4 INFO:battery:Max charge current is 79.167 but scaling back to 78.54316635764822 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 1.0079425578478864)
@4000000065e9ac8b2b37c244 INFO:battery:Battery currents:
[... cut ...]
@4000000065ead408226a6acc com.victronenergy.battery.ttyUSB1: 40.04, limit 44.0
@4000000065ead408226a7684 
@4000000065ead40922785d1c INFO:battery:Max charge current is 92.0 but scaling back to 28.976377952755904 as limit exceeded for battery com.victronenergy.battery.ttyUSB0 (overcurrent ratio: 3.1750000000000003)
@4000000065ead409228c5a4c INFO:battery:Battery currents:
@4000000065ead409228c6dd4 com.victronenergy.battery.ttyUSB0: 152.4, limit 48.0
@4000000065ead409228c798c com.victronenergy.battery.ttyUSB1: -98.09, limit 44.0
@4000000065ead409228c8544

Nurmi78, 2024-03-08:
as far as i can see its same behaviour we had at begining...battery CCL says 46A but battery charging Amps goes up to 50-52A

still dont know where this comes from :/

pulquero, 2024-03-08:
@4000000065ead409228c5a4c INFO:battery:Battery currents:
@4000000065ead409228c6dd4 com.victronenergy.battery.ttyUSB0: 152.4, limit 48.0
@4000000065ead409228c798c com.victronenergy.battery.ttyUSB1: -98.09, limit 44.0

Looks like there is an added complication of one battery charging, while the other is discharging. If it is discharging, is /Io/AllowToCharge still 1?

Nurmi78, 2024-03-09:
I can take a look next time it happens but Im pretty sure this parameter is OK

you see the CCLs of 48 and 44 - aggregator said a correct CCL of 92 but for some reason after ttyUSB0 battery came online again immediatley all current from PV plus all possible currecnt from other battery (DCL = 100Amps) goes tom this battery forcing BMS to disconnect again

while ttyUSB0 was disconnected I noticed that ttyUSB1 was not charged with fully possible 44A - charging was floating between 27-38 Amps which is kind of strange too

I did a uninstall and fresh install of aggregator - after that charging was back to normal behaviour

Nurmi78, 2024-03-09:
have another strange behaviour right now...after 4 hours of charging aggregator suddenly says "not connected" and in log is also says no batteries discoverd although both batteries are still there in Venus OS:

@4000000065ec510015b267bc INFO:battery:Battery currents:
@4000000065ec510015b27b44 com.victronenergy.battery.ttyUSB2: 38.12, limit 35.833
@4000000065ec510015b28ae4 com.victronenergy.battery.ttyUSB1: 30.36, limit 38.167
@4000000065ec510015b2969c 
@4000000065ec51011560407c INFO:battery:Max charge current is 74.0 but scaling back to 68.85593352376006 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0747076716992716)
@4000000065ec51011565321c INFO:battery:Battery currents:
@4000000065ec5101156541bc com.victronenergy.battery.ttyUSB2: 38.51, limit 35.833
@4000000065ec510115654d74 com.victronenergy.battery.ttyUSB1: 30.36, limit 38.167
@4000000065ec51011565592c 
@4000000065ec510315a6ee3c INFO:battery:Max charge current is 74.0 but scaling back to 70.26078431372548 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0532190997125557)
@4000000065ec510315abdfdc INFO:battery:Battery currents:
@4000000065ec510315abef7c com.victronenergy.battery.ttyUSB2: 37.74, limit 35.833
@4000000065ec510315abff1c com.victronenergy.battery.ttyUSB1: 30.36, limit 38.167
@4000000065ec510315ac0ad4 
@4000000065ec510415d35954 INFO:battery:Max charge current is 74.166 but scaling back to 71.13464341541756 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0426143499009293)
@4000000065ec510415dad364 INFO:battery:Battery currents:
@4000000065ec510415dae304 com.victronenergy.battery.ttyUSB2: 37.36, limit 35.833
@4000000065ec510415daeebc com.victronenergy.battery.ttyUSB1: 29.22, limit 38.333
@4000000065ec510415dafa74 
@4000000065ec510515f6a094 INFO:battery:Max charge current is 74.166 but scaling back to 70.77470780292943 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0479167248067423)
@4000000065ec51051643c694 INFO:battery:Battery currents:
@4000000065ec51051643da1c com.victronenergy.battery.ttyUSB2: 37.55, limit 35.833
@4000000065ec51051643e5d4 com.victronenergy.battery.ttyUSB1: 29.22, limit 38.333
@4000000065ec51051643f18c 
@4000000065ec51071655fabc INFO:battery:Max charge current is 74.166 but scaling back to 71.49825875706215 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0373119749951163)
@4000000065ec5107165ad104 INFO:battery:Battery currents:
@4000000065ec5107165adcbc com.victronenergy.battery.ttyUSB2: 37.17, limit 35.833
@4000000065ec5107165aec5c com.victronenergy.battery.ttyUSB1: 29.03, limit 38.333
@4000000065ec5107165af814 
@4000000065ec5109169bf4cc INFO:battery:Max charge current is 74.166 but scaling back to 71.13464341541756 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0426143499009293)
@4000000065ec510916a0cefc INFO:battery:Battery currents:
@4000000065ec510916a0de9c com.victronenergy.battery.ttyUSB2: 37.36, limit 35.833
@4000000065ec510916a0ee3c com.victronenergy.battery.ttyUSB1: 28.84, limit 38.333
@4000000065ec510916a0f9f4 
@4000000065ec510b16eb4d4c INFO:battery:Max charge current is 74.166 but scaling back to 70.06565457421566 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0585214746183687)
@4000000065ec510b16f33c8c INFO:battery:Battery currents:
@4000000065ec510b16f34c2c com.victronenergy.battery.ttyUSB2: 37.93, limit 35.833
@4000000065ec510b16f35bcc com.victronenergy.battery.ttyUSB1: 29.79, limit 38.333
@4000000065ec510b16f36784 
@4000000065ec510d174607dc INFO:battery:Max charge current is 74.166 but scaling back to 70.41839634340221 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0532190997125557)
@4000000065ec510d174d7e04 INFO:battery:Battery currents:
@4000000065ec510d174d8da4 com.victronenergy.battery.ttyUSB2: 37.74, limit 35.833
@4000000065ec510d174d995c com.victronenergy.battery.ttyUSB1: 29.6, limit 38.333
@4000000065ec510d174da514 
@4000000065ec511117dee894 INFO:battery:Max charge current is 74.166 but scaling back to 69.71642911857292 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0638238495241816)
@4000000065ec511117e3e9d4 INFO:battery:Battery currents:
@4000000065ec511117e3f974 com.victronenergy.battery.ttyUSB2: 38.12, limit 35.833
@4000000065ec511117e40914 com.victronenergy.battery.ttyUSB1: 30.17, limit 38.333
@4000000065ec511117e414cc 
@4000000065ec51121811a8c4 INFO:battery:Max charge current is 74.166 but scaling back to 70.06565457421566 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0585214746183687)
@4000000065ec511218167b24 INFO:battery:Battery currents:
@4000000065ec511218168ac4 com.victronenergy.battery.ttyUSB2: 37.93, limit 35.833
@4000000065ec51121816967c com.victronenergy.battery.ttyUSB1: 30.17, limit 38.333
@4000000065ec51121816a61c 
@4000000065ec51141873e534 INFO:battery:Max charge current is 74.166 but scaling back to 69.71642911857292 as limit exceeded for battery com.victronenergy.battery.ttyUSB2 (overcurrent ratio: 1.0638238495241816)
@4000000065ec51141878e28c INFO:battery:Battery currents:
@4000000065ec51141878f614 com.victronenergy.battery.ttyUSB2: 38.12, limit 35.833
@4000000065ec5114187901cc com.victronenergy.battery.ttyUSB1: 30.36, limit 38.333
@4000000065ec511418790d84 
[... cut ...]
@4000000065ec51a6242ed334 INFO:main:Waiting for batteries (attempt 12 of 30)...
@4000000065ec51a7242eb00c INFO:main:Waiting for batteries (attempt 13 of 30)...
@4000000065ec51a8242f9e54 INFO:main:Waiting for batteries (attempt 14 of 30)...
@4000000065ec51a924280504 INFO:main:Waiting for batteries (attempt 15 of 30)...
@4000000065ec51aa24262c5c INFO:main:Waiting for batteries (attempt 16 of 30)...
@4000000065ec51ab242dcd7c INFO:main:Waiting for batteries (attempt 17 of 30)...
@4000000065ec51ac24266ec4 INFO:main:Waiting for batteries (attempt 18 of 30)...

Nurmi78, 2024-03-09:
I did a complete Raspi reboot now and log files say:

@4000000065ec54382d349be4 INFO:main:Waiting for batteries (attempt 16 of 30)...
@4000000065ec54392d332cb4 INFO:main:Waiting for batteries (attempt 17 of 30)...
@4000000065ec543a2d35f3a4 INFO:main:Waiting for batteries (attempt 18 of 30)...
@4000000065ec543b082c6a34 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@4000000065ec543b2562657c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 2
@4000000065ec543b2d39fae4 INFO:main:Waiting for batteries (attempt 19 of 30)...
@4000000065ec543b2dad523c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065ec543b314ebfd4 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065ec543c2f8b516c CRITICAL:ve_utils:exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
@4000000065ec543c2f8b70ac Traceback (most recent call last):
@4000000065ec543c2f8b881c   File "/data/BatteryAggregator/ext/ve_utils.py", line 22, in exit_on_error
@4000000065ec543c2f8b9f8c     return func(*args, **kwargs)
@4000000065ec543c2f8baf2c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 347, in _execute_value_changes
@4000000065ec543c2f8bbecc     self.valueChangedCallback(serviceName, objectPath,
@4000000065ec543c2f8d2a14   File "/data/BatteryAggregator/battery_service.py", line 378, in _battery_value_changed
@4000000065ec543c2f8d456c     self._update_battery_value(dbusServiceName, dbusPath, value)
@4000000065ec543c2f8d60c4   File "/data/BatteryAggregator/battery_service.py", line 392, in _update_battery_value
@4000000065ec543c2f8d7834     self._update_battery_path(dbusServiceName, p, self.service[p])
@4000000065ec543c2f8d87d4 AttributeError: 'BatteryAggregatorService' object has no attribute '_update_battery_path'
@4000000065ec543d2f4a971c INFO:main:Starting...
@4000000065ec543d3070e2f4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec543d30a06704 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@4000000065ec543d30c592a4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 2
@4000000065ec543d32b2c474 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065ec543d32eadbd4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1
@4000000065ec543d34c2d024 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec543e2d3474d4 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec543f2d341714 INFO:main:Waiting for batteries (attempt 2 of 30)...
@4000000065ec54402d319674 INFO:main:Waiting for batteries (attempt 3 of 30)...
@4000000065ec54412d36b30c INFO:main:Waiting for batteries (attempt 4 of 30)...
@4000000065ec54422d2de524 INFO:main:Waiting for batteries (attempt 5 of 30)...
@4000000065ec54432d30c76c INFO:main:Waiting for batteries (attempt 6 of 30)...
@4000000065ec54442d33480c INFO:main:Waiting for batteries (attempt 7 of 30)...
@4000000065ec54452d323e6c INFO:main:Waiting for batteries (attempt 8 of 30)...

pulquero, 2024-03-12:
I've pushed a new version for you to try. It is now event-based rather than polling so minimal lag to the aggregator reflecting changes to the batteries. And, more actively managing CCL, taking into account battery internal resistances for better accuracy.

Nurmi78, 2024-03-12:
thanks...is it still version 3.0.14 ?

pulquero, 2024-03-12:
Yep, I've bumped it to 3.0.15 anyway.

closed 2024-03-24 by pulquero