pulquero/BatteryAggregator #19: AB ver. 2.5 not working?

entropia1ac, 2024-02-17:
Hi, 
I am trying to install BatteryAggregator on Cerbo ver. 3 (3.2) but after installation it is not visible in the device list and in the battery monitor. It is not visible in the logs either.
What could be the reason for this ?
Thank You for your great work!

[image]
[image]
[image]
[image]

pulquero, 2024-02-17:
Did you check for errors in /var/log/BatteryAggregator/current?

You can also try restarting it with 
svc -d /service/BatteryAggregator
svc -u /service/BatteryAggregator

entropia1ac, 2024-02-17:
Last login: Sat Feb 17 15:50:57 2024 from 192.168.1.75
root@einstein:~# svc -d /service/BatteryAggregator
svc: warning: unable to chdir to /service/BatteryAggregator: file does not exist
root@einstein:~#

pulquero, 2024-02-17:
Looks like it didnt install properly, in the package manager, try clicking Download, then Install again.

entropia1ac, 2024-02-17:
Reinstallation helped.Thank you

closed 2024-02-17 by entropia1ac