pulquero/BatteryAggregator #21: Restart BatteryAggregator

Dropdragger, 2024-02-27:
Hi,
I have an issue that the aggregator does not recognize the serial batteries anymore. I can't figure out why. Also after rebooting the Cerbo the status is the same. Is there shell script to restart the extension without reboot? The additional problem after rebooting is, that the serial batteries will be assigned to different ".ttyUSBx" devices and I had to change the config.json file. 

[image]

pulquero, 2024-02-27:
Check the log for any errors:

	cat /var/log/BatteryAggregator/current

Restart the service:

	svc -d /service/BatteryAggregator
	svc -u /service/BatteryAggregator

Dropdragger, 2024-02-27:
Hi pulquero,
thank you for your quick reply.

Now I have restarted the service with the two commands, but the incorrect behavior still exists. But I have found in the logfile that something seems to be wrong. I could not figure out what the reason is. Here the logs:

[image]

Dropdragger, 2024-02-27:
It seems that the upload of the file has a problem..
I have copied the console output text now.

2024-02-27 11:49:27.702358500 INFO:battery:Exit
2024-02-27 11:49:54.047971500 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
2024-02-27 11:49:54.275329500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-02-27 11:49:54.285128500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
2024-02-27 11:49:54.292667500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 1
2024-02-27 11:49:54.383173500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
2024-02-27 11:49:54.388250500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
2024-02-27 11:49:54.447879500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2024-02-27 11:49:54.450305500 INFO:battery:Registered Battery Aggregator com.victronenergy.battery.aggregator
2024-02-27 11:49:54.716038500 Traceback (most recent call last):
2024-02-27 11:49:54.716047500   File "/data/BatteryAggregator/battery_service.py", line 327, in publish
2024-02-27 11:49:54.716050500     self.update()
2024-02-27 11:49:54.716052500   File "/data/BatteryAggregator/battery_service.py", line 312, in update
2024-02-27 11:49:54.716055500     ccl = maxChargeCurrentAggr.values[i]
2024-02-27 11:49:54.716057500 IndexError: list index out of range

pulquero, 2024-02-27:
Pushed a new version with a fix.

Dropdragger, 2024-02-27:
That was extremly quick! - I will try..

Dropdragger, 2024-02-27:
Update done! - Service is working again...
Thanks a lot. Great work!

closed 2024-02-27 by pulquero