pulquero/BatteryAggregator #22: Start battery aggregator after init of battery

dersch81, 2024-02-28:
Always if i restart the GX device i have a battery value of 0% from the battery aggregator because it starts before the battery driver initalized. 

[image]

Then i have a drop to 0% soc in statistics and the system starts to load the batteries in alarm mode for a few seconds with full power out of the grid. 

Is there any way to wait for the battery driver to start the aggregator after it?

pulquero, 2024-02-28:
What is the behaviour of the battery/bms in dbus? Is it visible in dbus then the driver starts and initialises its values? What is ttyACM0? Are we expecting it to behave differently from the other two ACM devices (it appears, disappears and appears again). Is there some flag in dbus that signifies when the driver is ready (perhaps /Connected?)

dersch81, 2024-02-28:
I'm using dbus-serial-battery driver. I have 3 battery packs connected and ACM0, ACM1, ACM2 are the USB connected BMS. 

In dbus the batteries are available after dbus-serial-battery has been started. But the battery-aggregator has been already started then and shows 0% soc until dbus-serial-battery is online. 

There is a "Connected = 0/1" flag at each battery/bms entry in dbus.

pulquero, 2024-02-29:
Please try v2.7.1.

dersch81, 2024-02-29:
Thanks a lot. 

It upgraded automatically via setup helper. I watched a reboot process and the battery aggregator started after the serial battery driver. It started with the correct SOC value in the correct order.

dersch81, 2024-03-03:
No it doesn't work. I had 3 reboots today and every time i have 0% soc:

[image]

pulquero, 2024-03-03:
Log please

dersch81, 2024-03-03:
```
@4000000065e202561fdfeb74 *** CCGX booted (30253) ***
@4000000065e202610f0f7224 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065e202611035fc7c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065e2026209e0f804 INFO:battery:Waiting for batteries (attempt 1 of 5)...
@4000000065e2026309d791c4 INFO:battery:Waiting for batteries (attempt 2 of 5)...
@4000000065e2026409cc5e94 INFO:battery:Waiting for batteries (attempt 3 of 5)...
@4000000065e202650a0aa014 INFO:battery:Waiting for batteries (attempt 4 of 5)...
@4000000065e2026609f9df04 INFO:battery:Waiting for batteries (attempt 5 of 5)...
@4000000065e2026609f9f674 WARNING:battery:No batteries discovered!
@4000000065e2026609fa022c INFO:battery:Exit
@4000000065e48f9c2e610df4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065e48f9c2e8c8eac INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
@4000000065e48f9c3283c534 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 3
@4000000065e48f9e0d37627c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065e48f9f2c34f694 INFO:battery:Waiting for batteries (attempt 1 of 5)...
@4000000065e48f9f2e569e14 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065e48fa01fb13a54 INFO:battery:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065e48fa034f2504c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM2, scanning and storing items
@4000000065e48fa22237b26c INFO:dbusmonitor:       com.victronenergy.battery.ttyACM2 has device instance 5
@4000000065e48fa40b2399c4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
@4000000065e48faa1e0a6c0c INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 4
@4000000065e48fad1e14b0f4 INFO:dbusmonitor:com.victronenergy.battery.ttyACM0 disappeared from the dbus. Removing it from our lists
@4000000065e48fad1f5a737c INFO:dbusmonitor:com.victronenergy.battery.ttyACM2 disappeared from the dbus. Removing it from our lists
@4000000065e48fad27b634d4 INFO:dbusmonitor:com.victronenergy.battery.ttyACM1 disappeared from the dbus. Removing it from our lists
@4000000065e48fd22e50a2d4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
@4000000065e48fd30189c6a4 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 3
@4000000065e48fd31d40bb3c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
@4000000065e48fd33068c4d4 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 4
@4000000065e48fd52c466384 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM2, scanning and storing items
@4000000065e48fd6048f1764 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM2 has device instance 5

```

pulquero, 2024-03-04:
I'll increase the number of attempts, looks like maybe it needs to wait for longer. How longer does the serial battery take to initialise?

dersch81, 2024-03-04:
Good question, my feeling is it can be 30s in some cases. But normally it is faster. 

I used this aggregator before https://github.com/Dr-Gigavolt/dbus-aggregate-batteries and this symptom never happend. I don't know what the difference could be during the start process.

pulquero, 2024-03-04:
I've increased it to 30 and made it configurable in the JSON with "startupBatteryWait". The other aggregator you mention tries up to 10 times, with 5 sec waits.

dersch81, 2024-03-04:
Many thanks for your effort in this. To have it configurable is also pretty cool. 

I will test and report in the next days.

closed 2024-03-16 by pulquero