pulquero/BatteryAggregator #23: How to change CVL?

dersch81, 2024-02-29:
Hi, i want to balance my cells. But the battery aggregator limit CVl to 54v. For balancing i need 55.3v 

My BMS and DVCC is settable but for the aggregator i cant. How can i do this?

pulquero, 2024-02-29:
Currently, to avoid overvoltage, the aggregator picks the lowest voltage. I should be able to detect balancing and pick the highest to fix this.

dersch81, 2024-03-01:
It will also detect balancing if not the BMS is balancing but an additional NEEY Balancer?

pulquero, 2024-03-01:
See if v2.8 helps.

dersch81, 2024-03-03:
Seems to work

dersch81, 2024-04-27:
It's not working anymore. DVCC is set to 55.3v to balance but the BatteryAggregator has 54v 

[image]

[image]

I want to have DVCC settings as a primary value. Is this possible?

pulquero, 2024-04-28:
Switch to head and try v3.0.55. I've added dvcc support.

dersch81, 2024-04-28:
I have installed it and will watch it. Many thanks for your quick response

superkrischi, 2024-05-09:
Hello, 

i have the same issue. How can i install v3.0.55? SetupHelper shows only v3.0.52 as latest.

Thanks and greetings
Kris

dersch81, 2024-05-09:
Just change branch to main

pulquero, 2024-05-17:
Closing as no further issues reported.

dersch81, 2024-06-11:
Im still in MAIN and it worked very good until the last update to 3.0.66 with taking the CVL setting of DVCC into account. 

But with 3.0.66 my batteries are just charged very slow with 45A max. When i change the battery monitor of DVCC to some battery pack it is starting to charge with full possible power to my CVL. 

What has been changed?

pulquero, 2024-06-11:
The change in v3.0.66 should only affect use of the auxiliaryServices option. I think the issue maybe to do more with that BatteryAggregator restarted. I've just pushed a fix that I think will fix your issue, you should see CCL updates now in the log that'll ensure it is set to the correct value.

dersch81, 2024-06-11:
Thanks, i just installed 3.0.67 and will watch. But the CVL setting of DVCC is still not taken.

pulquero, 2024-06-11:
What about CCL? Are there CCL updates in the log?

dersch81, 2024-06-12:
That is the output:

```
cat /var/log/BatteryAggregator/current
@4000000066693df52654187c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyACM0', 'com.victronenergy.battery.ttyACM1', 'com.victronenergy.battery.ttyACM2']
@4000000066693df526543ba4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(2.6610937228150093, 'ir'), (3.1768595449884347, 'ir'), (3.2316610448230927, 'ir')]
@4000000066693df526545314 INFO:com.victronenergy.battery.aggregator:CCL estimates: [226.1929664392758, 270.03306132401696, 274.69118880996285]
@4000000066693df526563774 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyACM0': 95.0, 'com.victronenergy.battery.ttyACM1': 95.0, 'com.victronenergy.battery.ttyACM2': 95.0}
@4000000066693df5265652cc INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyACM0', 'com.victronenergy.battery.ttyACM1', 'com.victronenergy.battery.ttyACM2']
@4000000066693df526566a3c INFO:com.victronenergy.battery.aggregator:Current ratios: [(2.6610937228150093, 'ir'), (3.1768595449884347, 'ir'), (3.2316610448230927, 'ir')]
@4000000066693df5268912fc INFO:com.victronenergy.battery.aggregator:DCL estimates: [252.80390366742589, 301.8016567739013, 307.0077992581938]

```

Unfortunatley it doesn't changed. It is charging with very low A

[image]

[image]

There is enough PV power available. It is just consumed 100 % because the battery charging ist limited by the aggregator and my dynamic powered water heater is taking the surplus.

pulquero, 2024-06-12:
I've added a dedicated cvlMode to track the DVCC voltage. If you set {"cvlMode": "dvcc"} in config then the battery aggregator should track whatever you dynamically set the Max charge voltage to.

pulquero, 2024-06-12:
{
"cvlMode": "dvcc",
"currentRatioMethod": "count"
}

Note the trailing comma.

dersch81, 2024-06-13:
Charging with CVL of DVCC and high A is back again. Many thanks

closed 2024-06-13 by dersch81