pulquero/BatteryAggregator #24: Battery aggregator gets stuck when SOC near 100%

dbachats, 2024-03-03:
Dear pulquero sorry to disturb.
I noticed that since the last version Battery aggregator gets stuck when SOC is near 100%. The only way to unblocked is to reboot venus os or to reinstall via setup helper..
My system with two 16s batteries in parallel was working flawlessly until recently  thanks to your plugin.
Is there something we can do to fix ?
Thanks very much for you work.

pulquero, 2024-03-03:
Please check log for errors: cat /var/log/BatteryAggregator/current

dbachats, 2024-03-03:
Hi thanks for the reply

This is what i get inside the current file : 
@4000000065e47ec116463794 INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing
@4000000065e47ec12544b004 INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing
@4000000065e47ec1343cb034 INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing
@4000000065e47ec2079e7e2c INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing
@4000000065e47ec216844e1c INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing
@4000000065e47ec2256f6984 INFO:battery:Min charge voltage is 57.92 but using max of 57.92 as balancing

pulquero, 2024-03-03:
v2.8.2 reduced logging to free-up cpu usage.

dbachats, 2024-03-03:
Thanks i ll try and keep you informed.
Thanks again

dbachats, 2024-03-04:
Sorry to report , battery aggregator continue to get stuck and  i get the BMS lost alarm and the system goes in passthru . This is getting critical

coca7, 2024-03-04:
I observed this behavior before BMS getting lost and going to pass thru.  I had not considered it to be connected with Battery aggregator.  I have the [v1.2.20240227beta] and that improved communication with the BMS.  I have only had this about a week, but I have not seen the error since I updated.  I have 3 JBD BMS's.  I will have to look at my logs when I get back to my system.

dbachats, 2024-03-05:
Thanks for the info coca7

coca7, 2024-03-05:
This is what my log file says at  cat /

> @4000000065dde85636f1b644 *** CCGX booted (0) ***
@4000000065dde86510191224 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065dde8651efb638c Traceback (most recent call last):
@4000000065dde8651efb7ee4   File "/data/BatteryAggregator/battery_service.py", line 429, in <module>
@4000000065dde8651efb8a9c     main()
@4000000065dde8651efb926c   File "/data/BatteryAggregator/battery_service.py", line 412, in main
@4000000065dde8651efb9e24     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000065dde8651efba9dc   File "/data/BatteryAggregator/battery_service.py", line 224, in __init__
@4000000065dde8651efd113c     self._init_settings(conn)
@4000000065dde8651efd1cf4   File "/data/BatteryAggregator/ext/settableservice.py", line 10, in _init_settings
@4000000065dde8651efd28ac     self.settings = SettingsDevice(conn, self.supportedSettings, self._setting_changed)
@4000000065dde8651efd384c   File "/data/BatteryAggregator/ext/settingsdevice.py", line 51, in __init__
@4000000065dde8651efd4404     raise Exception("The settings service com.victronenergy.settings does not exist!")
@4000000065dde8651efd9ddc Exception: The settings service com.victronenergy.settings does not exist!
@4000000065dde86f2443591c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065dde871227e44d4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065dde871233f52b4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065dde871236ca82c INFO:battery:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065dde8c02ca3d17c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@4000000065dde8ce290c00d4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 3
@4000000065dde8cf15003a74 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065dde8d53473353c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
@4000000065dde8d61d69bb54 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000065dde8d83089f8d4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
@4000000065de44e33ad5b2ec INFO:dbusmonitor:com.victronenergy.battery.ttyUSB2 disappeared from the dbus. Removing it from our lists
@4000000065de452d058eeb94 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000065de4533109f1e8c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
@4000000065de62f925d0d304 INFO:dbusmonitor:com.victronenergy.battery.ttyUSB1 disappeared from the dbus. Removing it from our lists
@4000000065de63481ad73164 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065de634e3b1ec5f4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
root@einstein:/#

pulquero, 2024-03-05:
I'll add a wait for the settings service.

dbachats, 2024-03-06:
The situation just got ugly today.  Unfortunately the problem persists.. ...   and the charging voltage went to high !!!!   Likely the Bms's did the job cutting off.

pulquero, 2024-03-06:
What were the individual battery cvls, the aggregator cvl and the contents of the log?

pulquero, 2024-03-06:
So, looking at the log, the battery aggregator never started. Add "startupBatteryWait": 60 to config.json and try that. I assume the batteries themselves are present in dbus?

dbachats, 2024-03-06:
Yes they are present.
I m worried. I never saw my system that unstable since the beginning.(For almost two years everything was running smoothly)..
I

pulquero, 2024-03-06:
I've created a v2.4 tag for you, so if you like, in package manger, change "latest" to "v2.4" and install the version that was working fine. Verify if that still does work fine for you and that will help eliminate any possible external factors.

closed 2024-03-16 by pulquero