pulquero/BatteryAggregator #25: CCL & DCL are not aggregated

johnboyi, 2024-03-06:
Hi!

I have 2 Systems running, and on both (both 3xMultiplus5000, one system with 2 Basengreen batteries, one with 3 Basengreen batteries) CCL & DCL are not aggregated? Any idea?

[image]

dbus-spy
[image]

Aggregator:
[image]

1st battery:
[image]
[image]
[image]

pulquero, 2024-03-06:
Check Io/AllowToCharge on each battery, if it is zero/off then its CCL is ignored (it should probably be taken to be zero, but I can fix that if this is the case).
Also, check the log
cat /var/log/BatteryAggregator/current

johnboyi, 2024-03-06:
[image]
Can this be changed?

here´s the log: 
[image]

pulquero, 2024-03-06:
If we are talking about a battery then the logic in dbus-serialbattery is

        self._dbusservice["/Io/AllowToCharge"] = (
            1
            if self.battery.charge_fet
            and self.battery.control_allow_charge
            and self.block_because_disconnect is False
            else 0
        )

It looks like these variables are not supported by lifepower bms hence it is 0. I might try to contact dbus-serialbattery author to try to understand why this is the case.

pulquero, 2024-03-14:
Has the latest version helped?

johnboyi, 2024-03-14:
Hi! Yes! It works great! I thank you very very much!!! Robert

  #25)

johnboyi, 2024-03-14:
Now it works! Thank you!

closed 2024-03-14 by johnboyi