pulquero/BatteryAggregator #26: Wrong voltage reading

11011010, 2024-03-06:
I sometimes have a problem with completely wrong voltage readings, almost gave me a heart attack the first time it happened.

[image]

11011010, 2024-03-06:
root@einstein:~# @4000000065e873cb3190ca14 com.victronenergy.battery.ttyUSB0: 5.6, limit 1.08
-sh: @4000000065e873cb3190ca14: command not found
root@einstein:~# @4000000065e873cb3190d5cc com.victronenergy.battery.ttyUSB1: 22.7, limit 32.4
-sh: @4000000065e873cb3190d5cc: command not found
root@einstein:~# @4000000065e873cb3190dd9c
-sh: @4000000065e873cb3190dd9c: command not found
root@einstein:~# @4000000065e873cc05b92044 INFO:battery:Max charge current is 33.21 but scaling b                                                 ack to 4.803589285714287 as limit exceeded for battery com.victronenergy.battery                                                 .ttyUSB0 (overcurrent ratio: 6.913580246913579)
-sh: syntax error near unexpected token `('
root@einstein:~# @4000000065e873cc06d025f4 INFO:battery:Battery currents:
-sh: @4000000065e873cc06d025f4: command not found
root@einstein:~# @4000000065e873cc06d04534 com.victronenergy.battery.ttyUSB0: 5.6, limit 0.81
-sh: @4000000065e873cc06d04534: command not found
root@einstein:~# @4000000065e873cc06d050ec com.victronenergy.battery.ttyUSB1: 22.7, limit 32.4
-sh: @4000000065e873cc06d050ec: command not found
root@einstein:~# @4000000065e873cc06d05ca4
-sh: @4000000065e873cc06d05ca4: command not found
root@einstein:~# @4000000065e873cc319e00e4 INFO:battery:Max charge current is 33.48 but scaling b                                                 ack to 2.5827428571428572 as limit exceeded for battery com.victronenergy.batter                                                 y.ttyUSB0 (overcurrent ratio: 12.962962962962962)
-sh: syntax error near unexpected token `('
root@einstein:~# @4000000065e873cc31adcfc4 INFO:battery:Battery currents:
-sh: @4000000065e873cc31adcfc4: command not found
root@einstein:~# @4000000065e873cc31adef04 com.victronenergy.battery.ttyUSB0: 14.0, limit 1.08
-sh: @4000000065e873cc31adef04: command not found
root@einstein:~# @4000000065e873cc31adf6d4 com.victronenergy.battery.ttyUSB1: 18.8, limit 32.4
-sh: @4000000065e873cc31adf6d4: command not found
root@einstein:~# @4000000065e873cc31ae028c
-sh: @4000000065e873cc31ae028c: command not found
root@einstein:~# @4000000065e873cd31be0fec INFO:battery:Max charge current is 33.48 but scaling b                                                 ack to 12.468413793103448 as limit exceeded for battery com.victronenergy.batter                                                 y.ttyUSB0 (overcurrent ratio: 2.685185185185185)
-sh: syntax error near unexpected token `('
root@einstein:~# @4000000065e873cd31d3bacc INFO:battery:Battery currents:
-sh: @4000000065e873cd31d3bacc: command not found
root@einstein:~# @4000000065e873cd31d3da0c com.victronenergy.battery.ttyUSB0: 2.9, limit 1.08
-sh: @4000000065e873cd31d3da0c: command not found
root@einstein:~# @4000000065e873cd31d3e5c4 com.victronenergy.battery.ttyUSB1: 19.4, limit 32.4
-sh: @4000000065e873cd31d3e5c4: command not found
root@einstein:~# @4000000065e873cd31d3ed94
-sh: @4000000065e873cd31d3ed94: command not found
root@einstein:~# @4000000065e873fb3a913f7c Traceback (most recent call last):
-sh: syntax error near unexpected token `('
root@einstein:~# @4000000065e873fb3a91668c   File "/data/BatteryAggregator/battery_service.py", l                                                 ine 359, in publish
-sh: @4000000065e873fb3a91668c: command not found
root@einstein:~# @4000000065e873fb3a917244     self.update()
-sh: syntax error near unexpected token `('
root@einstein:~# @4000000065e873fb3a917a14   File "/data/BatteryAggregator/battery_service.py", l                                                 ine 324, in update
-sh: @4000000065e873fb3a917a14: command not found
root@einstein:~# @4000000065e873fb3a9185cc     batteryOvercurrentRatio = batteryCurrents[i]/ccl
-sh: @4000000065e873fb3a9185cc: command not found
root@einstein:~# @4000000065e873fb3a919184 ZeroDivisionError: float division by zero
-sh: @4000000065e873fb3a919184: command not found
root@einstein:~#

pulquero, 2024-03-06:
Pushed a fix for the error. Not sure where 59.15V is coming from, it should be min of the two battery voltages, unless balancing then it is the max of the two. Does it eventually drop to 54.8V? The current is also not correct.

11011010, 2024-03-07:
I'm also not sure where that is coming from. It happened one time in the night with 69v while both my batteries were 53v.
[image]

[image]