pulquero/BatteryAggregator #27: v3 error no Battery

stixif, 2024-03-09:
i use https://github.com/Louisvdw/dbus-serialbattery 
wit 2 batterys 

@4000000065ec5f0c16c863cc WARNING:main:No batteries discovered!
@4000000065ec5f0c1713209c INFO:main:Exit
@4000000065ec5f0d25a3373c INFO:main:Starting...
@4000000065ec5f0d271e270c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec5f0d274e33d4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
@4000000065ec5f0d278086a4 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 3
@4000000065ec5f0d2a987cd4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
@4000000065ec5f0d2acf6f3c INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 1
@4000000065ec5f0d2df7cca4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec5f0f16c44134 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec5f1016ccd484 INFO:main:Waiting for batteries (attempt 2 of 30)...

leiweke, 2024-03-09:
same here
2 x JKBMS
@4000000065ec655a2314b48c INFO:main:Starting...
@4000000065ec655a24074634 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec655a2424f7c4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000065ec655a244ad714 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
@4000000065ec655a26474354 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065ec655a2670f71c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
@4000000065ec655a283ffa5c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec655b30051164 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec655c30063e2c INFO:main:Waiting for batteries (attempt 2 of 30)...

pulquero, 2024-03-09:
I've pushed v3.10, if that doesn't help, please include your configs so I can check the configs I test with include your use-case.

leiweke, 2024-03-09:
just updated to 3.0.10
not working:
@4000000065ec6da8137e2684 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065ec6da9018317b4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1
@4000000065ec6da909b5d50c INFO:main:Waiting for batteries (attempt 29 of 30)...
@4000000065ec6da909fd7ccc INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065ec6da90f6e300c INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065ec6da911cdee14 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000065ec6da912f9fa94 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 2
@4000000065ec6dbc23765c8c CRITICAL:ve_utils:exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
@4000000065ec6dbc237677e4 Traceback (most recent call last):
@4000000065ec6dbc2376839c   File "/data/BatteryAggregator/ext/ve_utils.py", line 22, in exit_on_error
@4000000065ec6dbc23768f54     return func(*args, **kwargs)
@4000000065ec6dbc23769b0c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 345, in _execute_value_changes
@4000000065ec6dbc2376aaac     self.valueChangedCallback(serviceName, objectPath,
@4000000065ec6dbc23785c44   File "/data/BatteryAggregator/battery_service.py", line 378, in _battery_value_changed
@4000000065ec6dbc23786fcc     self._update_battery_value(dbusServiceName, dbusPath, value)
@4000000065ec6dbc23787b84   File "/data/BatteryAggregator/battery_service.py", line 392, in _update_battery_value
@4000000065ec6dbc23788b24     self._update_battery_path(dbusServiceName, p, self.service[p])
@4000000065ec6dbc237896dc AttributeError: 'BatteryAggregatorService' object has no attribute '_update_battery_path'
@4000000065ec6dbd2699ff04 INFO:main:Starting...
@4000000065ec6dbd27aeaf0c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec6dbd27c9a94c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000065ec6dbd27fed694 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 2
@4000000065ec6dbd2ae24f44 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000065ec6dbd2b14195c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1
@4000000065ec6dbd2dae11a4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec6dbe2036cda4 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec6dbf20656f24 INFO:main:Waiting for batteries (attempt 2 of 30)...

ThePhloggy, 2024-03-09:
I'm still encountering the same with v3.10
My config.ini is commented out for troubleshooting:
```
#{
#"excludedServices": ["com.victronenergy.battery.ttyS6", "com.victronenergy.battery.ttyS7"]
#}
```
I have 2X JBD "Overkill Solar" BMSs connected via dbus-serialbattery / Bluetooth.  Both are operating as expected.
I've tried  restarting the service, commenting out the SmartShunt exclusions in config.ini,  a reboot, and uninstall / reinstall  BatteryAggregator via SetupHelper.

I appreciate the work you've put into this project.  Thank you!

DenkBrettl, 2024-03-09:
Same problem here:

2024-03-09 14:08:53.181993500 INFO:main:Starting...
2024-03-09 14:08:53.199560500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-03-09 14:08:53.201487500 INFO:dbusmonitor:Found: com.victronenergy.battery.c8478ce2db1f, scanning and storing items
2024-03-09 14:08:53.204079500 INFO:dbusmonitor:       com.victronenergy.battery.c8478ce2db1f has device instance 4
2024-03-09 14:08:53.243377500 INFO:dbusmonitor:Found: com.victronenergy.battery.c8478ce2d1a9, scanning and storing items
2024-03-09 14:08:53.247156500 INFO:dbusmonitor:       com.victronenergy.battery.c8478ce2d1a9 has device instance 3
2024-03-09 14:08:53.301748500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2024-03-09 14:08:54.375556500 INFO:main:Waiting for batteries (attempt 1 of 30)...
[...]

The data from dbus-serialbattery is on the dbus though (verified with dbus-spy).

pulquero, 2024-03-09:
I've manged to reproduce it, hopefully 3.0.11 will do it for you too. It was an issue around startup order of aggregator vs batteries.

DenkBrettl, 2024-03-09:
Yes, just installed 3.0.11 and this one works - thanks so much for the fast fix! Phenomenal work!

leiweke, 2024-03-09:
yes installed 3.0.11 and it works perfect. Many Thanks for the fast fix. Sun is shining , charging works again ;-)

closed 2024-03-10 by pulquero