pulquero/BatteryAggregator #28: Crash every couple of seconds with the current version (3.0.11)

thkusch, 2024-03-09:
Hello!

I'm having issues with the version 3.0.11 (installed via package manager). The BatteryAggregator crashes and restarts every couple of seconds. Here is the log file:

```
@4000000065ec750e3b9526a4 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065ec750f048b6de4 INFO:com.victronenergy.battery.aggregator:Max charge current is 2.6 but scaling back to 0.818401937046005 as limit exceeded for battery com.victronenergy.battery.ttyACM0 (overcurrent ratio: 3.1769230769230767)
@4000000065ec750f048b910c INFO:com.victronenergy.battery.aggregator:Battery currents:
@4000000065ec750f048b9cc4 com.victronenergy.battery.ttyACM0: 8.26, limit 2.6
@4000000065ec750f048ba87c com.victronenergy.battery.ttyACM1: 6.95, limit None
@4000000065ec750f048bb434 com.victronenergy.battery.ttyACM2: 8.43, limit None
@4000000065ec750f048d2b34 
@4000000065ec750f0566f51c INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065ec751029ed1dbc INFO:com.victronenergy.battery.aggregator:Max charge current is 2.2 but scaling back to 0.6807313642756682 as limit exceeded for battery com.victronenergy.battery.ttyACM0 (overcurrent ratio: 3.231818181818182)
@4000000065ec751029f4ce7c INFO:com.victronenergy.battery.aggregator:Battery currents:
@4000000065ec751029f4e204 com.victronenergy.battery.ttyACM0: 7.11, limit 2.2
@4000000065ec751029f4edbc com.victronenergy.battery.ttyACM1: 5.96, limit None
@4000000065ec751029f4f974 com.victronenergy.battery.ttyACM2: 7.6, limit None
@4000000065ec751029f50914 
@4000000065ec751129b4912c INFO:com.victronenergy.battery.aggregator:Max charge current is 2.0 but scaling back to 0.48899755501222497 as limit exceeded for battery com.victronenergy.battery.ttyACM0 (overcurrent ratio: 4.09)
@4000000065ec751129bc49bc INFO:com.victronenergy.battery.aggregator:Battery currents:
@4000000065ec751129bc612c com.victronenergy.battery.ttyACM0: 8.18, limit 2.0
@4000000065ec751129bc6ce4 com.victronenergy.battery.ttyACM1: 5.46, limit None
@4000000065ec751129bc789c com.victronenergy.battery.ttyACM2: 6.94, limit None
@4000000065ec751129bc8454 
@4000000065ec75142aa9e5dc INFO:com.victronenergy.battery.aggregator:Max charge current is 1.6 but scaling back to 0.5589519650655023 as limit exceeded for battery com.victronenergy.battery.ttyACM0 (overcurrent ratio: 2.8625)
@4000000065ec75142aaa0904 INFO:com.victronenergy.battery.aggregator:Battery currents:
@4000000065ec75142aaa14bc com.victronenergy.battery.ttyACM0: 4.58, limit 1.6
@4000000065ec75142aaa245c com.victronenergy.battery.ttyACM1: 3.63, limit None
@4000000065ec75142aaa3014 com.victronenergy.battery.ttyACM2: 5.11, limit None
@4000000065ec75142aac9d2c 
@4000000065ec75152a0d24f4 INFO:com.victronenergy.battery.aggregator:Max charge current is 2.0 but scaling back to 1.0899182561307903 as limit exceeded for battery com.victronenergy.battery.ttyACM0 (overcurrent ratio: 1.835)
@4000000065ec75152a212ddc INFO:com.victronenergy.battery.aggregator:Battery currents:
@4000000065ec75152a21454c com.victronenergy.battery.ttyACM0: 3.67, limit 2.0
@4000000065ec75152a2154ec com.victronenergy.battery.ttyACM1: 2.64, limit None
@4000000065ec75152a2160a4 com.victronenergy.battery.ttyACM2: 4.12, limit None
@4000000065ec75152a216c5c 
@4000000065ec751e04fa0e34 CRITICAL:ve_utils:exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
@4000000065ec751e04fa25a4 Traceback (most recent call last):
@4000000065ec751e04fa315c   File "/data/BatteryAggregator/ext/ve_utils.py", line 22, in exit_on_error
@4000000065ec751e04fa40fc     return func(*args, **kwargs)
@4000000065ec751e04fa4cb4   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 345, in _execute_value_changes
@4000000065ec751e04fa586c     self.valueChangedCallback(serviceName, objectPath,
@4000000065ec751e04fba08c   File "/data/BatteryAggregator/battery_service.py", line 379, in _battery_value_changed
@4000000065ec751e04fbb02c     self._update_battery_value(dbusServiceName, dbusPath, value)
@4000000065ec751e04fbbfcc   File "/data/BatteryAggregator/battery_service.py", line 393, in _update_battery_value
@4000000065ec751e04fbcf6c     self._update_battery_path(dbusServiceName, p, self.service[p])
@4000000065ec751e04fbdb24 AttributeError: 'BatteryAggregatorService' object has no attribute '_update_battery_path'
```

Anything I can try?
Best regards,
Thomas

pulquero, 2024-03-09:
I've fixed that, but I see interesting behaviour with ccl. I dont know if you can ascertain if the aggregator is reporting a max charge current <= 2, but why the batteries seem to be receiving more. I don't know if this is normal for your setup or if there is something I'm missing to take into account???

thkusch, 2024-03-09:
Hello!

Thank you for the quick response!

To your question: there is a MTTP included. It is in slave mode (controlled by my Venus OS device). "DC-coupled PV - feed in excess" is active. Maybe this is the reason for the behaviour. I didn't notice any problem with your driver in the past. The charging current on the BMS went down to zero on full battery even if the MTTP delivered plenty of power.

Best regards and thanks again!
Thomas

closed 2024-03-10 by pulquero