pulquero/BatteryAggregator #29: wrong value for InstalledCapacity

stixif, 2024-03-10:
Hi, for InstalledCapacity i become the wrong value... 
(only from one Battery) not the sum of both...

stixif, 2024-03-10:
Capacity <- is correct

pulquero, 2024-03-10:
Check you dont have capacity set in config.json

pulquero, 2024-03-10:
OK, using defaults then. Any errors in the log?

stixif, 2024-03-10:
i think no...
@4000000065ec8ffa31f6cb5c INFO:main:Starting...
@4000000065ec8ffa33996384 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec8ffa33fb4dec INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec8ffc14b43584 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec8ffd14b2f91c INFO:main:Waiting for batteries (attempt 2 of 30)...
@4000000065ec8ffe14cf993c INFO:main:Waiting for batteries (attempt 3 of 30)...
@4000000065ec8ffe14cfb494 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
@4000000065ec8fff3428e1e4 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 3
@4000000065ec9000056873ec INFO:main:Waiting for batteries (attempt 4 of 30)...
@4000000065ec900005edb14c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065ec90000ebb2fd4 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065ec90001304280c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
@4000000065ec90002635e9ec INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 1
@4000000065ec8ffa31f6cb5c INFO:main:Starting...
@4000000065ec8ffa33996384 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000065ec8ffa33fb4dec INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000065ec8ffc14b43584 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000065ec8ffd14b2f91c INFO:main:Waiting for batteries (attempt 2 of 30)...
@4000000065ec8ffe14cf993c INFO:main:Waiting for batteries (attempt 3 of 30)...
@4000000065ec8ffe14cfb494 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
@4000000065ec8fff3428e1e4 INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 3
@4000000065ec9000056873ec INFO:main:Waiting for batteries (attempt 4 of 30)...
@4000000065ec900005edb14c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000065ec90000ebb2fd4 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000065ec90001304280c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
@4000000065ec90002635e9ec INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 1

pulquero, 2024-03-11:
This should be fixed now.