pulquero/BatteryAggregator #30: sudden change in CCL / DCL when loading

Micha854, 2024-03-19:
Hello, I observe a constant sudden change in the CCL and DCL values on the aggregator when charging the batteries.
[image]
[image]
[image]

it only happens when loading! is that normal? Normally the values should remain constant.

I use the current version of the aggregator

pulquero, 2024-03-19:
Yes. So, the aggregator will adjust the aggregated CCL such that the incoming current on any one battery doesn't exceed that battery's CCL. For parallel circuits, we have 1/R = 1/R_a + 1/R_b, so I calculate the resistance of each battery and scale back appropriately (in previous versions, I assumed the resistance of each battery is the same which doesnt work so well when mixing batteries of different capacities/age/etc). It's this continued calculation of the resistance that is causing the fluctuations.

It would be worth checking battery current against CCL to verify that it is working as expected, i.e. no battery has its CCL exceeded. Are the fluctuations having any unintended consequences? I could look into adding smoothing (though this will add additional delay).

Micha854, 2024-03-19:
Thank you for the explanation! I haven't noticed any disadvantages so far. However, I saw that the aggregator specifies a DCL of approx. 150A.

my pylon has DCL 100A
The self-made battery has DCL 200A

We have consumers that draw 200A.

I'm worried that problems may arise if DCL is limited to 150A

pulquero, 2024-03-19:
What are the capacities of the two batteries? 100AH and 200AH?

Micha854, 2024-03-19:
Both storage units have a capacity of 200Ah

pulquero, 2024-03-19:
The assumption being that you can pull 100A plus from one, and 100A or less from the other to give a total of 200A. The resistance calculation on the other-hand seems to indicate that you can't pull current in the required proportion.

I would monitor current against ccl/dcl to see how accurately it tracks. If it proves to be a problem, re-open this issue and I'll look into adding an alternative calculation based on relative battery capacities (equal => 50/50 current split) which might work better for you.

closed 2024-03-19 by pulquero