pulquero/BatteryAggregator #31: BatteryAggregator not running with Venus version 3.30

Dropdragger, 2024-03-21:
I have updated my Cerbo to 3.30 yesterday and noticed that the BatteryAggregator disappeared in the device menu of the console. Therefore I listed the corresponding log file and found that there is a problem with imported module "vedbus'. The logfile show the following message:

**2024-03-21 16:19:39.351043500 Traceback (most recent call last):
2024-03-21 16:19:39.351051500   File "/data/BatteryAggregator/battery_service.py", line 12, in <module>
2024-03-21 16:19:39.351054500     from vedbus import VeDbusService
2024-03-21 16:19:39.351057500 ModuleNotFoundError: No module named 'vedbus'**

Thereupon I switched back to 3.20 and the BatteryAggregator was working again. It seems that Victron had mage changes in the library.

pulquero, 2024-03-21:
Sometimes that happens, you just need to install it again. I think there is some sort of freak bug in the underlying package manager that it sometimes starts BatteryAggregator before it has finished unpacking all the files.

Dropdragger, 2024-03-21:
BTW it was the latest version 3.22 which I had run and noticed the error.
But you are right. - After switching back from 3.20 to 3.22 again. The plugin works again without re-installation.
Nevertheless, good to know that this may happen.
BR

closed 2024-03-21 by Dropdragger