pulquero/BatteryAggregator #33: How ti exclude shunt from aggregation, but use the SOC of the shunt

seamaster101, 2024-03-25:
@pulquero I did read the directions in read.me file and ended up with this:
/data/setupOptions/BatteryAggregator/config.json
{
"excludedServices":["com.victronenergy.battery.ttyUSB1"]
"primaryServices": {"com.victronenergy.battery.ttyUSB1": ["/SoC"]}
}

but i'm getting the average ion the 4 batteries SOC rather the SOC from the shunt as I presume it was already excluded from reading by the first line in the config.json. 
Any hints?

pulquero, 2024-03-25:
Remove the exludedServices line, it excludes it for *all* purposes. The fact that it's a primary service means it wont get aggregated.

seamaster101, 2024-03-26:
Unfortunately that did nit quite work as expected:
[image]

As you can see the current for the aggregator is showing as 0A and it should be the sum of B1 to B4. the SOC is 60% and should be the same as the Shunt at 63%

seamaster101, 2024-03-26:
I looked at it again and established that the shunt after reboot changed to USB3. this is little disturbing: does it mean that after everey reboot we have the chance these USB re-enumeration to change randomly?  

{
"primaryServices": {"com.victronenergy.battery.ttyUSB3": ["/SoC"]}
}
after editing the config to show above and restarting the service I got this:
[image]

I think it is using the shunt for the aggregation...

seamaster101, 2024-03-26:
> My understanding is the USB enumeration is based on the ports so should be stable. If the device instance ID is stable, I could look at adding support to use that as an alternative to the service name.
> 
> Look in dbus-spy, check the aggregator /System/Batteries path. It should have an array of the batteries it is aggregating over.

with the entry in my config:
[image]
without it (empty congig) it shows this:
[image]

I hope that helps to figure it out

pulquero, 2024-03-26:
That would suggest a config error/bug. Anything in the log?

seamaster101, 2024-03-26:
> That would suggest a config error/bug. Anything in the log?

How do I check the log?

pulquero, 2024-03-26:
cat /var/log/BatteryAggregator/current

seamaster101, 2024-03-26:
Here is the log when i have this entry in the config.json
{
"primaryServices": {"com.victronenergy.battery.ttyUSB1": ["/SoC"]}
}

`@400000006602fcad07fe16d4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB6 has device instance 8
@400000006602fcad0b14633c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB7, scanning and storing items
@400000006602fcad0b50fedc INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB7 has device instance 5
@400000006602fcad0e70f3ec INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB8, scanning and storing items
@400000006602fcad1093acdc INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB8 has device instance 6
@400000006602fcad149059d4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@400000006602fcad14df875c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 290
@400000006602fcad18f2601c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@400000006602fcae14b8ec8c INFO:main:Waiting for batteries (attempt 1 of 30)...
@400000006602fcae1545ee34 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@400000006602fcae1b63872c Traceback (most recent call last):
@400000006602fcae1b63a66c   File "/data/BatteryAggregator/battery_service.py", line 625, in wait_for_batteries
@400000006602fcae1b63b9f4     batteryAggr.register(timeout=15)
@400000006602fcae1b63c5ac   File "/data/BatteryAggregator/battery_service.py", line 338, in register
@400000006602fcae1b63d54c     self._refresh_values(paths_changed)
@400000006602fcae1b63e104   File "/data/BatteryAggregator/battery_service.py", line 365, in _refresh_values
@400000006602fcae1b63f48c     self._refresh_value(path)
@400000006602fcae1b65bd94   File "/data/BatteryAggregator/battery_service.py", line 353, in _refresh_value
@400000006602fcae1b65d504     v = self._get_value(dbusPath)
@400000006602fcae1b65dcd4   File "/data/BatteryAggregator/battery_service.py", line 348, in _get_value
@400000006602fcae1b65f05c     return self.service[dbusPath]
@400000006602fcae1b65f82c   File "/data/BatteryAggregator/ext/vedbus.py", line 161, in __getitem__
@400000006602fcae1b660bb4     return self._dbusobjects[path].local_get_value()
@400000006602fcae1b66752c KeyError: '/SoC'
root@raspberrypi2:~#`

seamaster101, 2024-03-26:
OK. I figured out one problem:
Victron shunt is not /SoC and it it is /Soc.  
So the proper entry for me is:
{
"primaryServices": {"com.victronenergy.battery.ttyUSB1": ["/Soc"]}
}
the other issue still exists:
[image]

it is aggregating the shunt regardless of the entry of "primaryServices" in the config.json

here is the log with the new entry in config.json

@400000006602fe4b26b4f174 INFO:main:Starting...
@400000006602fe4b2af0c27c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@400000006602fe4b2b240b64 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB5, scanning and storing items
@400000006602fe4b2b6ed3ec INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB5 has device instance 7
@400000006602fe4b2e7ac3fc INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB6, scanning and storing items
@400000006602fe4b2eb2d38c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB6 has device instance 8
@400000006602fe4b35295e0c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB7, scanning and storing items
@400000006602fe4b3563a404 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB7 has device instance 5
@400000006602fe4b38b73404 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB8, scanning and storing items
@400000006602fe4b38f85bb4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB8 has device instance 6
@400000006602fe4c00d40b34 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@400000006602fe4c010b47d4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 290
@400000006602fe4c0536ffc4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@400000006602fe4d14defea4 INFO:main:Waiting for batteries (attempt 1 of 30)...
@400000006602fe4d1564304c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@400000006602fe4d1d1ce374 Traceback (most recent call last):
@400000006602fe4d1d1cfecc   File "/data/BatteryAggregator/battery_service.py", line 625, in wait_for_batteries
@400000006602fe4d1d1d1254     batteryAggr.register(timeout=15)
@400000006602fe4d1d1d1e0c   File "/data/BatteryAggregator/battery_service.py", line 338, in register
@400000006602fe4d1d1d3194     self._refresh_values(paths_changed)
@400000006602fe4d1d1d3964   File "/data/BatteryAggregator/battery_service.py", line 365, in _refresh_values
@400000006602fe4d1d1d4cec     self._refresh_value(path)
@400000006602fe4d1d20fe3c   File "/data/BatteryAggregator/battery_service.py", line 353, in _refresh_value
@400000006602fe4d1d217f24     v = self._get_value(dbusPath)
@400000006602fe4d1d218adc   File "/data/BatteryAggregator/battery_service.py", line 348, in _get_value
@400000006602fe4d1d219a7c     return self.service[dbusPath]
@400000006602fe4d1d21a634   File "/data/BatteryAggregator/ext/vedbus.py", line 161, in __getitem__
@400000006602fe4d1d21b5d4     return self._dbusobjects[path].local_get_value()
@400000006602fe4d1d2236bc KeyError: 'SoC'
@400000006602fee6256dbbd4 INFO:main:Exit
@400000006602feeb099a177c INFO:main:Starting...
@400000006602feeb0bef5dd4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@400000006602feeb0d6d1c64 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB5, scanning and storing items
@400000006602feeb0dc477ac INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB5 has device instance 7
@400000006602feeb111b5b3c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB6, scanning and storing items
@400000006602feeb11638bb4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB6 has device instance 8
@400000006602feeb14cf146c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB7, scanning and storing items
@400000006602feeb1d5ac34c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB7 has device instance 5
@400000006602feeb21b7188c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB8, scanning and storing items
@400000006602feeb21f8e834 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB8 has device instance 6
@400000006602feeb271ab824 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@400000006602feeb28d36c74 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 290
@400000006602feeb2fbb71bc INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@400000006602feed14bd9fac INFO:main:Waiting for batteries (attempt 1 of 30)...
@400000006602feed15560f1c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@400000006602feed1d2f5a04 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator

seamaster101, 2024-03-27:
Perfect! I installed the new version and appears that the issue is solved! It works very well now. 

Thank you for your help with the issue and thank you for the battery aggregator in general!

closed 2024-03-27 by pulquero