pulquero/BatteryAggregator #34: One battery goes to protection, closing charge and agregator sets CCL to 0a on all batteries.

AlexdeKairos, 2024-04-01:
I'm testing new functionality on dbus-serialbattery. This functionality sets CCL to 0a when the BMS closes the Charging FETs (manually or automatically by the BMS). In this scenario I have found  that BatteryAggregator also sets CCL to 0a on ALL aggregated batteries and thus, charging stops in all batteries.  

Same happens is the BMS closes the discharge to protect the battery,  and Batteryagregator sets DCL to 0a on all batteries.

pulquero, 2024-04-01:
Current behaviour is because the batteries are in parallel, BatteryAggregator has to set CCL to 0 to ensure no current reaches the battery with 0 CCL. What would be good is to have some way for BatteryAggregator to know the state of charging fets so it knows it is effectively safe to ignore the CCL. Do you know if there is some dbus path I can check to ascertain the state?

pulquero, 2024-04-01:
Maybe check if any of these give an indication:

    /Io/AllowToCharge
    /System/NrOfModulesBlockingCharge
    /System/NrOfModulesBlockingDischarge
    /System/NrOfModulesOnline
    /System/NrOfModulesOffline

AlexdeKairos, 2024-04-01:
Yes. In IO menu of every aggregated battery Allow to charge and Allow to discharge will show "Yes" when fets are online (so batteries Allow to charge or discharge)  and show "No" when its not allowed to charge or discharge...

AlexdeKairos, 2024-04-01:
Tomorrow ill go to my installation and perform a dbus-spy to see values.

AlexdeKairos, 2024-04-01:
Looks to be this one
/Io/AllowToCharge

Can be Yes or No. Anyhow,  tomorrow ill have a look

AlexdeKairos, 2024-04-02:
Yes, I can confirm that this is the option where the BMS notifies that charging is not allowed

pulquero, 2024-04-03:
Released a new version that takes into account Allow status.

AlexdeKairos, 2024-04-03:
All working properly.  Next week ill do some advance testing with auto protections activation to see behaviour.

closed 2024-04-03 by pulquero