pulquero/BatteryAggregator #35: Since 3.0.18: ZeroDivisionError: float division by zero

DenkBrettl, 2024-04-03:
Hi there,

since 3.0.18 I have the following trace happening on startup:

```
2024-04-03 17:12:15.837288500 INFO:main:Starting...
2024-04-03 17:12:15.854473500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-04-03 17:12:15.856724500 INFO:dbusmonitor:Found: com.victronenergy.battery.c8478ce2db1f, scanning and storing items
2024-04-03 17:12:15.861574500 INFO:dbusmonitor:       com.victronenergy.battery.c8478ce2db1f has device instance 4
2024-04-03 17:12:15.954821500 INFO:dbusmonitor:Found: com.victronenergy.battery.c8478ce2d1a9, scanning and storing items
2024-04-03 17:12:15.959360500 INFO:dbusmonitor:       com.victronenergy.battery.c8478ce2d1a9 has device instance 3
2024-04-03 17:12:16.003356500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2024-04-03 17:12:17.438945500 INFO:main:Waiting for batteries (attempt 1 of 30)...
2024-04-03 17:12:17.445017500 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
2024-04-03 17:12:17.485994500 Traceback (most recent call last):
2024-04-03 17:12:17.485999500   File "/data/BatteryAggregator/battery_service.py", line 652, in wait_for_batteries
2024-04-03 17:12:17.486003500     batteryAggr.register(timeout=15)
2024-04-03 17:12:17.486006500   File "/data/BatteryAggregator/battery_service.py", line 345, in register
2024-04-03 17:12:17.486010500     self._refresh_values(paths_changed)
2024-04-03 17:12:17.486013500   File "/data/BatteryAggregator/battery_service.py", line 372, in _refresh_values
2024-04-03 17:12:17.486016500     self._refresh_value(path)
2024-04-03 17:12:17.486086500   File "/data/BatteryAggregator/battery_service.py", line 368, in _refresh_value
2024-04-03 17:12:17.486091500     self._update_active_values(dbusPath)
2024-04-03 17:12:17.486093500   File "/data/BatteryAggregator/battery_service.py", line 440, in _update_active_values
2024-04-03 17:12:17.486097500     defn.action(self)
2024-04-03 17:12:17.486099500   File "/data/BatteryAggregator/battery_service.py", line 194, in <lambda>
2024-04-03 17:12:17.486103500     '/Info/MaxChargeCurrent': PathDefinition(CURRENT, aggregatorClass=NullAggregator, triggerPaths={'/Info/MaxChargeCurrent', '/Dc/0/Current', '/Dc/0/Voltage', '/Io/AllowToCharge'}, action=lambda api: api._updateCCL()),
2024-04-03 17:12:17.486133500   File "/data/BatteryAggregator/battery_service.py", line 455, in _updateCCL
2024-04-03 17:12:17.486137500     total_ir = 1.0/sum([1.0/ir for ir in irs.values()]) if irs else 0
2024-04-03 17:12:17.486199500 ZeroDivisionError: float division by zero
````

Unfortunately, I don't have the time tonight to debug this further - I might find the time on Friday or the weekend.

DenkBrettl, 2024-04-03:
I'd love to downgrade, but I'm not sure how - if there was a v3017 tag for example, I could just ask SetupHelper to install that one instead of "latest".

DenkBrettl, 2024-04-03:
I tried inserting debug code, but when I killed battery_aggregator.py, a new error appeared which was ModuleNotFoundError: No module named 'vedbus'
I then reinstalled 3.0.18 through SetupHelper which also installed vedbus. Then upon restart two more ZeroDivisionError messages appeared and now it seems to be running.
Seems like some race condition to me. I can't spend more time on this tonight as said, maybe I can have a look tomorrow.

pulquero, 2024-04-03:
> I'd love to downgrade, but I'm not sure how - if there was a v3017 tag for example, I could just ask SetupHelper to install that one instead of "latest".

Tag created.

pulquero, 2024-04-03:
I've spotted a potential issue and pushed a fix for it. Not sure if it is the cause of your issue.

DenkBrettl, 2024-04-04:
Thanks for the quick response and fix! This installed over night, as expected, and is running stable since then, without any traces. I'll close this for now and re-open should there be issues again.

closed 2024-04-04 by DenkBrettl