pulquero/BatteryAggregator #36: After limitation or CCL or DCL (new feature), CVL goes also to 0V. after resuming normal condition, CVL remains 0V

AlexdeKairos, 2024-04-04:
Testing the new feature of checking FETs state to decide if CCL or DCL must be limited, I found that upon application of both CCL and DCL to 0, CVL goes ALSO to 0v. I think that's a mistake and should not happen. All chargers go to 0V and undervoltage alarm triggers. But after resuming normal condition, the CVL remains 0 even if CCL and DCL goes back to normal. So no charging happens.

AlexdeKairos, 2024-04-04:
[image]
[image]

pulquero, 2024-04-04:
Pushed fix.

AlexdeKairos, 2024-04-04:
Seems to be fine on v3.0.20

closed 2024-04-04 by pulquero