pulquero/BatteryAggregator #37: CVL drops and spikes, CCL and DCL swings

leiweke, 2024-04-04:
Hello yesterday I got a Battery low alarm from my Victron (wasn´t at home) with SOC 80%. Today I checked the VRM and found that BatteryAggregator is dropping CVL at that time, even the 2 Batteries are at 89% and 66% SOC, please see the picture. 
I also realised that Batteries BMS and Serial Battery have constant CVL and Battery Aggregators CVL is sometimes dropping, the same with CCL and DCL.
Setup helper was on auto update so I don´t know exactly which version, as there is nor tagging of the version, I assume it was 3.0.18 or 19. no config file, two 16S JK BMS with Serial Battery and USB connection.
I went back to 3.0.17 today as this was the only tagged version, but looks the same.
Any ideas?
please use version tagging in the future as it makes it easier to compare versions.
Thanks a lot
Sebastian

leiweke, 2024-04-04:
pictures from VRM from today
[image]

pulquero, 2024-04-04:
Try 3.0.20, see #36.

leiweke, 2024-04-04:
Thanks I just installed 3.0.20. 
looks a bit better, but CVL and DCL still jumping each second , but with a small aptitude.
see pictures.
Why is this? as the BMS parameters are stable and not changing.
[image]

[image]

pulquero, 2024-04-04:
CCL/DCL jumping: this is explained in #30. In short, I am calculating the internal resistances of the batteries to accurately limit the overall current such that the proportions for each battery dont exceed their individual CCL/DCL.

Does the variability cause issues for you?

leiweke, 2024-04-05:
Many Thanks for the explanation, makes it more clear.
please help me how to setup the right CCL/DCL.
I have 2 x Blocks of 16x280Ah, one Victron MP2 that can charge with 70A.
Serial Battery config is set to 80A.
if I check the individual Batterie BMS´s in Victron, I see each reporting 80A
Batterie Aggregator makes it to 160A and then recalculation ongoing.
But I don´t understand why the changes/variability is so high.
Please look at the screen shots, tonight at 0:00am
[image]

and at 2:12am :
[image]

Ampere Load is constant at around 7 Amp. Voltage at 0:00am is 53,05V and at 2:12am is 52,90V so a minimal drop.
But BatteryAgg. is reporting CCL/DCL down until 2:12 and then (no change on load) up until 7:46 (Coffeemaker on)
I don´t understand why on constant night load around 7-8Amp CCL/DCL is changing so high.
see the pic:
[image]

pulquero, 2024-04-05:
I'm currently in the process of revamping how this part of the code works. It's a much better approach using a mixture of techniques, which should control the variability much better. I'm testing it at the moment, hopefully in a day or two, I'll make it available pre-release for you to test it out too.

leiweke, 2024-04-06:
Great thanks for the effort, maybe also the recalculation only needs to be once every minute or so. 
please use Tags on the software versions. makes is easier to test different software versions  and compare.
Many Thanks

leiweke, 2024-04-06:
picture from todays night, looks good.
[image]

pulquero, 2024-04-07:
If you set tag to 'main', please try v3.0.33. Instead of doing the calculation on-demand, it is now done ahead of time, using a sliding window. The values are also published at /System/InternalResistances. For batteries of equal age and capacity, the values should be approx equal. Note, it may take some time for the samples to build up before values appear.

leiweke, 2024-04-09:
thanks updated to main 3.0.33 and checked /System/InternalResistances, shows: 
{"value":"[null, 0.009443896834824132]"}

I have two packs, why is the first value "null" ? and what says the value 0.009443896834824132?

pulquero, 2024-04-09:
It's null because its yet to build up sufficient samples to determine a value for the first battery. 0.009 Ohms is the estimated internal resistance of your second battery. Generally, as batteries age, the internal resistance increases. As a rule of thumb, batteries in parallel should have approx the same internal resistance. The internal resistance also varies with SoC and temperature.

leiweke, 2024-04-09:
okay now the first one is also there:
{"value":"[0.0245854010614145, 0.017369320182247922]"}
first one is 1 year older then the second block

pulquero, 2024-04-09:
Nice, that makes sense then. Is the CCL/DCL much more stable now, too?

leiweke, 2024-04-09:
will publish a pic in the evening.
But I have an open point. 
I have one Victron MP2 that can deliver 70A Charge for this two packs.
Serial Battery is configured to 80A Charge and BMS also.
BattAgg. makes 160 A Charge limit. This makes no sense. if batteries are at 90+ % its still pushing 70A into the Batteries.
Any ideas?

pulquero, 2024-04-09:
Check the voltages, if the voltage is within limits, I assume it is fine. Voltage limit should force the current to tail off. At some point, the voltage is going to have to rise, or the current drop off.

pulquero, 2024-04-09:
I assume the CCLs for the individual batteries are changing too? The internal resistance is not updated during charging so the changes shouldnt be due to anything the aggregator is doing.

leiweke, 2024-04-09:
as said I use Serial-Battery with RS485 and has set 80Amp in the config.ini file. so it BattAgg doubles it.
how to avoid this CCL and DCL doubling by BattAgg?

Internal Resistance update:
{"value":"[null, 1.3944688539531036]"}

Thanks

pulquero, 2024-04-09:
The doubling sounds like it is because BattAgg is aggregating over two batteries but the batteries are configured with the combined total CCL. Is Serial-Battery config meant to be for each individual battery or the combined total? I imagine it should be set to 40A so the individual batteries report a CCL of 40A and BattAgg reports 80A.

leiweke, 2024-04-09:
I tested it and set it to 40Amp Charge and 100Amp Discharge:
@400000006615940d3445188c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006615940d344b2b3c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006615940d345047d4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A (read from BMS)

now BattAgg use the combined values, but if I shut down one Batt for e.g. service than the other only get 40Amps max even if the MP2 can deliver 70Amps. not good :-(

pulquero, 2024-04-09:
Sorry, I'm getting a bit lost. What is the rated charge current for one battery? If it is 80A, then the combined rating will be 160A. If it is 40A, then the combined rating is 80A. The doubling is correct in either case. I don't think you mean to charge a 40A battery at 70A???

leiweke, 2024-04-10:
sorry for the confusion, some facts what I have:
1 x Victron MP2 5kw can deliver 70A charge 
2 x 16S 280Ah with JK BMS with max 150A charge/discharge connected with RS485
Victron OS 3.30 on RPI4, RS485 connection to JK BMS´s
SerialBattery to connect the BMS´s to Victron https://github.com/Louisvdw/dbus-serialbattery

pulquero, 2024-04-10:
New version has dynamic logging that you can switch on/off by setting /LogLevel to DEBUG. See if this helps to work out what is going on.

closed 2024-04-28 by pulquero