pulquero/BatteryAggregator #38: Lost batteries and shutdown

Albarge, 2024-04-09:
Hi, I just updated today from v3.0.15 to v3.0.33 and maybe an hour later I noticed on VRM battery Aggregator was showing 0v and a few seconds later the Quattro shutdown, then came back on again. Looks like it couldn't find the batteries for some reason, but what could cause this?

Here's my log:

@4000000066156e780d9d34e4 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000066156e780f017bec INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000066156e7819738e6c INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@4000000066156eb403ba4ad4 INFO:main:Exit
@4000000066156ed80e3d3da4 *** CCGX booted (0) ***
@4000000066156ee518e4fe54 INFO:main:Starting...
@4000000066156ee522004ad4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000066156ee5227868d4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000066156ee626d74e54 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000066156ee726c42be4 INFO:main:Waiting for batteries (attempt 2 of 30)...
@4000000066156ee826ca65a4 INFO:main:Waiting for batteries (attempt 3 of 30)...
@4000000066156ee926da5f7c INFO:main:Waiting for batteries (attempt 4 of 30)...
@4000000066156eea114257b4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
@4000000066156eea126f239c INFO:dbusmonitor:       com.victronenergy.battery.ttyS7 has device instance 288
@4000000066156eea2c275e1c INFO:main:Waiting for batteries (attempt 5 of 30)...
@4000000066156eeb26c9d134 INFO:main:Waiting for batteries (attempt 6 of 30)...
@4000000066156eec26c46e4c INFO:main:Waiting for batteries (attempt 7 of 30)...
@4000000066156eed26bd2aec INFO:main:Waiting for batteries (attempt 8 of 30)...
@4000000066156eee26c76034 INFO:main:Waiting for batteries (attempt 9 of 30)...
@4000000066156eef26bfde54 INFO:main:Waiting for batteries (attempt 10 of 30)...
@4000000066156ef026c30304 INFO:main:Waiting for batteries (attempt 11 of 30)...
@4000000066156ef126cf8a0c INFO:main:Waiting for batteries (attempt 12 of 30)...
@4000000066156ef226bc1594 INFO:main:Waiting for batteries (attempt 13 of 30)...
@4000000066156ef326c3ed64 INFO:main:Waiting for batteries (attempt 14 of 30)...
@4000000066156ef4271132a4 INFO:main:Waiting for batteries (attempt 15 of 30)...
@4000000066156ef526bd2aec INFO:main:Waiting for batteries (attempt 16 of 30)...
@4000000066156ef626bb2b34 INFO:main:Waiting for batteries (attempt 17 of 30)...
@4000000066156ef726ce3e04 INFO:main:Waiting for batteries (attempt 18 of 30)...
@4000000066156ef826c2a15c INFO:main:Waiting for batteries (attempt 19 of 30)...
@4000000066156ef926bbf654 INFO:main:Waiting for batteries (attempt 20 of 30)...
@4000000066156efa26ca88cc INFO:main:Waiting for batteries (attempt 21 of 30)...
@4000000066156efb26c13de4 INFO:main:Waiting for batteries (attempt 22 of 30)...
@4000000066156efc26be5b9c INFO:main:Waiting for batteries (attempt 23 of 30)...
@4000000066156efd26bcc55c INFO:main:Waiting for batteries (attempt 24 of 30)...
@4000000066156efe26bd231c INFO:main:Waiting for batteries (attempt 25 of 30)...
@4000000066156f001bed9ad4 INFO:main:Waiting for batteries (attempt 26 of 30)...
@4000000066156f011be9030c INFO:main:Waiting for batteries (attempt 27 of 30)...
@4000000066156f021beab0bc INFO:main:Waiting for batteries (attempt 28 of 30)...
@4000000066156f031be65b5c INFO:main:Waiting for batteries (attempt 29 of 30)...
@4000000066156f041bfe7b24 INFO:main:Waiting for batteries (attempt 30 of 30)...
@4000000066156f041bfea234 WARNING:main:No batteries discovered!
@4000000066156f041c063f6c INFO:main:Exit
@4000000066156f0a35b6cd14 INFO:main:Starting...
@4000000066156f0b00b73464 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000066156f0b02378b04 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@4000000066156f1634e74044 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
@4000000066156f1705c8d3cc INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
@4000000066156f170672c3b4 INFO:dbusmonitor:       com.victronenergy.battery.ttyS7 has device instance 288
@4000000066156f170ffcf6d4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@4000000066156f1710abefcc INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 3
@4000000066156f191fadfe34 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@4000000066156f192b418e14 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
@4000000066156f19332135e4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000066156f1b1be17d44 INFO:main:Waiting for batteries (attempt 1 of 30)...
@4000000066156f1b205aad3c INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@4000000066156f1b321718e4 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
@40000000661584cb1919319c INFO:dbusmonitor:com.victronenergy.battery.ttyUSB2 disappeared from the dbus. Removing it from our lists
@40000000661584cc1e064974 INFO:dbusmonitor:com.victronenergy.battery.ttyUSB1 disappeared from the dbus. Removing it from our lists
@40000000661584cc348e27ac INFO:dbusmonitor:com.victronenergy.battery.ttyUSB0 disappeared from the dbus. Removing it from our lists
@40000000661584cd0a312374 INFO:dbusmonitor:com.victronenergy.battery.ttyS7 disappeared from the dbus. Removing it from our lists
@40000000661584da1c78893c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
@40000000661584da1febaf2c INFO:dbusmonitor:       com.victronenergy.battery.ttyS7 has device instance 288
@400000006615853402f3beb4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@400000006615853a1278eb84 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 3
@400000006615853b003ac4ec INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
@400000006615854138d8d17c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
@40000000661585420fc842a4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
@40000000661585440aef5eac INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1

Albarge, 2024-04-09:
and VRM when it lost the BMS's:

[image]

pulquero, 2024-04-09:
To me, it looks like the individual batteries crashed. Check them for any logs or alarms. Check charts for the individual batteries, see if there are any clues. I dont know if a possible cause could be a spike in cpu load.

Albarge, 2024-04-09:
Thanks, I use dbus-serialbattery for the three batteries, I have had problems with different versions, but it's been stable for the last few weeks until the aggregator update today, maybe a coincidence though. I'll look at the serialbattery logs

ondrom111, 2024-04-09:
it says low battery for me
[image]

pulquero, 2024-04-18:
I'm closing this for now, re-open if it happens again.

closed 2024-04-18 by pulquero

Albarge, 2024-04-18:
Thanks, yeah I ended up doing a factory reset, as it seems the Cerbo just isn't powerful enough to run GuiMods, SerialBattery x 3, BatteryAggregator and all my other connections, the fresh reset helped a little with high CPU, and I haven't had another shutdown since, but `uptime` is still consistently above 2 which is very laggy on the 2 core Cerbo.