pulquero/BatteryAggregator #39: low battery

ondrom111, 2024-04-10:
it says low battery for me. I have 2 batteries, one has 96 percent, the other 100 percent.
[image]
[image]

pulquero, 2024-04-10:
The low voltage alarm on the aggregator is triggered if it is triggered on any individual battery. Please check the state of the individual batteries.

ondrom111, 2024-04-10:
all are ok

pulquero, 2024-04-10:
I've released a new version that has dynamic logging. In dbus-spy, set /LogLevel to DEBUG. The next time it happens please attach the log to this issue.

pulquero, 2024-04-19:
Is the problem still present in the latest version?

closed 2024-04-28 by pulquero