pulquero/BatteryAggregator #40: CCL = 0 and DCL =0

Rubens974, 2024-04-10:
Hi,
Can you help please?
I have 2 batteries in parallel
I've just upgraded from version 3.0.17 to 3.0.43 but the CCL and DCL remain at 0.
[image]
[image]

Great work thank you
Richard

pulquero, 2024-04-10:
Hi, I've added more detailed logging for CCL/DCL in v3.0.44. Set /LogLevel to DEBUG, attach contents of  /var/log/BatteryAggregator/current. Thanks.

Rubens974, 2024-04-10:
Thank you so much.
Here is the logfile
[current.txt](https://github.com/pulquero/BatteryAggregator/files/14934956/current.txt)

pulquero, 2024-04-10:
Fixed the error in the log.

Rubens974, 2024-04-10:
TypeError: 'Logger' object is not callable
How do I solve this error?

Rubens974, 2024-04-10:
CRITICAL
ZeroDivisionError: float division by zero.
[current.TXT](https://github.com/pulquero/BatteryAggregator/files/14937138/current.TXT)

pulquero, 2024-04-10:
Fixed

Rubens974, 2024-04-11:
Morning,
Here is my log file: no more error at this time. Thank you
[current0.txt](https://github.com/pulquero/BatteryAggregator/files/14941413/current0.txt)
[current1.txt](https://github.com/pulquero/BatteryAggregator/files/14941416/current1.txt)

Unfortunately I still have the DCL and CCL issue. it's impossible to use the batteries (charging or discharging).

Thanks

Rubens974, 2024-04-11:
At this time I've downgraded to version 3.0.17 and it works like a charm.

pulquero, 2024-04-11:
Hmm, it might be an initialisation ordering problem. The old version updates the CCL more frequently, so I'm guessing if it wasnt available at startup, it would quickly get picked up afterwards. When you next get a chance, if you can add "logLevel": "DEBUG" to the config file and we'll be able to see what happens during startup.

pulquero, 2024-04-11:
Unfortunately, that doesn't include full debug info since startup. Debug was turned on some time later.

Rubens974, 2024-04-11:
"logLevel": "DEBUG" is configured in dbus-spy after batteryaggregator is started. How can I proceed differently?

Rubens974, 2024-04-11:
Here are some logs after a restart

[@40000000661808f02dd94ff4.s.txt](https://github.com/pulquero/BatteryAggregator/files/14948487/%4040000000661808f02dd94ff4.s.txt)
[@400000006618092a2e02d4dc.s.txt](https://github.com/pulquero/BatteryAggregator/files/14948488/%40400000006618092a2e02d4dc.s.txt)
[@400000006618095e2e26ef0c.s.txt](https://github.com/pulquero/BatteryAggregator/files/14948491/%40400000006618095e2e26ef0c.s.txt)
[current.txt](https://github.com/pulquero/BatteryAggregator/files/14948492/current.txt)

Rubens974, 2024-04-11:
logs of v3.0.47
[@4000000066181d5a2f3afeec.s.txt]
[@4000000066181cd02f350794.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949365/%404000000066181cd02f350794.s.txt)
[@4000000066181ca92df334dc.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949366/%404000000066181ca92df334dc.s.txt)
[@4000000066181c712df2c77c.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949367/%404000000066181c712df2c77c.s.txt)
[@4000000066181c312e14ce6c.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949368/%404000000066181c312e14ce6c.s.txt)
[@4000000066181bfb2e3b74f4.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949369/%404000000066181bfb2e3b74f4.s.txt)
[@4000000066181bc52df1e4ec.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949372/%404000000066181bc52df1e4ec.s.txt)
[@4000000066181b972e3ae854.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949373/%404000000066181b972e3ae854.s.txt)
[@4000000066181b632f228164.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949375/%404000000066181b632f228164.s.txt)
[@4000000066181b2f2dc4e94c.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949376/%404000000066181b2f2dc4e94c.s.txt)
[@4000000066181afd2e106584.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949377/%404000000066181afd2e106584.s.txt)
(https://github.com/pulquero/BatteryAggregator/files/14949360/%404000000066181d5a2f3afeec.s.txt)
[@4000000066181d062e15eb94.s.txt](https://github.com/pulquero/BatteryAggregator/files/14949362/%404000000066181d062e15eb94.s.txt)

pulquero, 2024-04-11:
> "logLevel": "DEBUG" is configured in dbus-spy after batteryaggregator is started. How can I proceed differently?

You need to config it in the config file, changing it in dbus-spy is too late.
```
{"logLevel": "DEBUG"}

Rubens974, 2024-04-12:
Okay thank you.

[@400000006618ac992e3d8c1c.txt](https://github.com/pulquero/BatteryAggregator/files/14953661/%40400000006618ac992e3d8c1c.txt)
[@400000006618acca394ce83c.txt](https://github.com/pulquero/BatteryAggregator/files/14953662/%40400000006618acca394ce83c.txt)
[@400000006618acf63a480564.txt](https://github.com/pulquero/BatteryAggregator/files/14953663/%40400000006618acf63a480564.txt)
[current.txt](https://github.com/pulquero/BatteryAggregator/files/14953675/current.txt)

pulquero, 2024-04-12:
Looks like the issue is that the absence of /Io/AllowToCharge is interpreted as not allowed to charge. I need to distinguish between 0 and nothing. This was related to a recent change to enable proper support for AllowToCharge.
```
@400000006618ac99272f2a84 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Io/AllowToCharge updated with {com.victronenergy.battery.socketcan_can9: None} now has values {'com.victronenergy.battery.socketcan_can8': None, 'com.victronenergy.battery.socketcan_can9': None} with result None
```
Fixed.

Rubens974, 2024-04-12:
Great
You are the best
So none is not 0!
This has solved the problem, topic can be closed now

Thank you so much.

pulquero, 2024-04-12:
Pleasure.

closed 2024-04-12 by pulquero