pulquero/BatteryAggregator #41: Feature request. Provide Battery information to victron CAN bus for other devices

AlexdeKairos, 2024-04-11:
Hi!

I have a feature request.  
I have several devices connected via CANBUS to victron cerbo. Some of them are not directly compatible with victron but read CAN BUS info provided. In my case 2 alternators. If battery information is published in CAN, the alternators can read and adapt their charging profile to the battery needs. I use your BatteryAgregator to mix 2 Dbus-serialbattery info for cerbo. It works great and with latest improvements it's becoming extremely useful. 
Do you think it would be possible to use Battery Aggregator instead of publish the batt info in dbus, do it in CAN bus or both at the same time? 
Many devices like inverters can read CAN directly and even if not compatible with victron., connected to CAN ports will allow them to adapt to battery needs.

CVL, CCL, Batt Temperaure, DVL, DCL for instance.

All the best
Alex

pulquero, 2024-04-11:
Probably possible, unfortunately I don't have CAN bus myself to do testing. If you provide details about your CAN devices, maybe I can see if there is any related example code I can take a look at.

AlexdeKairos, 2024-04-11:
I'll have a look to do a CAN DUMP I'd that could help. Unfortunately I'm just a user with ideas! 
I'll look to find battery related CAN BUS messages.

AlexdeKairos, 2024-04-15:
Hi! I have been looking around but no idea how to help you. Dbus-spy shows all can connected devices and some link info regarding batteries.  No current but CVL is there already. But no idea how to help. Victron Can is based on nmea2000 sentences a can bus network used in marine world.