pulquero/BatteryAggregator #43: Setting battery capacity issue

campervantime, 2024-04-14:
Hi

I am using your BatteryAggregator to give me a single battery monitor view from 2x smartshunts that I have. One is installed on a 70ah AGM and the other on a 92ah AGM and both are working correctly.

I installed BatteryAggregator and did not create a config.json, and everything appeared to be working apart from Aggregator didn't know what battery capacity I had installed, it just appears as 0 when I look in the setup menu for Aggregator. Time remaining, SOC, current etc all seem to be operating though.

I then tried to create a config.json (see below) but I got errors (see below) and the battery capacity still wasn't set. 

Is setting the capacity really needed in my application or is it only useful where there is no source for time remaining from the BMSs/Shunts?

Thanks

Config.json.........
{
	"capacity": 162
	"primaryServices": {"com.victronenergy.battery.ttyS5": ["/Soc"]}
}

my error log

@40000000661aa78133b8d64c INFO:main:Starting...
@40000000661aa78135d97044 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@40000000661aa7813637ad44 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS5, scanning and storing items
@40000000661aa781367fe1a4 INFO:dbusmonitor:       com.victronenergy.battery.ttyS5 has device instance 277
@40000000661aa7820063059c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@40000000661aa78200a3799c INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 288
@40000000661aa78204f03a94 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@40000000661aa7830743db0c INFO:main:Waiting for batteries (attempt 1 of 30)...
@40000000661aa78307eec8dc INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@40000000661aa7830a10ca34 Traceback (most recent call last):
@40000000661aa7830a10e974   File "/data/BatteryAggregator/battery_service.py", line 771, in wait_for_batteries
@40000000661aa7830a10f914     batteryAggr.register(timeout=15)
@40000000661aa7830a1100e4   File "/data/BatteryAggregator/battery_service.py", line 382, in register
@40000000661aa7830a110c9c     self.service['/InstalledCapacity'] = self._configuredCapacity
@40000000661aa7830a111854   File "/data/BatteryAggregator/ext/vedbus.py", line 164, in __setitem__
@40000000661aa7830a133364     self._dbusobjects[path].local_set_value(newvalue)
@40000000661aa7830a134304 KeyError: '/InstalledCapacity'

pulquero, 2024-04-14:
That's a bug, fix pushed. BTW, dont know if it is a copy+paste error, but you are missing a comma in your config after the capacity line:
{
"capacity": 162 **,**
"primaryServices": {"com.victronenergy.battery.ttyS5": ["/Soc"]}
}

campervantime, 2024-04-14:
Thank you - I'll give it another try. Is the capacity figure used in my scenario for any purpose?

campervantime, 2024-04-14:
Have also just tried but no joy - 

@40000000661bb9f9385cf674 INFO:main:Starting...
@40000000661bb9f93b0e33c4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@40000000661bb9f93b6e31fc INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS5, scanning and storing items
@40000000661bb9fa001264fc INFO:dbusmonitor:       com.victronenergy.battery.ttyS5 has device instance 277
@40000000661bb9fa04839de4 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@40000000661bb9fa04c44894 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 288
@40000000661bb9fa0914f574 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@40000000661bb9fb238fe79c INFO:main:Waiting for batteries (attempt 1 of 30)...
@40000000661bb9fb240415cc INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
@40000000661bb9fb2e02d0f4 Traceback (most recent call last):
@40000000661bb9fb2e02f804   File "/data/BatteryAggregator/battery_service.py", line 770, in wait_for_batteries
@40000000661bb9fb2e0307a4     batteryAggr.register(timeout=15)
@40000000661bb9fb2e030f74   File "/data/BatteryAggregator/battery_service.py", line 414, in register
@40000000661bb9fb2e031b2c     self._refresh_values(paths_changed)
@40000000661bb9fb2e0322fc   File "/data/BatteryAggregator/battery_service.py", line 467, in _refresh_values
@40000000661bb9fb2e03329c     self._refresh_value(path)
@40000000661bb9fb2e06b8f4   File "/data/BatteryAggregator/battery_service.py", line 463, in _refresh_value
@40000000661bb9fb2e06d064     self._update_active_values(dbusPath)
@40000000661bb9fb2e06d834   File "/data/BatteryAggregator/battery_service.py", line 542, in _update_active_values
@40000000661bb9fb2e06e7d4     defn.action(self)
@40000000661bb9fb2e06ebbc   File "/data/BatteryAggregator/battery_service.py", line 200, in <lambda>
@40000000661bb9fb2e06f774     '/Info/MaxChargeCurrent': PathDefinition(CURRENT, aggregatorClass=NullAggregator, triggerPaths={'/Info/MaxChargeCurrent', '/Io/AllowToCharge'}, action=lambda api: api._updateCCL()),
@40000000661bb9fb2e07979c   File "/data/BatteryAggregator/battery_service.py", line 587, in _updateCCL
@40000000661bb9fb2e07ab24     currentRatios = self._get_current_ratios(connectedBatteries)
@40000000661bb9fb2e07b2f4   File "/data/BatteryAggregator/battery_service.py", line 558, in _get_current_ratios
@40000000661bb9fb2e096c5c     aggr_cap = self.aggregators["/InstalledCapacity"]
@40000000661bb9fb2e097814 KeyError: '/InstalledCapacity'

that's with using the following config.json:

{
	"capacity": 162 ,
	"primaryServices": {"com.victronenergy.battery.ttyS5": ["/Soc"]}
}

pulquero, 2024-04-14:
Pushed fix for that.

campervantime, 2024-04-14:
Thank you :) 

When I try to update on the Cerbo via Package Manager it is still only showing .49 - I presume there should be a .50 now?

pulquero, 2024-04-14:
My bad yet again, i forgot to tag it. Try now.

campervantime, 2024-04-14:
Perfect - works great now, thank you!

closed 2024-04-14 by pulquero