pulquero/BatteryAggregator #44: Charging voltage change from absorption voltage to rest voltage

RRonGit, 2024-04-16:
Hello everyone,

I have the following environment:

Raspi4 with VenusOS Verion v3.30
2x Seplos battery packs --> BMS Seplos BMS V16 (10E) Software version: 16.6
venus-os_dbus-serialbattery installed by mr-manuel Version: 1.2.20240409dev
BatteryAggregator installed via Setup Helper Version: v3.0.50

The BatteryAggregator aggregates the two batteries well so far.

Settings of venus-os_dbus-serialbattery:

In the config.ini of venus-os_dbus-serialbattery "Charge Voltage limitation (affecting CVL)" is enabled:

CVCM_ENABLE = True

This causes the battery to charge up to MAX_CELL_VOLTAGE (can seen as absorption voltage).

Then under certain circumstances the charging voltage changes to FLOAT_CELL_VOLTAGE (can be seen as resting voltage).

The problem:

If the first battery changes to FLOAT_CELL_VOLTAGE, but the second battery is still at MAX_CELL_VOLTAGE (because it still wants to do balancing), the BatteryAggregator sets the charging voltage equal to FLOAT_CELL_VOLTAGE.

The second battery cannot then complete the balancing.

It would be better if the BatteryAggregator only switches to FLOAT_CELL_VOLTAGE when both batteries switch to FLOAT_CELL_VOLTAGE.

Is this setting possible with the BatteryAggregator?

pulquero, 2024-04-16:
The current behaviour is:
If the aggregated /Balancing is 1, the max of the voltages is used, else the min.
The aggregated /Balancing is the max of the individual /Balancing.

Can you verify the behaviour of the individual battery /Balancing? 

On the aggregator, you can also set /LogLevel to INFO to see what is happening in practice (please attach any log).

RRonGit, 2024-04-18:
I noticed the following: 
one of the two batteries was balancing the cells. The other one doesn't.
In VenusOS I see that the battery that is being balanced has "Allow to balance" set to none.

[image]

Is this "Allow to balance" value the trigger for the BatteryAggregator to switch between the absorption voltage and the resting voltage?

pulquero, 2024-04-18:
No, /Io/AllowToBalance is different from /Balancing. To confirm, are you saying it is 'no' when it is balancing or 'no' when it is not balancing? You should use dbus-spy to check /Balancing.

Albarge, 2024-04-19:
I'm not sure I understand how the logic works exactly, but if any battery is still balancing and a different one switches to float early, the best solution is to increase the pre set `MAX_VOLTAGE_TIME_SEC = 900` (15 minutes) to allow all the batteries to balance for longer.

Otherwise MAX_CELL_VOLTAGE cannot be adjusted (limited lower) by the CVL functions which are arguably more important for cell protection, and a really useful function of sbattery: https://github.com/pulquero/BatteryAggregator/issues/45#issue-2251852273

RRonGit, 2024-04-21:
That increases the parameter MAX_VOLTAGE_TIME_SEC is a good idea and I've tried it now. 
Unfortunately, after reaching MAX_CELL_VOLTAGE (sbattery charge mode LINEAR_LIMITATION_ENABLE = True) it switches to float immediately and not after the time set in MAX_VOLTAGE_TIME_SEC. So that's not a solution either.

Albarge, 2024-04-21:
This is probably a Serial Battery settings issue. Maybe this thread is useful to read and change to suggested settings to make it work:

https://github.com/Louisvdw/dbus-serialbattery/discussions/1036#discussion-6510569

RRonGit, 2024-04-24:
Now it's working.
I adjusted the following settings in sbatterie config.ini:

; --------- Charge mode ---------
LINEAR_LIMITATION_ENABLE = True

; --------- Charge Voltage limitation (affecting CVL) ---------
CVCM_ENABLE = True

; -- CVL reset based on cell voltage diff (linear mode)
MAX_VOLTAGE_TIME_SEC = 3600

; --------- Voltage drop ---------
VOLTAGE_DROP = 0.050

Thank you very much for the help/support.

closed 2024-04-24 by pulquero