pulquero/BatteryAggregator #47: CCL and DCL in new Capacity and Count modes does not take in to account "Allow to charge" and "Allow to discharge"

AlexdeKairos, 2024-04-25:
New modes does not take into account the Allow to charge and Allow to discharge if any of the parallel batteries goes into protection and the total CCL and DCL remains well pass the actual limits.

pulquero, 2024-04-25:
Would it be useful if I added the ability to be able to manually specify something like the "capacities" so it could be manually tuned?

AlexdeKairos, 2024-04-25:
Battery capacity and BMS capacity  are published in dbus and can be easily read. I think its a matter of limiting the computed values (whatever system is selected by user IR, Capacity and Count) and verify before publishing if battery is under protection for charging and/or discharging,  and limit the value to the max bms capacity,  if applicable.  That should work with any computation. With dbus-serialbattery there is already implemented a max charge protection that could be based on SOC, Battery temperature or cell voltage.  In this scenario batteries are well protected and the direct sum of CCL and DCL is enough.  Other batteries like CAB BUS connected will take real benefit of your extra level of protection. So behaviour can be automated and all info read from dbus or CAN. I don't actually know if there is other connection type available for batteries...

pulquero, 2024-04-27:
I already do this: for each battery I calculate the overall current reqd to reach its ccl then pick the lowest overall current as the aggregated ccl. I don't think there is anything additional I can do to know whether it is going to result in an overcurrent for a particular battery.

Maybe what I could do is start at 90% of that value and slowly increase upwards.

AlexdeKairos, 2024-04-28:
Yes, all good but when a Battery goes into protection, its capacity
shouldn't be taken in consideration for the total CCL if the charge is
closed and DCL if the discharge is closed. At the moment if I close the
charge of one battery to simulate a protection. The CCL does not change and
remaining battery goes to protection due to overcharge.
If Capacity is used, the ah remaining in that battery (protected one)
should be removed from the computation. Only remaining batteries ah should
be used.
Same with Count mode...

pulquero, 2024-04-28:
The code already tracks /Io/AllowToCharge etc and excludes disconnected batteries from consideration. Please turn on debug logging and capture what dbus messages are flying about so we can verify what is actually happening.

AlexdeKairos, 2024-04-30:
Hello! Testing day today. But unfortunatelly i I follow your debug activation sentence, I get an invalid json file and I'm unable to do the test. Thats probably a silly question but I'm stuk.

Here is my json file. It only includes the exclusion of the installed victron shunt on possition 6.
Any  try to include debug: 	"logLevel": "DEBUG"
becomes in log as invalid JSON.
Can you create me a valid JSON with my setup?

[config.json](https://github.com/pulquero/BatteryAggregator/files/15161528/config.json)

pulquero, 2024-04-30:
Sure,
`
{
"logLevel": "DEBUG",
	"excludedServices": ["com.victronenergy.battery.ttyS6"]
}	
`

Note, you can change the loglevel dynamically too. BatteryAggregator has a LogLevel dbus path that you can switch between DEBUG and INFO.

AlexdeKairos, 2024-04-30:
Ok here the logs. The test has been performed firs, disconecting and conecting charge and discharge of battery ONE. this test resoult was correct. So behaviour as expected. The test is repeated with battery TWO. Here the reduced limit is not applied and regardless of the disconection (chrge or discharge) there is no significant reduction in CCL and DCL

AlexdeKairos, 2024-04-30:
Count mode is the only one that seems to work.

Here logs 

@400000006630d8783033031c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Power
@400000006630d8783045f2c4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM1: -90.11} now has values {'com.victronenergy.battery.ttyACM0': -78.93, 'com.victronenergy.battery.ttyACM1': -90.11} with result -169.04000000000002
@400000006630d878347ce1cc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MinCellVoltage
@400000006630d878348db27c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinCellVoltage updated with {com.victronenergy.battery.ttyACM1: 3.329} now has values {'com.victronenergy.battery.ttyACM0': 3.326, 'com.victronenergy.battery.ttyACM1': 3.329} with result 3.326
@400000006630d87834a04464 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MaxCellVoltage
@400000006630d87834b7337c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyACM1: 3.33} now has values {'com.victronenergy.battery.ttyACM0': 3.331, 'com.victronenergy.battery.ttyACM1': 3.33} with result 3.331
@400000006630d87930df8efc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Current
@400000006630d879318a9c0c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyACM1: -6.68} now has values {'com.victronenergy.battery.ttyACM0': -5.93, 'com.victronenergy.battery.ttyACM1': -6.68} with result -12.61
@400000006630d8793221d6e4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Power
@400000006630d87932391bec DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM1: -88.91} now has values {'com.victronenergy.battery.ttyACM0': -78.93, 'com.victronenergy.battery.ttyACM1': -88.91} with result -167.84
@400000006630d8793541fad4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MinCellVoltage
@400000006630d87935bb0334 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinCellVoltage updated with {com.victronenergy.battery.ttyACM1: 3.328} now has values {'com.victronenergy.battery.ttyACM0': 3.326, 'com.victronenergy.battery.ttyACM1': 3.328} with result 3.326
@400000006630d87935d91284 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MaxCellVoltage
@400000006630d87935ed80fc DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyACM1: 3.329} now has values {'com.victronenergy.battery.ttyACM0': 3.331, 'com.victronenergy.battery.ttyACM1': 3.329} with result 3.331
@400000006630d87a0c802454 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Current
@400000006630d87a0d2d5444 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyACM0: -6.02} now has values {'com.victronenergy.battery.ttyACM0': -6.02, 'com.victronenergy.battery.ttyACM1': -6.68} with result -12.7
@400000006630d87a0e4e394c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Power
@400000006630d87a0e71132c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM0: -80.13} now has values {'com.victronenergy.battery.ttyACM0': -80.13, 'com.victronenergy.battery.ttyACM1': -88.91} with result -169.04
@400000006630d87a10a066ac DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxVoltageCellId
@400000006630d87a10d253ec DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyACM0: C3} now has values {'com.victronenergy.battery.ttyACM0': 'C3', 'com.victronenergy.battery.ttyACM1': 'C1'} with result C3
@400000006630d87a2feb6ee4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Current
@400000006630d87a3052c01c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyACM1: -6.86} now has values {'com.victronenergy.battery.ttyACM0': -6.02, 'com.victronenergy.battery.ttyACM1': -6.86} with result -12.879999999999999
@400000006630d87a312e918c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Power
@400000006630d87a323c8ebc DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM1: -91.31} now has values {'com.victronenergy.battery.ttyACM0': -80.13, 'com.victronenergy.battery.ttyACM1': -91.31} with result -171.44
@400000006630d87a334298c4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MaxCellVoltage
[... cut ...] ': 3.331, 'com.victronenergy.battery.ttyACM1': 3.33} with result 3.331
@400000006630d880306bca44 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /System/MinVoltageCellId
@400000006630d8803088e37c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinVoltageCellId updated with {com.victronenergy.battery.ttyACM1: C4} now has values {'com.victronenergy.battery.ttyACM0': 'C4', 'com.victronenergy.battery.ttyACM1': 'C4'} with result C4

pulquero, 2024-04-30:
What I'm expecting to see are /Io/AllowToCharge messages everytime a battery connects/disconnects. These are one of the messages that trigger a recalculation of the CCL. In the first log there was only ever one for disabling ACM1. Never one for enabling it. I think you need to ascertain whether this is being correctly set on the batteries, and maybe if for some reason the aggregator is not picking them up. Or whether the aggregator should be listening to some other alternative message instead.

AlexdeKairos, 2024-04-30:
Hi.
Battery is connecting and disconnecting both of  them. I have checked with bluetooth app and in cerbo UI and they repond properly. So Battery aggregator is not caching the changes...

I send you an additional. Just changed to mode to IR

AlexdeKairos, 2024-04-30:
here last log
@4000000066311ddb0085cfdc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Power
@4000000066311ddb0085f6ec DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM0: -25.74} now has values {'com.victronenergy.battery.ttyACM0': -25.74, 'com.victronenergy.battery.ttyACM1': 0.0} with result -25.74
@4000000066311ddb036fdc24 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxVoltageCellId
@4000000066311ddb03824ae4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyACM0: C3} now has values {'com.victronenergy.battery.ttyACM0': 'C3', 'com.victronenergy.battery.ttyACM1': 'C3'} with result C3
@4000000066311ddb37734254 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Voltage
@4000000066311ddb378c8ee4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Voltage updated with {com.victronenergy.battery.ttyACM1: 13.68} now has values {'com.victronenergy.battery.ttyACM0': 13.69, 'com.victronenergy.battery.ttyACM1': 13.68} with result 13.684999999999999
@4000000066311ddb39be4b94 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Current
@4000000066311ddb39f3d2b4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyACM0: -1.97} now has values {'com.victronenergy.battery.ttyACM0': -1.97, 'com.victronenergy.battery.ttyACM1': 0.0} with result -1.97
@4000000066311ddb3a22be6c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Power
@4000000066311ddb3a6afe84 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM0: -26.97} now has values {'com.victronenergy.battery.ttyACM0': -26.97, 'com.victronenergy.battery.ttyACM1': 0.0} with result -26.97
@4000000066311ddc37c45054 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Current
@4000000066311ddc37ddb83c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyACM0: -2.06} now has values {'com.victronenergy.battery.ttyACM0': -2.06, 'com.victronenergy.battery.ttyACM1': 0.0} with result -2.06
@4000000066311ddc38ac85a4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /Dc/0/Power
@4000000066311ddc38ef4394 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyACM0: -28.2} now has values {'com.victronenergy.battery.ttyACM0': -28.2, 'com.victronenergy.battery.ttyACM1': 0.0} with result -28.2
@4000000066311ddc39805834 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxVoltageCellId
@4000000066311ddc3990c73c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyACM0: C1} now has values {'com.victronenergy.battery.ttyACM0': 'C1', 'com.victronenergy.battery.ttyACM1': 'C3'} with result C3
@4000000066311ddc3a073b74 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MinCellVoltage
@4000000066311ddc3b0828e4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinCellVoltage updated with {com.victronenergy.battery.ttyACM0: 3.42} now has values {'com.victronenergy.battery.ttyACM0': 3.42, 'com.victronenergy.battery.ttyACM1': 3.42} with result 3.42
@4000000066311ddd027e38c4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxCellVoltage
@4000000066311ddd0296394c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyACM0: 3.424} now has values {'com.victronenergy.battery.ttyACM0': 3.424, 'com.victronenergy.battery.ttyACM1': 3.424} with result 3.424
@4000000066311ddd02be33ac DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM1 /Dc/0/Voltage
@4000000066311ddd02d4f7cc DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Voltage updated with {com.victronenergy.battery.ttyACM1: 13.69} now has values {'com.victronenergy.battery.ttyACM0': 13.69, 'com.victronenergy.battery.ttyACM1': 13.69} with result 13.69
@4000000066311ddd37c4831c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxVoltageCellId
@4000000066311ddd37e17544 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyACM0: C3} now has values {'com.victronenergy.battery.ttyACM0': 'C3', 'com.victronenergy.battery.ttyACM1': 'C3'} with result C3
@4000000066311ddd38f3ac7c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxCellVoltage
@4000000066311ddd390f4acc DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyACM0: 3.425} now has values {'com.victronenergy.battery.ttyACM0': 3.425, 'com.victronenergy.battery.ttyACM1': 3.424} with re [... cut ...] ACM0': 'C3', 'com.victronenergy.battery.ttyACM1': 'C3'} with result C3
@4000000066311deb393587dc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyACM0 /System/MaxVoltageCellId
@4000000066311deb39a312d4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyACM0: C1} now has values {'com.victronenergy.battery.ttyACM0': 'C1', 'com.victronenergy.battery.ttyACM1': 'C3'} with result C3

pulquero, 2024-05-02:
To check for inconsistencies between the batteries and aggregator, I've create a dbus_dump.py tool available in the latest version. Please run:

`python3 dbus_dump.py com.victronenergy.battery`

before and after a battery connection/disconnection. It'll write files to /data/dbus_dump_*. Please attach these files to this issue, thanks.

AlexdeKairos, 2024-05-07:
may be something misssing?

python3: can't open file 'dbus_dump.py': [Errno 2] No such file or directory

pulquero, 2024-05-08:
You probably havent changed directories, try `cd /data/BatteryAggregator` first. Sure, happy to have a remote session via discord or something.

AlexdeKairos, 2024-05-08:
Hi. I wont be able to test untill next week initially. We keep in touch to prepare a remote session

AlexdeKairos, 2024-05-15:
@pulquero  Hi, if you want. About 1030 am, I'll be able for a remote session

pulquero, 2024-05-15:
Good with me.

AlexdeKairos, 2024-05-15:
im online right now. Alexdekairos in discord

closed 2024-06-06 by pulquero