pulquero/BatteryAggregator #48: load to SOC reset

rolandlueders, 2024-05-01:
Hi!
i have three banks 24V 300Ah with JK-BMS in parallel. In Serialbattery is a feature to SOC reset a jk BMS. There is a timeplan to load to an higher Voltage. The cell voltage ist controlled and the end of this feature is reached, when the Cells are balanced better as 10mV and the "Load to SOC reset Voltage" is reched. The problem is that it is stopped when only one of the Batteries reached this target. It would be better when all batteries had reached the criteria.

pulquero, 2024-05-02:
To understand what is going on, please switch to the head branch to download the beta/early release version. Then run:

python3 dbus_dump.py com.victronenergy.battery

before and after the problem. It'll write files to /data/dbus_dump_*. Please attach these files to this issue, thanks.

rolandlueders, 2024-05-03:
one Battery finished the "Load to SOC Reset" a few days before. Now the Carge Voltage Limit ist 27.44V
[image]
The other two Batteries are in ""Bulk and SOC Reset" mode with a Charge Voltage Limit of 27.84V 
[image]
But this Voltage will not reached, because BatteryAggregator limits to 27,44V
[image]

pulquero, 2024-05-06:
If the batteries are not being balanced, then the min CVL is used to prevent overvoltage. Wouldnt a voltage of 27.84 cause overvoltage on the second battery? Won't that be a problem?

rolandlueders, 2024-05-07:
in Dbus-Serialbattery is a feature named "SOC_RESET_AFTER_DAYS ". I set it to7. That means a timer ist set to 7 and the CVL will be rised to "SOC_RESET_VOLTAGE" which i set to 3.48V per cell so CVL should be 27.84V. I use it to force balancing every 7 days.  When the battery reaches 27.48V and is well balanced the timer will be cleared. So the next balancing is in 7 Days. If it is cloudly the timer will not be cleared so it will try next day again. But i have tree BMSes. When the first Battery reaches the critera, the other Batteries can not load to 27,84V because the timer is cleared now. But CVL should not reduced untill all three Batteries are balanced at 27.84V.  I asked Mr.Manuel but he says ist is a Problem of Batteryaggreator.

pulquero, 2024-05-07:
Ok, that makes sense, but I would have thought the other two batteries should still report as being balanced, and then the aggregator would do the right thing. Or why dont the other batteries have a high CVL if they havent finished balancing yet? Maybe you could attach the serialbattery config of your batteries and I can try to trace through the serialbattery code to understand the mechanics better.

rolandlueders, 2024-05-08:
The other batteries are not finished with balancing. In one case the balancing was interrupted by a heating wascher. In another case the first battery reached float. I say an other issue with a similar problem in #44. Here is my config.ini:
MAX_BATTERY_CHARGE_CURRENT =150.0
MAX_BATTERY_DISCHARGE_CURRENT =180.0
MAX_CELL_VOLTAGE   = 3.43 ; this voltage sould be as low as possible because the batteries in  summer are always full
SOC_RESET_VOLTAGE = 3.48 ; by now the voltage rises rapidly
SOC_RESET_AFTER_DAYS = 7 ; for testing
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = false
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = false
CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.2,  0.3,    1
BMS_TYPE = Jkbms

rolandlueders, 2024-05-08:
I closed this issue by mistake.
I forgot to send the BMS values
Balance Trig. Volt. 0.005
Start Balance Volt. 3.47

pulquero, 2024-05-17:
Try v3.0.61.

closed 2024-05-29 by rolandlueders