pulquero/BatteryAggregator #49: V

superkrischi, 2024-05-10:
To provide a debug log, use dbus-spy to change the D-Bus path `/LogLevel` for the BatteryAggregator to `DEBUG`. Log location is `/var/log/BatteryAggregator/current`.

superkrischi, 2024-05-10:
Hello, 

i use v3.0.57. My System had 4x16 Cells with JKBMS in parallel.

Aggregator is always at 55.2V, all packs are now in float. 
Sometimes after reboot one of the packs has no CVL in the parameters.

Thanks a lot and greetings
Kristian

Here is the log.

@40000000663df5290f073c94 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyUSB3: C4} now has values {'com.victronenergy.battery.ttyUSB4': 'C1', 'com.victronenergy.battery.ttyUSB5': None, 'com.victronenergy.battery.ttyUSB1': 'C5', dbus.String('com.victronenergy.battery.ttyUSB3'): 'C4'} with result C5
@40000000663df52934fa957c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB4 /Dc/0/Voltage 55.37
@40000000663df5293511f1f4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Voltage updated with {com.victronenergy.battery.ttyUSB4: 55.37} now has values {'com.victronenergy.battery.ttyUSB4': 55.37, 'com.victronenergy.battery.ttyUSB5': 55.37, 'com.victronenergy.battery.ttyUSB1': 55.37, dbus.String('com.victronenergy.battery.ttyUSB3'): 55.36} with result 55.36749999999999
@40000000663df529353fe794 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB4 /Dc/0/Power 105.76
@40000000663df529354eef3c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB4: 105.76} now has values {'com.victronenergy.battery.ttyUSB4': 105.76, 'com.victronenergy.battery.ttyUSB5': -89.7, 'com.victronenergy.battery.ttyUSB1': 105.76, dbus.String('com.victronenergy.battery.ttyUSB3'): 100.76} with result 222.58
@40000000663df52935747c84 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB4 /System/MinVoltageCellId C4
@40000000663df529357c0a1c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinVoltageCellId updated with {com.victronenergy.battery.ttyUSB4: C4} now has values {'com.victronenergy.battery.ttyUSB4': 'C4', 'com.victronenergy.battery.ttyUSB5': None, 'com.victronenergy.battery.ttyUSB1': 'C9', dbus.String('com.victronenergy.battery.ttyUSB3'): 'C6'} with result C4
@40000000663df52937289cf4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Current 2.13
@40000000663df529372c13ac DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB1: 2.13} now has values {'com.victronenergy.battery.ttyUSB4': 1.91, 'com.victronenergy.battery.ttyUSB5': -1.62, 'com.victronenergy.battery.ttyUSB1': 2.13, dbus.String('com.victronenergy.battery.ttyUSB3'): 1.82} with result 4.24
@40000000663df52937498aa4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Power 117.94
@40000000663df5293752a2c4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB1: 117.94} now has values {'com.victronenergy.battery.ttyUSB4': 105.76, 'com.victronenergy.battery.ttyUSB5': -89.7, 'com.victronenergy.battery.ttyUSB1': 117.94, dbus.String('com.victronenergy.battery.ttyUSB3'): 100.76} with result 234.76
@40000000663df5293793c2a4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxVoltageCellId C3
@40000000663df529379b486c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyUSB1: C3} now has values {'com.victronenergy.battery.ttyUSB4': 'C1', 'com.victronenergy.battery.ttyUSB5': None, 'com.victronenergy.battery.ttyUSB1': 'C3', dbus.String('com.victronenergy.battery.ttyUSB3'): 'C4'} with result C4
@40000000663df52937b767a4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.464
@40000000663df52937c25c54 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.464} now has values {'com.victronenergy.battery.ttyUSB4': 3.461, 'com.victronenergy.battery.ttyUSB5': None, 'com.victronenergy.battery.ttyUSB1': 3.464, dbus.String('com.victronenergy.battery.ttyUSB3'): 3.462} with result 3.464
@40000000663df52a0db4339c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB3 /Dc/0/Current -1.82
@40000000663df52a0dd6ee3c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB3: -1.82} now has values {'com.victronenergy.battery.ttyUSB4': 1.91, 'com.victronenergy.battery.ttyUSB5': -1.62, 'com.victronenergy.battery.ttyUSB1': 2.13, dbus.String('com.victronenergy.battery.ttyUSB3'): -1.82} with result 0.5999999999999999
@40000000663df52a0e60c364 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB3 /Dc/0/Power -100.76
@40000000663df52a0e68aad4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB3: -100.76} now has values {'com.victronenergy.battery.ttyUSB4': 105.76, 'com.victronenergy.battery.ttyUSB5': -89.7, 'com.victronenergy.battery.ttyUSB1': 117.94, dbus.String('com.victronenergy.battery.ttyUSB3'): -100.76} with result 33.239999999999995
[... cut ...]
@40000000663df52a36278c5c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB4 /System/MaxCellVoltage 3.462
@40000000663df52a363ba8cc DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB4: 3.462} now has values {'com.victronenergy.battery.ttyUSB4': 3.462, 'com.victronenergy.battery.ttyUSB5': None, 'com.victronenergy.battery.ttyUSB1': 3.464, dbus.String('com.victronenergy.battery.ttyUSB3'): 3.462} with result 3.464

pulquero, 2024-05-10:
Can you include some screen shots to show exactly what you mean, please.

superkrischi, 2024-05-12:
Hello, 

thanks for your answer.
Since downgrade to v3.0.52 all is perfect. 

Also tested on a 12V Victron Setup, the same.

so i need actual no more help.

Thanks a lot and greetings 
Kristian

closed 2024-05-12 by superkrischi

pulquero, 2024-05-12:
Just to verify, are you saying v3.0.57 does not work, but v3.0.52 does? Somewhat ironically, v3.0.57 contains support for DVCC, but it seems to have caused you issues. I may look into adding a config option to disable DVCC support in BatteryAggregator then.