pulquero/BatteryAggregator #50: Since v3.0.60 Again CCL = 0 and DCL =0 #40

Rubens974, 2024-05-16:
Hi,
Since v3.0.60 I have again the same issue like in #40 .
Here are logs
[@40000000664641eb35bfc5f4.s.txt](https://github.com/pulquero/BatteryAggregator/files/15338887/%4040000000664641eb35bfc5f4.s.txt)
[@400000006646423711b8f2fc.s.txt](https://github.com/pulquero/BatteryAggregator/files/15338888/%40400000006646423711b8f2fc.s.txt)
[@400000006646428535aaee04.s.txt](https://github.com/pulquero/BatteryAggregator/files/15338889/%40400000006646428535aaee04.s.txt)
[current.txt](https://github.com/pulquero/BatteryAggregator/files/15338890/current.txt)

Regards

pulquero, 2024-05-16:
Sorry about that. What is the value of /Info/MaxChargeCurrent and /Io/AllowToCharge for one of your batteries? I'm guessing one is 0 the other is null and I'm checking the wrong one.

Rubens974, 2024-05-17:
Hi,
Bank 1: CCL 200A  and CVL 54.5V
Bank 2: CCL 175A and CVL 53.2V
BatteryAggregator: null
[image]
[image]

Logs
[@400000006646ba1f00deb5ac.s.txt](https://github.com/pulquero/BatteryAggregator/files/15342982/%40400000006646ba1f00deb5ac.s.txt)
[@400000006646ba6e24f70d04.s.txt](https://github.com/pulquero/BatteryAggregator/files/15342984/%40400000006646ba6e24f70d04.s.txt)
[current.txt](https://github.com/pulquero/BatteryAggregator/files/15342985/current.txt)

Thank you

pulquero, 2024-05-17:
It was an entirely different issue to what I thought, should be fixed in v3.0.61.