pulquero/BatteryAggregator #51: CCL to low?

endurance1968, 2024-05-27:
Hi,

issues and a problem to understand how CCL calculation works. From the logs I get

```
@40000000665411ed08b8c3e4 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB0': 140.0}
@40000000665411ed08ced83c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0']
@40000000665411ed090dfc4c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.0, 'capacity')]
@40000000665411ed091e8a94 INFO:com.victronenergy.battery.aggregator:DCL estimates: [140.0]
@40000000665411ed09c3b7bc INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB0': 150.0}
@40000000665411ed0a2e395c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0']
@40000000665411ed0a2e606c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.0, 'capacity')]
@40000000665411ed0a2e700c INFO:com.victronenergy.battery.aggregator:CCL estimates: [150.0]
@40000000665411ed0a9d52c4 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB0': 140.0}
@40000000665411ed0ab09c44 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0']
@40000000665411ed0ac24b9c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.0, 'capacity')]
@40000000665411ed0b0d6dfc INFO:com.victronenergy.battery.aggregator:DCL estimates: [140.0]
@40000000665411ed0b67365c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB0': 150.0}
@40000000665411ed0bce54cc INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0']
@40000000665411ed0c94350c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.0, 'capacity')]
@40000000665411ed0ca57704 INFO:com.victronenergy.battery.aggregator:CCL estimates: [150.0]
```

If I then check the Venus system CCL is set to 100 only (should be 150A) 

any idea what to check?

Config json:

```
{
        "excludedServices": ["com.victronenergy.battery.socketcan_can0_vi0_uc288664"],
        "logLevel": "DEBUG"
}
```
installed version is 3.0.61

pulquero, 2024-05-27:
If you are using DVCC, then the CCL will be capped by that (under DVCC menu, you can set a max current limit, in dbus, "com.victronenergy.settings", "/Settings/SystemSetup/MaxChargeCurrent").

endurance1968, 2024-05-27:
OK got it - tested and confirmed. :)

closed 2024-05-27 by pulquero