pulquero/BatteryAggregator #52: CCL and DCL to high

locke987, 2024-05-30:
The CCL and DCL paramter settings are higher in the aggregator as they are on the 2 Bat. Packs.
I've installed the serial bat driver as well, both Bat Packs have a setting of max 75A for discharge.
So DCL should not be higher than 150A, but is 225A, CCL is also higher than expected.
Hardware: RPi4
Setup: DVCC yes
Version:  3.0.61

[image]
[image]
[image]

config.json:
root@raspberrypi4:~# cat /data/setupOptions/BatteryAggregator/config.json

{
        "loglevel": "INFO",
        "currentRatioMethod": "capacity"
}

Log:
root@raspberrypi4:~# cat /var/log/BatteryAggregator/current

@400000006658481a14989734 INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@400000006658481a1498ba5c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658481a1498cde4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658481a1498dd84 INFO:com.victronenergy.battery.aggregator:CCL estimates: [104.39999999999999, 120.75]
@4000000066584824158744c4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 36.825, dbus.String('com.victronenergy.battery.ttyUSB4'): 40.25}
@400000006658482415924144 INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@4000000066584824159b075c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658482415a2fa84 INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658482415aae9c4 INFO:com.victronenergy.battery.aggregator:CCL estimates: [110.47500000000001, 120.75]
@4000000066584829346bb744 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 36.825, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@400000006658482934982e14 INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@400000006658482934aae324 WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658482934c9447c INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658482934d3375c INFO:com.victronenergy.battery.aggregator:CCL estimates: [110.47500000000001, 123.75]
@400000006658483b173229ec INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 38.75, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@400000006658483b173e66bc INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@400000006658483b1747964c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658483b17500674 INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658483b17583fec INFO:com.victronenergy.battery.aggregator:CCL estimates: [116.25, 123.75]
@400000006658484016e2b22c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 36.15, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@40000000665848401721ca84 INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@40000000665848401721e9c4 WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@40000000665848401721fd4c INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658484017220cec INFO:com.victronenergy.battery.aggregator:CCL estimates: [108.44999999999999, 123.75]
@400000006658485218f4ec74 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 34.125, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@40000000665848521901d524 INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@4000000066584852191424a4 WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@4000000066584852191dbdac INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@40000000665848521928b25c INFO:com.victronenergy.battery.aggregator:CCL estimates: [102.375, 123.75]
@400000006658485e19c6abc4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 36.15, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@400000006658485e19d19c8c INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@400000006658485e19e23e5c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658485e19ecfc5c INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658485e19f6f324 INFO:com.victronenergy.battery.aggregator:CCL estimates: [108.44999999999999, 123.75]
@40000000665848621a7cfba4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 34.125, dbus.String('com.victronenergy.battery.ttyUSB4'): 41.25}
@40000000665848621a88137c INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@40000000665848621ab3b75c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@40000000665848621aca4c9c INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@40000000665848621ad4d3ec INFO:com.victronenergy.battery.aggregator:CCL estimates: [102.375, 123.75]
@400000006658486538c0cd0c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {dbus.String('com.victronenergy.system'): None, dbus.String('com.victronenergy.battery.ttyUSB3'): 34.125, dbus.String('com.victronenergy.battery.ttyUSB4'): 39.75}
@400000006658486538cb351c INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.system'), dbus.String('com.victronenergy.battery.ttyUSB3'), dbus.String('com.victronenergy.battery.ttyUSB4')]
@400000006658486538d95e1c WARNING:com.victronenergy.battery.aggregator:/InstalledCapacity is not available for com.victronenergy.system
@400000006658486538ecea04 INFO:com.victronenergy.battery.aggregator:Current ratios: [(3, 'count'), (3, 'count'), (3, 'count')]
@400000006658486538f7abec INFO:com.victronenergy.battery.aggregator:CCL estimates: [102.375, 119.25]
root@raspberrypi4:~#

pulquero, 2024-05-30:
Fixed in v3.0.62. It was due to com.victronenergy.system being counted as a battery, so it thought there were 3 batteries in total.

locke987, 2024-05-31:
Thank you, I installed the newer version and it looks better now.

closed 2024-05-31 by pulquero