pulquero/BatteryAggregator #57: Feature Request - config check

endurance1968, 2024-06-08:
Typically struggling with the correct format within the config.json. Unfortunately the logfile is overwritten quickly so I never saw (if it exists) errors while startup.

Woul be nice to have an option to just check the config file an printout in case of issues or show a short summary of the resulting behaviour.

Maybe this exists already??

pulquero, 2024-06-11:
To check syntax, there are various online validators. Maybe in the same vein, an online config wizard/generator would help.

endurance1968, 2024-06-11:
Currently it seems people are trying to adapt the config and then wonder that nothing changes (I assume the Aggregator falls back to all defaults in case of issues). Online validator could help checking syntax but e.g.  not on finding typos on the keys evaluated by the BA. A config generator of course would be super nice but do not know how much effort this would be.

KaiGrassnick, 2024-07-02:
@endurance1968 i created a PR merging these features into this Project. Feel free to check them out. Maybe you could also give some feedback about the first version of the config generator https://kaigrassnick.github.io/BatteryAggregator/config_generator/index.html

closed 2024-07-02 by pulquero