pulquero/BatteryAggregator #59: Compatible with python new version?

Rubens974, 2024-06-20:
Hi there
When I use Python3.12 this is what happen:

_# python3.12 /data/BatteryAggregator/battery_service.py
Traceback (most recent call last):
  File "/data/BatteryAggregator/battery_service.py", line 8, in <module>
    import dbus
  File "/usr/local/lib/python3.12/site-packages/dbus/__init__.py", line 73, in <module>
    import dbus.types as types
  File "/usr/local/lib/python3.12/site-packages/dbus/types.py", line 8, in <module>
    from _dbus_bindings import (
ModuleNotFoundError: No module named '_dbus_bindings'

Regards

Rubens974, 2024-06-20:
Downgrading to python 3.11 solves my problem.

pulquero, 2024-06-20:
I think this depends on a new version of Venus to support 3.12.

Rubens974, 2024-06-23:
In fact I think python3.12 is a bit faster and sometimes the command
 "dbus -y com.victronenergy.system /Dc/Battery/Soc GetValue  
doesn't do anything.
This can be solved by adding a 0.5s pause.