pulquero/BatteryAggregator #62: Troubleshooting support: Logfiles to analyze crash

endurance1968, 2024-07-05:
From time to time Battery Aggregator seems to crash.
This morning I got an alarm beep from my garage and saw that the BMSs are on alarm with low voltage.
Looking at the Cerbo I saw that the BA showed "no connection" (typically an indication of crash wo auto recovery). To fix I manually restarted BA.

Where to find old logs (if any)?
Can I influence the log rotation (length and number of files) on a Venus device?

pulquero, 2024-07-05:
This file controls the logging: https://github.com/pulquero/BatteryAggregator/blob/main/service/log/run.

Multilog docs: https://manpages.debian.org/stretch/daemontools/multilog.8.en.html.

endurance1968, 2024-07-08:
thanks helps - i will try to tune parameters

closed 2024-07-08 by endurance1968