pulquero/BatteryAggregator #64: Regular but unreproducable crash

endurance1968, 2024-07-10:
Cerbo

```
@40000000668ed8fa386e674c Traceback (most recent call last):
@40000000668ed8fa386e8e5c   File "/data/BatteryAggregator/battery_service.py", line 12, in <module>
@40000000668ed8fa386e9a14     from vedbus import VeDbusService
@40000000668ed8fa386ea1e4 ModuleNotFoundError: No module named 'vedbus'
@40000000668ed8fc213267cc Traceback (most recent call last):
@40000000668ed8fc21328edc   File "/data/BatteryAggregator/battery_service.py", line 12, in <module>
@40000000668ed8fc21329e7c     from vedbus import VeDbusService
@40000000668ed8fc2132a64c ModuleNotFoundError: No module named 'vedbus'
@40000000668ed8fd300ce164 Traceback (most recent call last):
@40000000668ed8fd300d048c   File "/data/BatteryAggregator/battery_service.py", line 12, in <module>
@40000000668ed8fd300d142c     from vedbus import VeDbusService
@40000000668ed8fd300d1bfc ModuleNotFoundError: No module named 'vedbus'
```

It happens after a few days. I updated now to 3.0.70 (from 68) - lets see

endurance1968, 2024-08-21:
until now stable - I had the impression it was related to availability of an update via the paketmanager. So it would interesting to see if a crash happens again as soon there is an update avaible. if not this ticket can be closed.

endurance1968, 2024-08-27:
I close this - now version 75 is out and had no issues. In case it happens again I might reopen.

closed 2024-08-27 by endurance1968

KaiGrassnick, 2024-09-12:
Same issue here. On latest version. Happend second time after about 2 weeks running without a flaw