pulquero/BatteryAggregator #65: Charge Current Limit to high

wetty, 2024-08-14:
Hello,

I have two JKBMS with 18S LiFePo and SerialBattery 1.2.20240408 running.
Both Batteries are nearly identical. Battery two is 6 month younger and the cells might have 5-10Ah more.

In SerialBattery the CCL ist reduced according the Cell Voltage.

Battery Aggregator is reducing the current to twice the value of the minimum CCL of the two batteries. In my case this was yesterday 7.2A for Battery 2 which results in BatteryAggregator as 14.4A

Unfortunately Battery 1 is just consuming 4.03A of this 14.4A CCL which resulted in 10.27A for Battery 2. But this in this case is higher then the CCL Setting 7.2A for this Battery.

Worst case would be that Battery 1 is not charging any more so the CCL of 14.4A will be consumed from Battery 2.

So is there a way to reduce the CCL dynamicall to the lowest value of one of the Batteries, so that the lowest value of one of the batteries does not exceed?
Do I misunderstand something? Is it an configuration Issue?

Battery 1
[image]

Battery 2
[image]

Battery Aggregator with CCL Limit twice the value of Battery 2 
[image]

Battery 2 loading with higher current, as the CCL for this Battery
[image]

Please include system details:

Hardware: Cerbo/RPi???
RPi 3, Venus OS 3.50-13

Setup: DVCC?
DVCC ON
All Settings OFF, also tested with SVS and SVC On
Controlling BMS Battery Aggregator

Please provide output from dbus-spy.
com.victronenergy.adc
com.victronenergy.battery.aggregator                                                                                                                                                                                                                      Battery Aggregator
com.victronenergy.battery.ttyUSB1                                                                                                                                                                                                                       SerialBattery(Jkbms)
com.victronenergy.battery.ttyUSB2                                                                                                                                                                                                                       SerialBattery(Jkbms)
com.victronenergy.fronius
com.victronenergy.grid.cg_BX12200350015                                                                                                                                                                                             Carlo Gavazzi EM24 Ethernet Energy Meter
com.victronenergy.hub4

com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                                                                                                                                                                                                             Package Manager
com.victronenergy.platform                                                                                                                                                                                                                                             Venus
com.victronenergy.pvinverter.pv_121_1803953                                                                                                                                                                                                            Fronius Symo 20.0-3-M
com.victronenergy.settings
com.victronenergy.shutdown                                                                                                                                                                                                                                   ShutdownMonitor
com.victronenergy.system
com.victronenergy.tailscaleGX                                                                                                                                                                                                                            TailscaleGX-control
com.victronenergy.temperature.RpiCpu06                                                                                                                                                                                                                          Raspberry Pi
com.victronenergy.vebus.ttyUSB0

Please attach any relevant logs. Log location is `/var/log/BatteryAggregator/current`.
@4000000066bc64682ec73e1c INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.battery.ttyUSB2'), dbus.String('com.victronenergy.battery.ttyUSB1')]
@4000000066bc64682ec7d674 INFO:com.victronenergy.battery.aggregator:Current ratios: [(2, 'count'), (2, 'count')]
@4000000066bc64682ec7e9fc INFO:com.victronenergy.battery.aggregator:CCL estimates: [120.0, 120.0]

wetty, 2024-08-16:
Hi, and where can I add this option?

wetty, 2024-08-16:
> Hello, I have a similar problem. You could add the option to specify the Max Charge Current based on voltage

And Where can I add this option? In the BatteryAggregator config?

adcrafter27, 2024-08-16:
> > Hello, I have a similar problem. You could add the option to specify the Max Charge Current based on voltage
> 
> And Where can I add this option? In the BatteryAggregator config?

The question was asked to @pulquero  because I also have a problem

wetty, 2024-08-16:
> > > Hello, I have a similar problem. You could add the option to specify the Max Charge Current based on voltage
> > 
> > 
> > And Where can I add this option? In the BatteryAggregator config?
> 
> The question was asked to @pulquero because I also have a problem

pulquero, 2024-08-16:
What is the reason for having a different CCL for each battery? Is it because of their different ages or something else?

adcrafter27, 2024-08-16:
> What is the reason for having a different CCL for each battery? Is it because of their different ages or something else?

For my case I just want to control the total Max Charge Current of all batteries together.
I have different batteries with different Max Charge Current.

I have
2x JKBMS 15S
and 6x Pylontech 15S 

and I would like to create my own Max Charge Current based on the total battery volts.

e.g
50Volts -> 100A
51Volt -> 70A
52Volt -> 30A
52.5 volts ->0A (in my case 100% full)

This way I can charge my batteries more slowly when they get full

wetty, 2024-08-17:
The charge current is reduced based on the cell voltage. In serialbattery it is the
 --------- Cell Voltage Current limitation (affecting CCL/DCL) ---------
 setting. So if one cell in one of the batteries hits the voltage limit the current gets reduced.
As the two batteries are mostly not loading with exact the same current, an overall limitation of the current to twice the ccl might end up in the worst case, that one battery consume all the current and the other consumes 0A

So if one battery has a current limit the overall current should be reduced until this battery gets just this limited amount. Its not a one time calculation, but more a continous process the regulate the overall current.

As in my example. Battery 2 has a limit of 7.2A (because of the cell voltage). BatteryAggregator sets the overall limit to 14.4A
Battery 1 is only consuming 4A of this 14A, and Battery 2 consumes 10A, which is to high, according to the limit of 7.2A. 
In this example the correct overall limit might be 11A, 4A for Batterie 1 and 7A for Batterie 2.

pulquero, 2024-08-19:
Have you tried the "ir" current ratio method as this is meant to dynamically correct for changes in battery resistance during charging?

pulquero, 2024-08-19:
> For my case I just want to control the total Max Charge Current of all batteries together. I have different batteries with different Max Charge Current.
>
> I have 2x JKBMS 15S and 6x Pylontech 15S [...]

This sounds like you want a custom charge curve, please create a separate issue for this.

wetty, 2024-08-20:
> Have you tried the "ir" current ratio method as this is meant to dynamically correct for changes in battery resistance during charging?

Currently it is running on ir. (was not on my problem). I have to check, but in my opinion the ir mode is not the fix. 

I think it is easier then calculating the internal resistance. I simplified the problem a little bit and set the MAX_BATTERY_CHARGE_CURRENT    =  15.0
to limit the current for the batteries to 15 A. At the moment they are charge to 64% and could consume more than the 15A. The current from the batteries differs a bit. At this SOC just a few Amps.

On the attached screenshot you see that one batterie is loading with nearly 16A and the other with 13.5A
And this is the point. Here the overall current should be reduced that much, that the current for battery 1 in my examples just loads with 15A. 

[image]

pulquero, 2024-08-21:
If the 

> > Have you tried the "ir" current ratio method as this is meant to dynamically correct for changes in battery resistance during charging?
> 
> Currently it is running on ir. (was not on my problem). I have to check, but in my opinion the ir mode is not the fix.

The benefit of IR is that it is a feed-forward mechanism, not feedback. I have had issues in the past trying to implement feedback. The reaction time of the feedback loop is relatively slow because messages are being sent/recvd over dbus, so you can still end up with overcurrent. If IR is close enough then maybe mopping up any residual error with a feedback loop is feasible. I'll have to experiment..

wetty, 2024-08-21:
> > e.g 50Volts -> 100A 51Volt -> 70A 52Volt -> 30A 52.5 volts ->0A (in my case 100% full)
> > This way I can charge my batteries more slowly when they get full
> 
> This sounds like you want a custom charge curve, please create a separate issue for this.

This custom charge curve can be done via serialbattery. Thats what I do and what causes this issue here 😃 

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING      = 3.55, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1

adcrafter27, 2024-08-21:
> This custom charge curve can be done via serialbattery. Thats what I do and what causes this issue here 😃
>
> ; Set steps to reduce battery current ; The current will be changed linear between those [...]

Yes, that's how I do it with my JKBMS, but the Pylontech doesn't run on serial battery. and I would have liked to see a fixed CCL across all batteries (e.g. 20A, 30A, 50A...), not worth a % of BMS

pulquero, 2024-08-24:
@wetty Please try branch "feedback" in the package manger. I've done very limited testing so far, so please be wary in case the feedback cycle goes out of control.

wetty, 2024-08-25:
The feedback version is working perfect. 🙂
Batteries are nearly 100% and current limit was not exceeded.

wetty, 2024-08-26:
First screenshot shows that the second battery is loading with 18.86A. And the second show that the CCL for this battery is set to 19.2A. I will check this the next days.
[image]
[image]

wetty, 2024-09-03:
After several days of testing it seems that the CCL is a little bit to low..
BatteryAggregator limit it to 13.4A
One of the batteries limit ist 12A
This one ist just loading with 9A and the second battery is loading with 4.4A. 
So the overall CCL could be a little bit higher. 
But all in all the feedback version does a good job :-)

pulquero, 2024-09-03:
It is now.

closed 2024-09-03 by pulquero

tzapu, 2025-04-03:
this is off topic, sorry for that, but how do you get that parameters page @wetty ? mine only has CVL CCL and DCL on it

wetty, 2025-04-03:
Hi,
I also use SerialBattery with my jkbms. Maybe this shows the additional parameter.

commit def61341 (2024-08-26): Fixes #65. Add feedback loop to correct IR.
commit 56135187 (2024-08-26): Fixes #65. Add feedback loop to correct IR.