pulquero/BatteryAggregator #67: Agregator logs zero until first battery comes online.

rldoose, 2024-08-25:
Please include system details:
Three JBD serial batteries 

Hardware: Cerbo/RPi???
I have run it on both Cerbo and RPi with same results

Setup: DVCC?
NA

Please provide output from dbus-spy.
NA

Please attach any relevant logs. Log location is `/var/log/BatteryAggregator/current`.
NA

The zero volts is annoying, and just plain wrong.  Agregator should not report until there is something to report. This wacks the scale of my graphs which must now go from zero to 14.2 instead of about 12 to 14.2.  

[image]
[image]

pulquero, 2024-08-25:
Wow, "just plain wrong", harsh. The only thing that is plain wrong is that statement. The aggregator is a battery, a battery not reporting a voltage can cause issues. What I can do is probably delay dbus registration until a valid voltage is available.

rldoose, 2024-08-25:
You are right, I was unnecessarily harsh.  Especially directed at someone providing something for free,  Please accept my apologies.

commit 6793d2a2 (2024-08-26): Fixes #67. Wait for batteries to have voltage before registering on dbus.
closed 2024-08-26 by pulquero