pulquero/BatteryAggregator #70: Battery aggregator includes Shunt on reboot due to USB enumeration

seamaster101, 2024-09-13:
Please include system details:

Hardware: RPi3B running Venus OS 3.5 20

Setup: DVCC Yes

Please provide output from dbus-spy.
*****************************************************
Services
com.victronenergy.adc                                                                                                   com.victronenergy.battery.aggregator                                                                  Battery Aggregatorcom.victronenergy.battery.ttyUSB1                                                                   SmartShunt 500A/50mVcom.victronenergy.battery.ttyUSB5__0                                                                SerialBattery(JKBMS)com.victronenergy.battery.ttyUSB6__0                                                                SerialBattery(JKBMS)com.victronenergy.battery.ttyUSB7__0                                                                SerialBattery(JKBMS)com.victronenergy.battery.ttyUSB8__0                                                                SerialBattery(JKBMS)com.victronenergy.ble
com.victronenergy.fronius
com.victronenergy.generator.startstop0                                                              Generator start/stopcom.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                                                         Package Managercom.victronenergy.platform                                                                                         Venuscom.victronenergy.settings
com.victronenergy.solarcharger.ttyUSB2                                                SmartSolar MPPT VE.Can 150/85 rev2com.victronenergy.system
com.victronenergy.tank.mopeka_dd01922ecfda                                                                 Mopeka sensorcom.victronenergy.tank.mopeka_e31f484e3988                                                                 Mopeka sensorcom.victronenergy.tank.socketcan_can1_vi0_uc828_di1                                                          Tank sensorcom.victronenergy.tank.socketcan_can1_vi1_uc928_di0                                                          Tank sensorcom.victronenergy.temperature.RPi_cpu_id01                                           Raspberry Pi 3 Model B Plus Rev 1.3com.victronenergy.vebus.ttyUSB0                                                       Quattro-II 12/3000/120-2x50 2x120Vcom.victronenergy.vecan.can0
com.victronenergy.vecan.can1
**********************************************
this is my config file:
*************************
{
        "excludedServices": ["com.victronenergy.battery.ttyUSB2"]
}
*************************
the problem I run into is that on every reboot USB re-enumerates and what was the Victron shunt before as specified in my config ttyUSB2, now has changed and you can see Victron shunt now is 500A/50mVcom.victronenergy.battery.ttyUSB5__0. For that reason i cant really tell the Aggregator to exclude the Victron shunt and what end up happening, is it works perfect some times, but most of the times after reboot just excluded one of my batteries and instead includes the shunt, therefore reports nearly twice the current as measured by the shunt.  
[image]

pulquero, 2024-09-14:
There are a few details I dont quite follow here.
"Venus OS 3.5 20"  - the latest version is 3.42
where are USB5-8 coming from? are you using a USB hub?

if the service name is unstable then you need to find something that is stable as an alternative and I can add that to the code. My first suggestion would be to check if the device instance ID changes.

seamaster101, 2024-09-14:
"Venus OS 3.5 20"  was the latest beta to which I upgraded few weeks ago, but the behavior of my system is exactly the same as when using the release version before that. The issue did not start when I upgraded.  
 I'm using 4 Batteries with JK BMSs. these are communicating via RS485 to USB, I have a USB hub connected to the RPI for that .
I also have 4ch ve.direct to USB [https://www.fraron.de/en/battery-accessories/battery-monitor/victron-ve-direct-interface-1-x-usb-to-4-x-ve-direct/a-85938332/](url) where my smart shunt, my solar controller, my small inverter are connected. 
On restart the names of the devices are consistent every time, but the USB enumeration is not the same. In the config file for the aggregator one can only specify to ignore USB# but if that # is different the next time the system restart, the aggregator  ignores some other device that took the number of the USB port of the shunt, which was specified to be ignored in the config file initially. 
I hope that explains the issue better

pulquero, 2024-09-15:
Ah, it looks like it might be something to do with the 4-channel adapter then. For educational purposes, you could try connecting a single device to it to see if each channel has a fixed designation or whether the first USB# just goes to the first device.

OK, I can see if I can add support to use assigned names as an alternative to service names.

seamaster101, 2024-09-15:
I’m traveling at the moment with no physical access to do the test but I can do that on Friday when I return. 
min a meantime, I have remote access so if you implement assigned names I would be able to test that right away remotely

pulquero, 2024-09-16:
Try v3.3 in main. Instead of using service names in config, you can use custom names by prepending with "name:".  e.g. "name:My Battery" instead of "com.victronenergy.battery.ttyS7".

seamaster101, 2024-09-17:
I will try and report. Thank you!Jordan

seamaster101, 2024-09-17:
I can only download V3.2 in Package editor. How do I install v3.3?

endurance1968, 2024-09-17:
Just to clarify with servicenames you mean - the name visible in dbus-spy or...?
[image]

seamaster101, 2024-09-28:
Hi,
I upgraded to V3.7 then I edited the cofig.ini file 
I made it contain so it only has one line:
{
"excludedServices":["name:Bank1_Shunt"]
}
the name of the device I want to exclude is Bank1_Shunt
I did check validity of the config file by running 
root@raspberrypi2:~# bash /data/BatteryAggregator/check_config.sh
SUCCESS: The Config file is valid

I tried restarting the RPI or restarting just the service by 
svc -d /service/BatteryAggregator 
and then 
svc -u /service/BatteryAggregator
no error messages but the server don't start
Where im making a mistake?

pulquero, 2024-09-28:
silly mistake, fixed in v3.8.

seamaster101, 2024-09-29:
> silly mistake, fixed in v3.8.
@pulquero 
upgraded to v3.8
still not working
********************
root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066f8c06e137e70bc dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: org.freedesktop.DBus does not understand message GetValue
@4000000066f8c0710f6f3d94 INFO:main:Starting...
@4000000066f8c0711521e78c Traceback (most recent call last):
@4000000066f8c071152202e4   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f8c0711522166c     main()
@4000000066f8c07115221a54   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f8c07115222ddc     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f8c07115224164   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f8c07115244504     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f8c0711524588c   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f8c0711524aa94     service_name = self._get_custom_names().get(name)
@4000000066f8c0711524ba34   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f8c07115252b7c     custom_name = self.conn.call_blocking(service_name, "/CustomName", None, "GetValue", "", [])
@4000000066f8c071152542ec   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f8c07115255674     reply_message = self.send_message_with_reply_and_block(
@4000000066f8c0711525622c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: org.freedesktop.DBus does not understand message GetValue
@4000000066f8c0730aec70ac INFO:main:Starting...
@4000000066f8c0730d888404 Traceback (most recent call last):
@4000000066f8c0730d889b74   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f8c0730d88ab14     main()
@4000000066f8c0730d88b2e4   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f8c0730d88c284     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f8c0730d88d9f4   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f8c0730d8abe54     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f8c0730d8ad1dc   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f8c0730d8ae564     service_name = self._get_custom_names().get(name)
@4000000066f8c0730d8af11c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f8c0730d8d90fc     custom_name = self.conn.call_blocking(service_name, "/CustomName", None, "GetValue", "", [])
@4000000066f8c0730d8da86c   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f8c0730d8dbbf4     reply_message = self.send_message_with_reply_and_block(
@4000000066f8c0730d8dcb94 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: org.freedesktop.DBus does not understand message GetValue
@4000000066f8c0743aa603e4 INFO:main:Starting...
@4000000066f8c07500e5b2bc Traceback (most recent call last):
@4000000066f8c07500e5c644   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f8c07500e5d9cc     main()
@4000000066f8c07500e5e19c   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f8c07500e5f13c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f8c07500e604c4   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f8c07500e7dd6c     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f8c07500e7f0f4   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f8c07500e8047c     service_name = self._get_custom_names().get(name)
@4000000066f8c07500e8141c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f8c07500e8817c     custom_name = self.conn.call_blocking(service_name, "/CustomName", None, "GetValue", "", [])
@4000000066f8c07500e898ec   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f8c07500e8ac74     reply_message = self.send_message_with_reply_and_block(
@4000000066f8c07500e8b82c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: org.freedesktop.DBus does not understand message GetValue
@4000000066f8c076204a3e34 INFO:main:Starting...
@4000000066f8c0762321e774 Traceback (most recent call last):
@4000000066f8c0762321fee4   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f8c0762322126c     main()
@4000000066f8c07623221654   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f8c076232229dc     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f8c07623223d64   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f8c07623240e3c     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f8c076232425ac   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
[... cut ...]
@4000000066f8c0822e8f45fc     custom_name = self.conn.call_blocking(service_name, "/CustomName", None, "GetValue", "", [])
@4000000066f8c0822e8f5d6c   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f8c0822e8f70f4     reply_message = self.send_message_with_reply_and_block(
@4000000066f8c0822e8f8094 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: org.freedesktop.DBus does not understand message GetValue
*****************************************************************

pulquero, 2024-09-29:
This is a bit more challenging. I guess it might be related to Venus 3.5. I've included a change that Victron has recently made to their python stuff. See if that works. If not, you may need to downgrade so we can eliminate that as a cause.

seamaster101, 2024-09-29:
@pulquero unfortunately did not work. 
here is my config and the name of my shunt as it appears  in the Victron. I want to confirm all my settings are correct before I downgrade.
Config:
***************
{
"excludedServices":["name:Bank1_Shunt"]
}
***************
 
[image]

and the log:
root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066f99ba5084b67cc INFO:main:Starting...
@4000000066f99ba50a961f04 Traceback (most recent call last):
@4000000066f99ba50a963674   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f99ba50a9649fc     main()
@4000000066f99ba50a964de4   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f99ba50a96616c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f99ba50a9674f4   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f99ba50a97ff7c     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f99ba50a981304   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f99ba50a98268c     service_name = self._get_custom_names().get(name)
@4000000066f99ba50a98362c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f99ba50a98a774     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f99ba50a98bee4   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f99ba50a98d26c     reply_message = self.send_message_with_reply_and_block(
@4000000066f99ba50a98de24 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f99ba6311e45ac INFO:main:Starting...
@4000000066f99ba7043f0f1c Traceback (most recent call last):
@4000000066f99ba7043f2e5c   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f99ba7043f41e4     main()
@4000000066f99ba7043f45cc   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f99ba7043f556c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f99ba7043f6cdc   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f99ba704410aec     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f99ba70441225c   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f99ba7044135e4     service_name = self._get_custom_names().get(name)
@4000000066f99ba70441419c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f99ba70441aefc     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f99ba70441c66c   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f99ba70441d9f4     reply_message = self.send_message_with_reply_and_block(
@4000000066f99ba70441e994 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f99ba906cf3b94 INFO:main:Starting...
@4000000066f99ba908c1dfec Traceback (most recent call last):
@4000000066f99ba908c1f75c   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f99ba908c206fc     main()
@4000000066f99ba908c20ecc   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f99ba908c21e6c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f99ba908c231f4   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f99ba908c3acdc     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f99ba908c3c064   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f99ba908c3d3ec     service_name = self._get_custom_names().get(name)
@4000000066f99ba908c3e38c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f99ba908c450ec     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f99ba908c4685c   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f99ba908c47be4     reply_message = self.send_message_with_reply_and_block(
@4000000066f99ba908c48b84 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f99bab130e4db4 INFO:main:Starting...
@4000000066f99bab15b1852c Traceback (most recent call last):
@4000000066f99bab15b1a084   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f99bab15b1b024     main()
@4000000066f99bab15b1b7f4   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f99bab15b1c794     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f99bab15b1db1c   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f99bab15b33aac     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
[... cut ...]
@4000000066f99bab15b3715c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f99bab15b3dad4     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f99bab15b3f244   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f99bab15b405cc     reply_message = self.send_message_with_reply_and_block(
@4000000066f99bab15b4156c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue

seamaster101, 2024-09-29:
@pulquero 
I downgraded Venus OS to the latest release V3.42
No luck. here is the log:

root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066f9b9552cf96b8c INFO:main:Starting...
@4000000066f9b9552f64c63c Traceback (most recent call last):
@4000000066f9b9552f64e57c   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f9b9552f64f904     main()
@4000000066f9b9552f64fcec   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f9b9552f651074     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f9b9552f6523fc   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f9b9552f67855c     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f9b9552f6798e4   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f9b9552f67ac6c     service_name = self._get_custom_names().get(name)
@4000000066f9b9552f67bc0c   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f9b9552f682d54     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f9b9552f6844c4   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f9b9552f68584c     reply_message = self.send_message_with_reply_and_block(
@4000000066f9b9552f6867ec dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f9b9580204bf1c INFO:main:Starting...
@4000000066f9b9580533f284 Traceback (most recent call last):
@4000000066f9b958053409f4   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f9b95805341d7c     main()
@4000000066f9b95805342164   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f9b958053434ec     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f9b95805344874   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f9b9580535c744     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f9b9580535dacc   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f9b9580535ee54     service_name = self._get_custom_names().get(name)
@4000000066f9b9580535fdf4   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f9b9580536676c     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f9b95805367edc   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f9b95805369264     reply_message = self.send_message_with_reply_and_block(
@4000000066f9b9580536a204 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f9b9593713b564 INFO:main:Starting...
@4000000066f9b959393ac3b4 Traceback (most recent call last):
@4000000066f9b959393adb24   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f9b959393aeac4     main()
@4000000066f9b959393af294   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f9b959393b0234     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f9b959393b15bc   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f9b959393c9c5c     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f9b959393cafe4   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f9b959393cc36c     service_name = self._get_custom_names().get(name)
@4000000066f9b959393ccf24   File "/data/BatteryAggregator/battery_service.py", line 250, in _get_custom_names
@4000000066f9b959393d406c     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f9b959393d53f4   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f9b959393d677c     reply_message = self.send_message_with_reply_and_block(
@4000000066f9b959393d771c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
@4000000066f9b95b39b20ec4 INFO:main:Starting...
@4000000066f9b95c01235fcc Traceback (most recent call last):
@4000000066f9b95c01237b24   File "/data/BatteryAggregator/battery_service.py", line 972, in <module>
@4000000066f9b95c01238ac4     main()
@4000000066f9b95c01239294   File "/data/BatteryAggregator/battery_service.py", line 944, in main
@4000000066f9b95c0123a234     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066f9b95c0123b5bc   File "/data/BatteryAggregator/battery_service.py", line 361, in __init__
@4000000066f9b95c01255f84     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066f9b95c012576f4   File "/data/BatteryAggregator/battery_service.py", line 256, in resolve_service_name
@4000000066f9b95c01258a7c     service_name = self._get_custom_names().get(name)
[... cut ...]
@4000000066f9b95c0126077c     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066f9b95c01261eec   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066f9b95c01263274     reply_message = self.send_message_with_reply_and_block(
@4000000066f9b95c01264214 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownInterface: org.freedesktop.DBus does not understand message GetValue
root@raspberrypi2:~#

pulquero, 2024-09-29:
I think I might have it this time...

seamaster101, 2024-09-30:
@pulquero 
hi, I hate to tell, you but the service did not start again. 
Let me know if you want me give you remote access to see whyat is going on.
here is the log:
 cat /var/log/BatteryAggregator/current
@4000000066fb16a2281d1c94 INFO:main:Starting...
@4000000066fb16a22a29e83c Traceback (most recent call last):
@4000000066fb16a22a2a0394   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fb16a22a2a171c     main()
@4000000066fb16a22a2a1b04   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fb16a22a2a2e8c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fb16a22a2a4214   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fb16a22a2bf3ac     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fb16a22a2c0b1c   File "/data/BatteryAggregator/service_name_resolver.py", line 30, in resolve_service_name
@4000000066fb16a22a2c1ea4     service_name = self._get_custom_names().get(name)
@4000000066fb16a22a2c2e44   File "/data/BatteryAggregator/service_name_resolver.py", line 24, in _get_custom_names
@4000000066fb16a22a2c9f8c     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066fb16a22a2cb6fc   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066fb16a22a2cce6c     reply_message = self.send_message_with_reply_and_block(
@4000000066fb16a22a2cda24 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetValue" with signature "" on interface "com.victronenergy.BusItem" doesn't exist
@4000000066fb16a22a2dc09c
@4000000066fb16a429d3b5d4 INFO:main:Starting...
@4000000066fb16a50ea62cec Traceback (most recent call last):
@4000000066fb16a50ea64844   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fb16a50ea65bcc     main()
@4000000066fb16a50ea65fb4   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fb16a50ea6733c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fb16a50ea6c544   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fb16a50ea8d884     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fb16a50ea8ec0c   File "/data/BatteryAggregator/service_name_resolver.py", line 30, in resolve_service_name
@4000000066fb16a50ea8ff94     service_name = self._get_custom_names().get(name)
@4000000066fb16a50ea90f34   File "/data/BatteryAggregator/service_name_resolver.py", line 24, in _get_custom_names
@4000000066fb16a50ea97c94     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066fb16a50ea99404   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066fb16a50ea9a78c     reply_message = self.send_message_with_reply_and_block(
@4000000066fb16a50ea9b72c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetValue" with signature "" on interface "com.victronenergy.BusItem" doesn't exist
@4000000066fb16a50eaa99bc
@4000000066fb16a730986c0c INFO:main:Starting...
@4000000066fb16a73818bd9c Traceback (most recent call last):
@4000000066fb16a73818d50c   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fb16a73818e894     main()
@4000000066fb16a73818ec7c   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fb16a738190004     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fb16a73819138c   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fb16a7381afbd4     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fb16a7381b1344   File "/data/BatteryAggregator/service_name_resolver.py", line 30, in resolve_service_name
@4000000066fb16a7381b26cc     service_name = self._get_custom_names().get(name)
@4000000066fb16a7381b3284   File "/data/BatteryAggregator/service_name_resolver.py", line 24, in _get_custom_names
@4000000066fb16a7381ba7b4     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066fb16a7381bbf24   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066fb16a7381bd2ac     reply_message = self.send_message_with_reply_and_block(
@4000000066fb16a7381be24c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetValue" with signature "" on interface "com.victronenergy.BusItem" doesn't exist
@4000000066fb16a7381dda34
@4000000066fb16aa0923225c INFO:main:Starting...
@4000000066fb16aa16a8b66c Traceback (most recent call last):
@4000000066fb16aa16a8d5ac   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fb16aa16a8e54c     main()
@4000000066fb16aa16a8ed1c   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fb16aa16a8fcbc     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fb16aa16a91044   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fb16aa16aaf4a4     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
[... cut ...]
@4000000066fb16aa16ab98b4     custom_name = self.conn.call_blocking(service_name, "/CustomName", VE_INTERFACE, "GetValue", "", [])
@4000000066fb16aa16abb024   File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 666, in call_blocking
@4000000066fb16aa16abc3ac     reply_message = self.send_message_with_reply_and_block(
@4000000066fb16aa16abd34c dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetValue" with signature "" on interface "com.victronenergy.BusItem" doesn't exist
@4000000066fb16aa16acaa24
root@raspberrypi2:~#

pulquero, 2024-10-01:
It was either going to work or do this. And if it does this then I do...v3.11.

seamaster101, 2024-10-01:
@pulquero 
hi thank you for keep trying..V3.11 service still failed, but this time at least we have more solid hint. I don't know why, but it reports  "ValueError: Unknown name: name:Bank1_Shunt"/ See below
root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066fc66bb112d1a34   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fc66bb1204e464     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fc66bb120503a4   File "/data/BatteryAggregator/service_name_resolver.py", line 35, in resolve_service_name
@4000000066fc66bb1205172c     raise ValueError(f"Unknown name: {name}")
@4000000066fc66bb120526cc ValueError: Unknown name: name:Bank1_Shunt
@4000000066fc66bd38f75dcc INFO:main:Starting...
@4000000066fc66be0e25e12c Traceback (most recent call last):
@4000000066fc66be0e25f89c   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fc66be0e26083c     main()
@4000000066fc66be0e26100c   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fc66be0e261fac     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fc66be0e263334   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fc66be0e277b54     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fc66be0e278edc   File "/data/BatteryAggregator/service_name_resolver.py", line 35, in resolve_service_name
@4000000066fc66be0e27a64c     raise ValueError(f"Unknown name: {name}")
@4000000066fc66be0e27b204 ValueError: Unknown name: name:Bank1_Shunt
@4000000066fc66c0295c285c INFO:main:Starting...
@4000000066fc66c1092c4e04 Traceback (most recent call last):
@4000000066fc66c1092c6574   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fc66c1092c78fc     main()
@4000000066fc66c1092c7ce4   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fc66c1092c906c     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fc66c1092ca3f4   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fc66c1092e8084     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fc66c1092e940c   File "/data/BatteryAggregator/service_name_resolver.py", line 35, in resolve_service_name
@4000000066fc66c1092ea794     raise ValueError(f"Unknown name: {name}")
@4000000066fc66c1092eb34c ValueError: Unknown name: name:Bank1_Shunt
@4000000066fc66c236efd1e4 INFO:main:Starting...
@4000000066fc66c30a8c012c Traceback (most recent call last):
@4000000066fc66c30a8c1c84   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fc66c30a8c300c     main()
@4000000066fc66c30a8c33f4   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fc66c30a8c4394     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fc66c30a8c5b04   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fc66c30a8de974     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fc66c30a8dfcfc   File "/data/BatteryAggregator/service_name_resolver.py", line 35, in resolve_service_name
@4000000066fc66c30a8e1084     raise ValueError(f"Unknown name: {name}")
@4000000066fc66c30a8e1c3c ValueError: Unknown name: name:Bank1_Shunt
@4000000066fc66c432caf3dc INFO:main:Starting...
@4000000066fc66c439b74e84 Traceback (most recent call last):
@4000000066fc66c439b76dc4   File "/data/BatteryAggregator/battery_service.py", line 929, in <module>
@4000000066fc66c439b77d64     main()
@4000000066fc66c439b78534   File "/data/BatteryAggregator/battery_service.py", line 901, in main
@4000000066fc66c439b794d4     batteryAggr = BatteryAggregatorService(dbusConnection(), DEFAULT_SERVICE_NAME, config)
@4000000066fc66c439b7a85c   File "/data/BatteryAggregator/battery_service.py", line 318, in __init__
@4000000066fc66c439b988d4     excludedServices.append(serviceNameResolver.resolve_service_name(service_name))
@4000000066fc66c439b9a044   File "/data/BatteryAggregator/service_name_resolver.py", line 35, in resolve_service_name
@4000000066fc66c439b9b3cc     raise ValueError(f"Unknown name: {name}")
@4000000066fc66c439b9bf84 ValueError: Unknown name: name:Bank1_Shunt

pulquero, 2024-10-01:
Stupid error, forgot to strip off the "name:" prefix. v3.12.

seamaster101, 2024-10-02:
@pulquero 
Maybe V3.13 will be the one. 
still no work with 3.12
here is the log:

cat /var/log/BatteryAggregator/current
@4000000066fcbb942c1b197c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942cd74174     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942cd75ccc TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942cd77054 ERROR:dbusmonitor:Ignoring com.victronenergy.modbustcp because of error while scanning:
@4000000066fcbb942cde9c44 Traceback (most recent call last):
@4000000066fcbb942cdeb79c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942cded6dc     return self.scan_dbus_service_inner(serviceName)
@4000000066fcbb942cdee294   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in scan_dbus_service_inner
@4000000066fcbb942cdef61c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942cdf0d8c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942cdfe84c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942cdfffbc TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942ce00f5c ERROR:dbusmonitor:Ignoring :1.4 because of error while scanning:
@4000000066fcbb942ce01efc Traceback (most recent call last):
@4000000066fcbb942ce02e9c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942ce0bb3c     return self.scan_dbus_service_inner(serviceName)
@4000000066fcbb942ce0cadc   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in scan_dbus_service_inner
@4000000066fcbb942ce0e24c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce1c0f4   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942ce1d47c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce1ebec TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942ce1fb8c ERROR:dbusmonitor:Ignoring :1.39 because of error while scanning:
@4000000066fcbb942ce268ec Traceback (most recent call last):
@4000000066fcbb942ce274a4   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942ce2882c     return self.scan_dbus_service_inner(serviceName)
@4000000066fcbb942ce293e4   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in scan_dbus_service_inner
@4000000066fcbb942ce2a76c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce310e4   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942ce3246c     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce337f4 TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942ce34b7c ERROR:dbusmonitor:Ignoring :1.104 because of error while scanning:
@4000000066fcbb942ce3fb44 Traceback (most recent call last):
@4000000066fcbb942ce406fc   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942ce4169c     return self.scan_dbus_service_inner(serviceName)
@4000000066fcbb942ce4263c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in scan_dbus_service_inner
@4000000066fcbb942ce43dac     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce4a724   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942ce4baac     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce4d21c TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942ce4e98c ERROR:dbusmonitor:Ignoring :1.5 because of error while scanning:
@4000000066fcbb942ce562a4 Traceback (most recent call last):
@4000000066fcbb942ce56e5c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942ce581e4     return self.scan_dbus_service_inner(serviceName)
@4000000066fcbb942ce58d9c   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in scan_dbus_service_inner
@4000000066fcbb942ce5a124     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce76644   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 205, in <genexpr>
@4000000066fcbb942ce779cc     if (len(self.ignoreServices) != 0 and any(serviceName.startswith(x) for x in self.ignoreServices)):
@4000000066fcbb942ce7913c TypeError: startswith first arg must be str or a tuple of str, not NoneType
@4000000066fcbb942ce7a0dc ERROR:dbusmonitor:Ignoring com.victronenergy.battery.ttyUSB1 because of error while scanning:
@4000000066fcbb942ce81224 Traceback (most recent call last):
@4000000066fcbb942ce81ddc   File "/data/BatteryAggregator/ext/dbusmonitor.py", line 183, in scan_dbus_service
@4000000066fcbb942ce83164     return self.scan_dbus_service_inner(serviceName)
[... cut ...]
@4000000066fcbba718a5a394 INFO:main:Waiting for batteries (attempt 18 of 30)...
@4000000066fcbba8189c972c INFO:main:Waiting for batteries (attempt 19 of 30)...
@4000000066fcbba9189e0e2c INFO:main:Waiting for batteries (attempt 20 of 30)...
@4000000066fcbbaa189f3af4 INFO:main:Waiting for batteries (attempt 21 of 30)...
@4000000066fcbbab189f5264 INFO:main:Waiting for batteries (attempt 22 of 30)...
@4000000066fcbbac18a3b37c INFO:main:Waiting for batteries (attempt 23 of 30)...
@4000000066fcbbad18997664 INFO:main:Waiting for batteries (attempt 24 of 30)...

pulquero, 2024-10-02:
try the latest one

seamaster101, 2024-10-03:
@pulquero 
V3.13 was the "charm"! it works as expected now. The service started with no issues
Thank you for the feature and perseverance ! 
I think this one could be closed now.

commit d0bcc313 (2024-09-16): Fixes #70.
commit c6a7c35d (2024-09-20): Fixes #70.
commit ab0091ff (2024-09-27): Fixes #70.
commit 0cb82547 (2024-09-28): Fixes #70.
closed 2024-10-03 by pulquero