pulquero/BatteryAggregator #71: Cosntant crash using 3.2 version - module not found

endurance1968, 2024-09-16:
Just installed 3.2 version but it is constantly crashing due to module not found

Hardware: Cerbo 3.41
BA 3.2  

Traceback (most recent call last):
@4000000066e7bc04363d0474   File "/data/BatteryAggregator/battery_service.py", line 23, in <module>
@4000000066e7bc04363d1414     from pydoc import locate

endurance1968, 2024-09-16:
Hmm - i installed the packes needed for config check and reinstalled version 3.2 and now it works again.
NMot sure if the error was fixed by installing the extra packages or by reinstalling v3.2

SlapJackNpNp, 2024-09-16:
I have two systems running battery aggregator and the modul is not found on either. What exactly did you do to fix this, which packages are you talkin about?

SlapJackNpNp, 2024-09-16:
I uninstalled 3.2, rebooted the cerbo and installed it again but it still doesn't work. The Pacakges also don't seem to be installable via package manager? This is unfortunate since i acces the systems remotely.

pulquero, 2024-09-16:
Try downloading and installing again via package manager. It seems it didnt download a newer version of the VELib library. Also look in the log for a line "++ Install Python library velib_python" and there were no errors.

erhankeseli, 2024-09-16:
I get another error:
Traceback (most recent call last):
@4000000066e7f6862cf97f14   File "/data/BatteryAggregator/battery_service.py", line 23, in <module>
@4000000066e7f6862cf98acc     from pydoc import locate
@4000000066e7f6862cf9929c ModuleNotFoundError: No module named 'pydoc'

pulquero, 2024-09-16:
> I get another error: Traceback (most recent call last): @4000000066e7f6862cf97f14 File "/data/BatteryAggregator/battery_service.py", line 23, in @4000000066e7f6862cf98acc from pydoc import locate @4000000066e7f6862cf9929c ModuleNotFoundError: No module named 'pydoc'

what version of venus are you running?

erhankeseli, 2024-09-16:
3.10-22

SlapJackNpNp, 2024-09-16:
Running bash /data/BatteryAggregator/check_config.sh and installing the two packages helped to get it running again. But if possible it would be great if the files would be included in the packagemaanger files

pulquero, 2024-09-16:
> 3.10-22

Try upgrading to 3.42.

Feil1, 2024-09-16:
i had the same problem, i also uninstalled, restartet, installed new nothing helped,
but bash /data/BatteryAggregator/check_config.sh    has helped and now its working again.

SlapJackNpNp, 2024-09-17:
I updated my cerbo to the latest v3.50~22 and the Aggregator stopped again. To get it working, i had to run the bash /data/BatteryAggregator/check_config.sh command again and install the packages to get it running. uninstalling and redownloading with package manager didn't help. Maybe it is important to notice that i don't use the venus-os large image but the regular one?

pulquero, 2024-09-17:
Ah maybe. I only test on large. Please paste log with install output.

SlapJackNpNp, 2024-09-17:
Uh sorry, but could you quickly explain what to put into the ssh to get the logs you need and in which condition? Atm i got everything running again. Can i still get the log you need?

PatrickGoettsch, 2024-09-18:
If you have root access via ssh
You just copy and paste this

bash /data/BatteryAggregator/check_config.sh

endurance1968, 2024-09-26:
just upgraded to latest Venus release - some issue. On the normal venus (not large)version you definitely have to install extra packages.
Simplest is to run as suggest the config checker - but this can only be a workaround.

SlapJackNpNp, 2024-09-27:
[image]
This is what i got with cat /var/log/BatteryAggregator/current
The Battery Aggregator is stopping every time i do a software update and only works after the config check command. I only use the regular venus-os and not the large one.

pulquero, 2024-09-27:
Ahhh, I know what needs to be done now...

commit 6b64dd4d (2024-09-27): Fixes #71.
commit 71450d54 (2024-09-28): Fixes #71.
closed 2024-09-27 by pulquero