pulquero/BatteryAggregator #72: CCL is not multiplied with number of parallel batteries

MarenB, 2024-09-22:
Please include system details:

Hardware: 
- 2x 240Ah Battery connected in parallel to Multiplus II
- Venus on RPi 3B+
- 2x JK BMS connected to the Pi via dbus-serialbattery driver

Setup: DVCC

While i.e. the total capacity is calculated correctly, the CCL is applied as if there was one battery only or if they were in series. I'd expect a total CCL of 140A, if each battery is set to 70A. 

The BMS setup:
[image]
[image]

We can see, the cell voltages are in a range, where CCL should be reduced to ~50-60%, which should still result in ~55A for both parallel batteries. However, it's limited to 27A already.

[image]
[image]
[image]

MarenB, 2024-09-22:
Ok, I feel stupid now. I guess, this was related to the fact I didn't provide a config-file with the setting "count"? 
Created that file now, will return later, if it didn't help.

pulquero, 2024-09-22:
Need to see the log to know what is going on, but CCL can be throttled back if it detects the CCL of an individual battery being exceeded. In this case, the CCL will be limited by the weaker battery. With 'count' mode, both batteries are assumed to be equally strong.

MarenB, 2024-09-26:
I think I must re-open this. Strangely, I sometimes get a CCL of 70A (which is actually per battery) and sometimes 100A.
Today I changed the logging level to DEBUG and restarted the Aggregator and after that it switched from 70A to 100A again.

This is the log-file:
`root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066f579a83142968c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.683
@4000000066f579a831503ea4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.683} now has values {'com.victronenergy.battery.ttyUSB2': 3.684, 'com.victronenergy.battery.ttyUSB1': 3.683} with result 3.684
@4000000066f579a9157c639c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Current 5.14
@4000000066f579a9157c8e94 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB2: 5.14} now has values {'com.victronenergy.battery.ttyUSB2': 5.14, 'com.victronenergy.battery.ttyUSB1': 5.54} with result 10.68
@4000000066f579a915a465cc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Power 113.49
@4000000066f579a915b52ac4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB2: 113.49} now has values {'com.victronenergy.battery.ttyUSB2': 113.49, 'com.victronenergy.battery.ttyUSB1': 122.32} with result 235.81
@4000000066f579a931466b04 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Current 5.04
@4000000066f579a9315883d4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB1: 5.04} now has values {'com.victronenergy.battery.ttyUSB2': 5.14, 'com.victronenergy.battery.ttyUSB1': 5.04} with result 10.18
@4000000066f579a932ffabf4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Power 111.28
@4000000066f579a9330a23a4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB1: 111.28} now has values {'com.victronenergy.battery.ttyUSB2': 113.49, 'com.victronenergy.battery.ttyUSB1': 111.28} with result 224.76999999999998
@4000000066f579a93320479c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.684
@4000000066f579a9332a7514 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.684} now has values {'com.victronenergy.battery.ttyUSB2': 3.684, 'com.victronenergy.battery.ttyUSB1': 3.684} with result 3.684
@4000000066f579ab16001a5c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Current 5.05
@4000000066f579ab1613f07c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB2: 5.05} now has values {'com.victronenergy.battery.ttyUSB2': 5.05, 'com.victronenergy.battery.ttyUSB1': 5.04} with result 10.09
@4000000066f579ab164d8e7c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Power 111.5
@4000000066f579ab165d558c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB2: 111.5} now has values {'com.victronenergy.battery.ttyUSB2': 111.5, 'com.victronenergy.battery.ttyUSB1': 111.28} with result 222.78
@4000000066f579ab31bd21a4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Current 5.12
@4000000066f579ab31bd48b4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB1: 5.12} now has values {'com.victronenergy.battery.ttyUSB2': 5.05, 'com.victronenergy.battery.ttyUSB1': 5.12} with result 10.17
@4000000066f579ab31dbbd94 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Power 113.05
@4000000066f579ab31eeb8f4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB1: 113.05} now has values {'com.victronenergy.battery.ttyUSB2': 111.5, 'com.victronenergy.battery.ttyUSB1': 113.05} with result 224.55
@4000000066f579ab323efbd4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.686
@4000000066f579ab324efd7c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.686} now has values {'com.victronenergy.battery.ttyUSB2': 3.684, 'com.victronenergy.battery.ttyUSB1': 3.686} with result 3.686
@4000000066f579ac3194dd0c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.684
@4000000066f579ac319b5164 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.684} now has values {'com.victronenergy.battery.ttyUSB2': 3.684, 'com.victronenergy.battery.ttyUSB1': 3.684} with result 3.684
[... cut ...]
@4000000066f5354f021b7784 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {dbus.String('com.victronenergy.battery.ttyUSB2'): 100.0, dbus.String('com.victronenergy.battery.ttyUSB1'): 100.0}
@4000000066f5354f02273b3c INFO:com.victronenergy.battery.aggregator:Connected batteries: [dbus.String('com.victronenergy.battery.ttyUSB2'), dbus.String('com.victronenergy.battery.ttyUSB1')]
@4000000066f5354f026b879c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.0000000000000033, 'ir'), (305868322391773.75, 'ir')]
@4000000066f5354f026baeac INFO:com.victronenergy.battery.aggregator:DCL estimates: [100.00000000000033, 3.0586832239177376e+16]
`

MarenB, 2024-09-26:
I don't get it. Now it's working. After I changed logging level to INFO, re-started the service once again and voila!

`root@raspberrypi2:~# cat /var/log/BatteryAggregator/current
@4000000066f57b8120fbe6ac DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Current 5.37
@4000000066f57b8120fc11a4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB1: 5.37} now has values {'com.victronenergy.battery.ttyUSB2': 5.47, 'com.victronenergy.battery.ttyUSB1': 5.37} with result 10.84
@4000000066f57b8120fc4084 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /Dc/0/Power 118.73
@4000000066f57b8121145494 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB1: 118.73} now has values {'com.victronenergy.battery.ttyUSB2': 120.89, 'com.victronenergy.battery.ttyUSB1': 118.73} with result 239.62
@4000000066f57b812158db8c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MinVoltageCellId C3
@4000000066f57b812165108c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinVoltageCellId updated with {com.victronenergy.battery.ttyUSB1: C3} now has values {'com.victronenergy.battery.ttyUSB2': 'C3', 'com.victronenergy.battery.ttyUSB1': 'C3'} with result C3
@4000000066f57b81218b74ac DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxVoltageCellId C1
@4000000066f57b8121a25bf4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxVoltageCellId updated with {com.victronenergy.battery.ttyUSB1: C1} now has values {'com.victronenergy.battery.ttyUSB2': 'C5', 'com.victronenergy.battery.ttyUSB1': 'C1'} with result C5
@4000000066f57b81222a3934 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MinCellVoltage 3.683
@4000000066f57b81223c55ec DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.683} now has values {'com.victronenergy.battery.ttyUSB2': 3.68, 'com.victronenergy.battery.ttyUSB1': 3.683} with result 3.68
@4000000066f57b81225c034c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.686
@4000000066f57b81226eeb24 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.686} now has values {'com.victronenergy.battery.ttyUSB2': 3.686, 'com.victronenergy.battery.ttyUSB1': 3.686} with result 3.686
@4000000066f57b82044f62cc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /System/MaxCellVoltage 3.687
@4000000066f57b8204640024 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB2: 3.687} now has values {'com.victronenergy.battery.ttyUSB2': 3.687, 'com.victronenergy.battery.ttyUSB1': 3.686} with result 3.687
@4000000066f57b822033a8f4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MinVoltageCellId C6
@4000000066f57b822040cc3c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinVoltageCellId updated with {com.victronenergy.battery.ttyUSB1: C6} now has values {'com.victronenergy.battery.ttyUSB2': 'C3', 'com.victronenergy.battery.ttyUSB1': 'C6'} with result C3
@4000000066f57b822074ff84 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MinCellVoltage 3.682
@4000000066f57b8220828474 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MinCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.682} now has values {'com.victronenergy.battery.ttyUSB2': 3.68, 'com.victronenergy.battery.ttyUSB1': 3.682} with result 3.68
@4000000066f57b8220b6522c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB1 /System/MaxCellVoltage 3.687
@4000000066f57b8220cadfe4 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /System/MaxCellVoltage updated with {com.victronenergy.battery.ttyUSB1: 3.687} now has values {'com.victronenergy.battery.ttyUSB2': 3.687, 'com.victronenergy.battery.ttyUSB1': 3.687} with result 3.687
@4000000066f57b83046f661c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Current 5.81
@4000000066f57b83046f8d2c DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Current updated with {com.victronenergy.battery.ttyUSB2: 5.81} now has values {'com.victronenergy.battery.ttyUSB2': 5.81, 'com.victronenergy.battery.ttyUSB1': 5.37} with result 11.18
@4000000066f57b8304cf412c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB2 /Dc/0/Power 128.4
@4000000066f57b8304da9f54 DEBUG:com.victronenergy.battery.aggregator:Aggregator for /Dc/0/Power updated with {com.victronenergy.battery.ttyUSB2: 128.4} now has values {'com.victronenergy.battery.ttyUSB2': 128.4, 'com.victronenergy.battery.ttyUSB1': 118.73} with result 247.13
[... cut ...]
@4000000066f57baa29c4fc4c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 70.0, 'com.victronenergy.battery.ttyUSB1': 70.0}
@4000000066f57baa29d10654 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB1']
@4000000066f57baa29dcc624 INFO:com.victronenergy.battery.aggregator:Current ratios: [(2, 'count'), (2, 'count')]
@4000000066f57baa29e69dac INFO:com.victronenergy.battery.aggregator:CCL estimates: [140.0, 140.0]
@4000000066f57baa2a0c752c INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator
`

pulquero, 2024-09-27:
Some of the ir estimates are definitely dodgy. I've improved how the calculation is performed, try v3.6.

pulquero, 2024-09-30:
v3.10 should be fine.

closed 2024-10-12 by pulquero