pulquero/BatteryAggregator #74: missing pydoc on venosos >3.41 on rpi

tazinblack, 2024-09-28:
BatteryAggregator stops working after venusos update to > 3.41 (i think, sorry for not checking version before update)

Hardware: RPi

Setup: DVCC 

Please provide output from dbus-spy:

`com.victronenergy.adc                                                                                         
com.victronenergy.battery.ttyACM0                                                         SerialBattery(Jkbms)
com.victronenergy.battery.ttyUSB0                                                                   SmartShunt
com.victronenergy.battery.ttyUSB2                                                         SerialBattery(Jkbms)
com.victronenergy.fronius
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                                               Package Manager
com.victronenergy.platform                                                                               Venus
com.victronenergy.settings
com.victronenergy.shutdown                                                                     ShutdownMonitor
com.victronenergy.solarcharger.ttyUSB1                                                            MPPT Pergola
com.victronenergy.system
com.victronenergy.vebus.ttyUSB3                                                     MultiPlus-II 48/5000/70-50`

Please attach any relevant logs. Log location is `/var/log/BatteryAggregator/current`:

`@4000000066f7e98a18b602fc Traceback (most recent call last):
@4000000066f7e98a18b62a0c   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e98a18b63d94     from pydoc import locate
@4000000066f7e98a18b6494c ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e98c0f36e3cc Traceback (most recent call last):
@4000000066f7e98c0f370adc   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e98c0f37224c     from pydoc import locate
@4000000066f7e98c0f372a1c ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e98e2053a08c Traceback (most recent call last):
@4000000066f7e98e2053c3b4   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e98e2053db24     from pydoc import locate
@4000000066f7e98e2053e6dc ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e9901acc4484 Traceback (most recent call last):
@4000000066f7e9901acc6b94   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e9901acc8304     from pydoc import locate
@4000000066f7e9901acc8ebc ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e992161b8db4 Traceback (most recent call last):
@4000000066f7e992161bb8ac   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e992161bd01c     from pydoc import locate
@4000000066f7e992161bd7ec ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e9940b92b714 Traceback (most recent call last):
@4000000066f7e9940b92e5f4   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e9940b92fd64     from pydoc import locate
@4000000066f7e9940b93091c ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e99604e2733c Traceback (most recent call last):
@4000000066f7e99604e29664   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e99604e2add4     from pydoc import locate
@4000000066f7e99604e2b5a4 ModuleNotFoundError: No module named 'pydoc'
@4000000066f7e9973b90ec9c Traceback (most recent call last):
@4000000066f7e9973b9113ac   File "/data/BatteryAggregator/battery_service.py", line 24, in <module>
@4000000066f7e9973b912b1c     from pydoc import locate
@4000000066f7e9973b9136d4 ModuleNotFoundError: No module named 'pydoc'`

I'm not a specialist, but may be a python update in the newer venusos version is causing this.
I was looking if I can install something like 'pydoc' or 'python3-pydoc' via opkg but did not find something like that.
Any ideas, who this can be fixed shorthand?

I'm going to install an older version on a different sd card in the meantime

tazinblack, 2024-09-28:
ok, can be closed. 
After running `bash /data/BatteryAggregator/check_config.sh` missing packages were installed

Also looks like a duplicate of [](https://github.com/pulquero/BatteryAggregator/issues/71#top)

pulquero, 2024-09-28:
Supposedly this is meant to be fixed in the setup script now.

closed 2024-09-28 by pulquero