pulquero/BatteryAggregator #76: CCL 30A and Battery charge is 88A

Rubens974, 2024-09-30:
Please include system details:

Hardware: RPi

Setup: DVCC
[image]

config
{
"logLevel": "DEBUG",
"capacity": 500,
"currentRatioMethod": "capacity",
"cvlMode": "min_when_balancing"
}

Hi, 
Can you tell me if this is normal?
 
[image]
[image]

Best regards

pulquero, 2024-09-30:
The current logic is if "limit charge current" is off then the ccl setting is ignored and it is assumed the bms provides the ccl.

Rubens974, 2024-09-30:
Thank you 
So I must activate "limit charge current" in DVCC menue
I will try this.

Regards

closed 2024-10-12 by pulquero